zwave-js/zwave-js #4795: Heatit Z-Dim2 250W

geirra, 2022-07-12:
### Checklist

- [X] It is not in the [configuration DB](https://devices.zwave-js.io/?search=3)

- [X] It was not [merged recently](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Apr+is%3Aclosed) or has a [pending PR](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Aopen+is%3Apr)

### Which device is missing?

Heatit Z-Dim2 250w

### Manufacturer ID

0x019b

### Product Type

0x0002

### Product ID

0x2201

### Firmware Version

1.20

### Is the device listed on the Z-Wave Alliance website?

https://products.z-wavealliance.org/products/4666

### Do you have a manual?

[1444440-Manual_Heatit_Z-Dim2  250W.pdf](https://github.com/zwave-js/node-zwave-js/files/9094141/1444440-Manual_Heatit_Z-Dim2.250W.pdf)

geirra, 2022-07-12:
Can help with the config file, but as i understand it we need to import it from the zwave db first?

AlCalzone, 2022-07-13:
@zwave-js-bot import config 4666 from zwa

geirra, 2022-07-19:
solved with merge of #4801

closed 2022-07-19 by geirra