zwave-js/zwave-js #4799: [Bug]: `getAvailableFirmwareUpdates` is calling `waitForWakeup` which throws

marcus-j-davies, 2022-07-13:
More than happy to provide a log, but its pretty trivial 😅 

 `getAvailableFirmwareUpdates` is calling into  `node.waitForWakeup()`
 https://github.com/zwave-js/node-zwave-js/blob/bd5aa02022ee1c4adc96cf884a2a956081bf2eb2/packages/zwave-js/src/lib/controller/Controller.ts#L4995
 
 but `waitForWakeup()` will throw for devices that don't sleep.
https://github.com/zwave-js/node-zwave-js/blob/bd5aa02022ee1c4adc96cf884a2a956081bf2eb2/packages/zwave-js/src/lib/node/Node.ts#L507
 
 And the exception is not handled in `getAvailableFirmwareUpdates`.

marcus-j-davies, 2022-07-14:
I have taken liberty of suggesting a patch in the form of a PR.

AlCalzone, 2022-07-14:
Nice, will check it on Monday

closed 2022-07-19 by AlCalzone