zwave-js/zwave-js #4805: [Bug]: Including a device twice makes it inaccessible until excluded and included again

Lovasz-Akos, 2022-07-14:
### Checklist

- [X]  I am not using Home Assistant. Or: a developer has told me to come here.
- [X] I have checked the troubleshooting section and my problem is not described there.
- [X] I have read the changelog and my problem is not mentioned there.

### Deploy method

Manually built (git clone - yarn build - yarn install

### Zwavejs2Mqtt version

6.13.0

### ZwaveJS version

9.5.0

### Describe the bug

Including a device that's already included will make it become inaccessible even if the given name and/or location matches the already included one.

### To Reproduce

Include a zwave device, then, include it once again.

### Expected behavior

Either the inclusion gets declined or the device gets a new name/location, and it still accessible, even if under a new path/id/mqtt topic.

### Additional context

Excluding then Including the device fixes this

Lovasz-Akos, 2022-07-15:
Inaccessible as in it doesn't get messages I'd try to send it. Be it just a ping or a value change. This includes sending the message to it's ID and if the location/name changed on the new inclusion, neither the old or the new pairing of name/location mqtt topic, ex.: `zwave/office/dimmer/38/1/targetValue/set`

Lovasz-Akos, 2022-07-15:
when a device is paired for a second time, the UI shows that device with ID `shows the ID of the device that was already paired` successfully included, but then it can't be controlled from APIs, MQTT, or the UI itself until it gets excluded and re-included under a new ID

AlCalzone, 2022-07-15:
I'll try to reproduce next week, but I'm afraid this is either device or controller specific behavior.

AlCalzone, 2022-07-26:
Can confirm that weird stuff happens when doing this. Restarting and/or re-interviewing gets things back to normal, but we should probably just not react when a node is "added" with an ID that is already part of the network.

PR #6180 (2023-08-17): fix: abort inclusion when node is already part of the network
fixes: #4805