zwave-js/zwave-js #4811: currentColor property reported as null

amilanov75, 2022-07-18:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within ZWaveJS2MQTT?

NO, my problem is NOT within ZWaveJS2MQTT

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

The currenColor properties for RGBW devices is being reported as "null" for all properties i.e. R, G, B and WW.

The expected behaviour is that a value between 0 and 255 should always be reported against each channel.

The tested device is Fibaro RGBW2.

Testing 1:
An interview of the device will return "null"
A getValueDB (in NR) will then return "null"

Testing 2:
A poll of the device will return a value
A getValueDB (in NR) will then return a value

It is not clear why the value is being dropped in the first instance i.e. why null is being reported as there is no sign of "mass" interviews happening to reset the property from a valid value to "null".

Versions:
NR v3.0.0 
nr-zwave-js v7.1.0
Nr driver v9.3.0
node-js v14.18.3

Image info - message received after a poll:
prevValue = null (this was before a poll)
new Value = 0 this from the poll
[image]

### Device information

refer above

### How are you using `node-zwave-js`?

- [ ] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [ ] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [X] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

refer above

### Did you change anything?

no

### Did this work before?

Don't know, this is a new device

### If yes, where did it work?

Not sure how long this has not been working for as my code will set the target colour whenever the light is turned on, if the target does not equal the current colour, so the issue would have been covered up.

### Attach Driver Logfile

no log provided

AlCalzone, 2022-07-18:
@marcus-j-davies the NR-plugin is listening for the `"value added"` event, right?
I do see these logs during the interview (same for the other channels), which means the values are being added/created there, as opposed to updated:
```
2022-07-18T11:12:47.920Z CNTRLR   [Node 042] [+] [Color Switch] currentColor[0]: 0                  [Endpoint 0]
2022-07-18T11:12:47.921Z CNTRLR   [Node 042] [+] [Color Switch] targetColor[0]: 0                   [Endpoint 0]
2022-07-18T11:12:47.921Z CNTRLR   [Node 042] [+] [Color Switch] currentColor: {"warmWhite":0}       [Endpoint 0]
2022-07-18T11:12:47.921Z CNTRLR   [Node 042] [+] [Color Switch] targetColor: {"warmWhite":0}        [Endpoint 0]
```

marcus-j-davies, 2022-07-18:
@AlCalzone

I do, but I pass them to user land as **VALUE_UPDATED**, as their makeup is extremely similar.

https://github.com/zwave-js/node-red-contrib-zwave-js/blob/b21cf7a4bb65d78a9ad8a8d25cf0e2607c4f0e2d/zwave-js/zwave-js.js#L2026

I worked on this issues this morning on discord, after an interview, `currentColor` is **null** (for all components) (using `getValue`), but only after a `pollValue` are they available (at least that's what I witnessed) 😅

marcus-j-davies, 2022-07-18:
Is this another endpoint madness thing?, as the endpoint in question is 1, but logs shows 1 & 0 🤷

AlCalzone, 2022-07-18:
I also thought of an endpoint thing first, but both endpoints have the values. Also, the driver only uses `null` instead of `undefined` in a handful of locations. Values aren't one of them.

Do you read all values after the node is ready? Or how are the values reaching userland?

amilanov75, 2022-07-19:
"when executing getValue by the user - its returning null for currentColor, until pollValue is executed.
it seems after an interview it reverts back to null @amilanov75 please confirm that is the case."

Interview does not revert back to null, which means that as it stands, I do not know what is triggering a null value being populated in the first instance.

amilanov75, 2022-07-19:
Yes, but that was before I re-instated my code to work as expected, after which the lighting function was actually sending colour commands to the lights.

Do you recall me saying I was seeing inconsistent results. I said that after I sent a command to set the colour for a light, as I was then seeing the colour values stick i.e. the nulls did not return.

So, "something" happens to trigger the null values, after which we can only temporarily replace them with actual values with a poll, but once an actual colour command is sent, the messages report correct values, until the "something" event happens again.

Hence my statement above: "I do not know what is triggering a null value being populated in the first instance."

AlCalzone, 2022-07-19:
Which value IDs are we talking about anyways?
I can see that the value events for endpoint 1 reach NR:
```
2022-07-18T11:12:38.440Z CNTRLR   [Node 042] [~] [Color Switch] duration: {"value":0,"unit":"second [Endpoint 1]
                                  s"} => {"value":0,"unit":"seconds"}
2022-07-18T11:12:38.440Z NDERED » [Node: 42] [VALUE_UPDATED] [DIRECT] Forwarding payload...
2022-07-18T11:12:38.440Z NDERED » [Node: 42] [VALUE_UPDATED] [EVENT] Forwarding payload...
2022-07-18T11:12:38.441Z CNTRLR   [Node 042] [+] [Color Switch] currentColor[4]: 0                  [Endpoint 1]
2022-07-18T11:12:38.441Z NDERED » [Node: 42] [VALUE_UPDATED] [DIRECT] Forwarding payload...
2022-07-18T11:12:38.441Z NDERED » [Node: 42] [VALUE_UPDATED] [EVENT] Forwarding payload...
2022-07-18T11:12:38.441Z CNTRLR   [Node 042] [+] [Color Switch] targetColor[4]: 0                   [Endpoint 1]
2022-07-18T11:12:38.441Z NDERED » [Node: 42] [VALUE_UPDATED] [DIRECT] Forwarding payload...
2022-07-18T11:12:38.442Z NDERED » [Node: 42] [VALUE_UPDATED] [EVENT] Forwarding payload...
2022-07-18T11:12:38.442Z CNTRLR   [Node 042] [~] [Color Switch] currentColor: {"warmWhite":0,"red": [Endpoint 1]
                                  0,"green":0,"blue":0} => {"warmWhite":0,"red":0,"green":0,"blue":0}
2022-07-18T11:12:38.442Z NDERED » [Node: 42] [VALUE_UPDATED] [DIRECT] Forwarding payload...
2022-07-18T11:12:38.442Z NDERED » [Node: 42] [VALUE_UPDATED] [EVENT] Forwarding payload...
2022-07-18T11:12:38.443Z CNTRLR   [Node 042] [~] [Color Switch] targetColor: {"warmWhite":0,"red":0 [Endpoint 1]
                                  ,"green":0,"blue":0} => {"warmWhite":0,"red":0,"green":0,"blue":0}
2022-07-18T11:12:38.443Z NDERED » [Node: 42] [VALUE_UPDATED] [DIRECT] Forwarding payload...
2022-07-18T11:12:38.443Z NDERED » [Node: 42] [VALUE_UPDATED] [EVENT] Forwarding payload...
2022-07-18T11:12:38.443Z CNTRLR   [Node 042] [~] [Color Switch] hexColor: "000000" => "000000"      [Endpoint 1]
2022-07-18T11:12:38.444Z NDERED » [Node: 42] [VALUE_UPDATED] [DIRECT] Forwarding payload...
2022-07-18T11:12:38.444Z NDERED » [Node: 42] [VALUE_UPDATED] [EVENT] Forwarding payload...
```

but the events for endpoint 0 do not since they duplicate endpoint 1 and are filtered on the Z-Wave JS side.

marcus-j-davies, 2022-07-19:
@AlCalzone to my knowledge it's all the property keys R,G,B,WW for currentColor on endpoint 1.

@amilanov75 I'm really hoping your flows are not setting null for currentColor and in error, but I think the driver validates the value for currentColor 🤞

amilanov75, 2022-07-19:
ok, so I have been looking at my lights and they are no ALL reporting values NOT null.

I have checked a light that I am fairly confident would not have been turned on in the last 24 hours and it has values again, so something has happened to start reporting value not nulls.

Either you can see something in the code, else it's an intermitted issue that may be hard to pin down.

marcus-j-davies, 2022-07-19:
Ok,

On the basis your flows (and dashboards) are not somehow setting `null` values for `targetColor`, which will then translate to `currentColor `(i think 😅 ) - but then the driver should validate the value for `targetColor` if using the ValueAPI

I'm at a complete loss.
data is coming from the driver and to the driver, the module doesn't massage them to introduce a `null` value in anyway.

I'm lost, as i can see nothing.

`getValueDB` is just fetching the value from the drivers data cache
https://github.com/zwave-js/node-red-contrib-zwave-js/blob/b21cf7a4bb65d78a9ad8a8d25cf0e2607c4f0e2d/zwave-js/zwave-js.js#L1393

`setValue` is just passing it directly to the  driver
https://github.com/zwave-js/node-red-contrib-zwave-js/blob/b21cf7a4bb65d78a9ad8a8d25cf0e2607c4f0e2d/zwave-js/zwave-js.js#L1200

amilanov75, 2022-07-19:
@marcus-j-davies  in the world of possible explanations, you have provided one that could work, albeit (hopefully) unlikely.
"Could I have introduced a bug into my code that sent a null value message to the controller. "

Have you tested if sending a null value, then translates into the null value being stored for future retrieval? If not, I can test it later to see if it does in fact do that.

I am scratching my head now to think about how to check the above hypothesis given all the code changes that I have been making on a daily basis.

Ok, let me test sending "null"  first....

amilanov75, 2022-07-19:
Actually no, I am not going to do test the sending "null" hypotheses.

Recall what happened yesterday, the null value kept coming back right after a poll set it correctly. 

So, unless a null value being sent, does something weird/unexpected i.e. somehow makes the null stick - which really would be out there, that can't be the explanation.

marcus-j-davies, 2022-07-19:
> Recall what happened yesterday, the null value kept coming back right after a poll set it correctly.

Ok - lets rule out any user flow being the cause.

Attach a debug node to the link node(s), right before they hit the controller.
then do stuff (including a poll - why not)

what we should NOT see is any message where the target is being set to `null`
I'm out of idea - so clutching at straws

[image]

amilanov75, 2022-07-19:
I thought about doing the same, then stopped myself as it didn't feel like a good use of time.

I'll do it later to be complete and see if anything ever pops up.

marcus-j-davies, 2022-07-19:
> IF you send a null value

The Driver should be validating values for the ValueAPI:setValue commands, and it does.

[image]

So I would expect the same for Color Switch:`targetColor`
but if its not, and for some reason your sending null values - then this may be the source of madness

marcus-j-davies, 2022-07-19:
@AlCalzone

I am not sure if this is of Value or should be the desired affect, but I have just been able to set `null` for each colour component 

(connected to a friends VPN and tested  - as I do not have RGBW controllers)

and `null` is indeed returned for current colour.

**Setting**

```javascript
{
    "mode": "ValueAPI",
    "method": "setValue",
    "params": [
        {
            "commandClassName": "Color Switch",
            "commandClass": 51,
            "endpoint": 0,
            "property": "targetColor",
            "propertyName": "targetColor"
        },
        {
            "warmWhite": null,
            "red": null,
            "green": null,
            "blue": null
        }
    ]
}

```

Current Color
```javascript
{
   "valueId": {
      "commandClassName": "Color Switch",
      "commandClass": 51,
      "endpoint": 0,
      "property": "currentColor",
      "propertyName": "currentColor"
   },
   "meta": {
      "type": "any",
      "readable": true,
      "writeable": false,
      "label": "Current Color"
   },
   "propertyid": "0-51-currentColor-",
   "endpoint": 0,
   "valueData": {
      "value": {
         "warmWhite": null,
         "red": null,
         "green": null,
         "blue": null
      }
   }
}
```

@amilanov75 
based on this, I am under the believe `null` was being sent from your flow somehow, and will indeed be committed.

AlCalzone, 2022-07-20:
Ohh, it just verifies that each key is valid, but not the actual values.

amilanov75, 2022-07-20:
Hi @marcus-j-davies, you beat me to it. I was just sitting down with a coffee about to test the above. Thank you for doing so yourself!.

For the time being and given that the issue happened so quickly, I would have to agree with you that my code (as we discussed, which was in transit and in which I'd already introduced a bug into that uncovered this issue was even happening) is currently the most likely candidate for causing null to be reported in the device messages. I can't think of any other event that happened over the 3 day period in question that could be reasonably attributed to the issue. I'd already updated NR to 3.0.0 one day before the three day window in which I determined that bug was happening, so that takes NR out of the question.

I will still put a look on my inbound messages, until the above bug is resolved, to see if "null" ever gets written, however unless a new bug is introduced into my code, I doubt this will be the case.

@marcus-j-davies  
In your testing, did you check after writing the null value if the exact same behaviour was happening as per the screenshots I shared with you in our PMs the other day? 
i.e. could you see a value in getValueDB by using getValue and then poll to see it return to null in the getValueDB?  
I think this is an important nuance of the bug that @AlCalzone should be aware of.
If not, I'll can run the tests again and provide fresh screenshoots + anything else @AlCalzone requests e.g. could also provide logs, if it helps - although not sure they will in this situation.

AlCalzone, 2022-07-20:
I'm pretty sure this is fixed with the linked PR. `setValue` didn't throw, although the passed value was invalid, which caused the `currentColor` to get updated optimistically with invalid values.