zwave-js/zwave-js #4835: S2 Multicast

mbbush, 2022-07-22:
I recently migrated my entire zwave network from hubitat to zwavejs. One of the main features I was looking forward to was multicast. I included most of my devices with S2, which my (old) hubitat didn't support, mostly for the message encapsulation and reliability benefits; I'm not worried about the security of my zwave network. Unfortunately, after setting everything up, I leaned that zwavejs doesn't (yet) support multicast to S2 devices. 

**Describe the solution you'd like**
S2 multicast is clearly on [the roadmap](https://github.com/zwave-js/node-zwave-js/issues/4312). I tried to find an issue tracking progress, or describing the challenges involved, and didn't find one. So maybe this can be that issue?

**Describe alternatives you've considered**
I could exclude and re-include every device in my network without security, or live without multicast support.

**Additional context**
I imagine the steps needed to send a S2 multicast would be, for each multicast message,
1. Partition the nodes by security class (none, S0, S2 Unauthenticated, S2 Authenticated, S2 Access Control)
2. For each security class with at least one target node in it, send one multicast message to those nodes.
Are there other significant complications I'm not aware of? Is S2 encryption of multicast messages qualitatively different than S2 encryption of unicast messages?

AlCalzone, 2022-07-22:
S2 multicast messaging is a bit more involved than singlecast, because the singlecast followup that is done by the firmware for "normal" commands must now be handled by the driver.

If you want to read a bit further:
https://github.com/zwave-js/specs/blob/master/ZWA_Z-Wave%20Transport-Encapsulation%20Command%20Class%20Specification.pdf, section 3.6.5.2

mbbush, 2022-07-22:
I see the mention of the the S2 SC-Followup messages in that specification, but I didn't find any mention of 
> the singlecast followup that is done by the firmware for "normal" commands 

in either the [network layer](https://raw.githubusercontent.com/zwave-js/specs/master/ZWA_Z-Wave%20Network%20Protocol%20Command%20Class%20Specification_SPE_11.0.0.pdf) or [MAC/PHY layer](https://github.com/zwave-js/specs/blob/master/T-REC-G.9959-201501.pdf) specifications. Where is that specified?

This does indeed seem like a lot of work. In addition to the recommendation for the followup message, it seems like managing the MPANs in addition to the SPANs is indeed a qualitative difference from singlecast S2. I did notice that the singlecast followup message is a "SHOULD" not a "MUST" in the specification, so perhaps it's worth starting with just the multicast nonces?

Is it just a matter of finding the time to do it? Are there any design questions that are still outstanding, or will you not know that until you get deep into the weeds of actually implementing it? I don't want to interfere unhelpfully, but I'm both curious about how this works under the hood, and experienced in designing software in general. My javascript skills in particular, however, are approximately zero.

AlCalzone, 2022-07-25:
> Where is that specified?

That is always the key question when it comes to Z-Wave. For this, it is hidden in the description for the `TRANSMIT_OPTION_ACK` of the `ZW_SendDataMulti` function in the 500 series appl. programmers Guide:
> The multicast frame will be followed by a number of single cast frames to each of the destination nodes and request acknowledge from each destination node.

Since multicast frames are not routed, this makes sure that each addressed device got the request, even those out of direct range. The initial multicast frame avoids waterfall effects due to latency in communication.

> Is it just a matter of finding the time to do it? 

Mostly this

closed 2023-03-07 by AlCalzone