zwave-js/zwave-js #4839: Missing device configuration: <enter manufacturer and device name>

technoman63, 2022-07-24:
### Checklist

- [X] It is not in the [configuration DB](https://devices.zwave-js.io/?search=3)

- [X] It was not [merged recently](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Apr+is%3Aclosed) or has a [pending PR](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Aopen+is%3Apr)

### Which device is missing?

iSurpass J825

### Manufacturer ID

0x021c

### Product Type

0x5010

### Product ID

0x1001

### Firmware Version

2.16

### Is the device listed on the Z-Wave Alliance website?

No

### Do you have a manual?

[J1825 Wall Reader Instruction.pdf](https://github.com/zwave-js/node-zwave-js/files/9175606/J1825.Wall.Reader.Instruction.pdf)

cyrilst, 2023-04-28:
Hello, here is an imported device configuration for iSurpass j1825 : https://github.com/zwave-js/node-zwave-js/compare/master...cyrilst:node-zwave-js:patch-1

Can I submit a pull request, or do I have to improve the file in some way ?

cyrilst, 2023-05-20:
FYI: pull request is here : https://github.com/zwave-js/node-zwave-js/pull/5823

closed 2023-05-23 by AlCalzone