zwave-js/zwave-js #4841: Issues with TKB Wall switches driver

GAguado, 2022-07-25:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

YES, BUT a Home Assistant developer has told me to come here

### Is your problem within ZWaveJS2MQTT?

NO, my problem is NOT within ZWaveJS2MQTT

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

I have found that the wall switch TKB TZ56D is included incorrectly as a TZ36D and with only the left pad functional (load switch). The right pad, and double right pad sensors are not included/discovered and therefore can not be used.

The same happens with the model TZ66D, any right pad or double right-click functionality is included. In this case, the model TZ66D is properly discovered.
This limits seriously the use of these popular wall switches.

Only entities discovered in both cases (after several trials):

[image]

I'm using the latest version 0.1.64

These devices are properly configured with other Z-Wave controllers

### Device information

Manufacturer: TKB
Model name: TZ56D / TZ66D
Firmware 2.5
0x010c00040008032503ffcc0000e9   [Node 008]

### How are you using `node-zwave-js`?

- [ ] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [X] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

Home Assistant 2022.7.6
Supervisor 2022.07.0
Operating System 8.2
Frontend 20220707.1 - latest

### Did you change anything?

no

### Did this work before?

Don't know, this is a new device

### Attach Driver Logfile

First two messages with left pad use, last two when trying to use the right pad (any message) or double-right:

Subscribed to Z-Wave JS Log Messages…
2022-07-25T08:01:25.028Z SERIAL « 0x010c00040008032503ffc70000e2                                      (14 bytes)
2022-07-25T08:01:25.033Z CNTRLR   [Node 008] [~] [Binary Switch] currentValue: false => true        [Endpoint 0]
2022-07-25T08:01:25.039Z SERIAL » [ACK]                                                                   (0x06)
2022-07-25T08:01:25.056Z DRIVER « [Node 008] [REQ] [ApplicationCommand]
                                  └─[BinarySwitchCCReport]
                                      current value: true
2022-07-25T08:01:27.545Z SERIAL « 0x010c0004000803250300c400001e                                      (14 bytes)
2022-07-25T08:01:27.547Z CNTRLR   [Node 008] [~] [Binary Switch] currentValue: true => false        [Endpoint 0]
2022-07-25T08:01:27.550Z SERIAL » [ACK]                                                                   (0x06)
2022-07-25T08:01:27.553Z DRIVER « [Node 008] [REQ] [ApplicationCommand]
                                  └─[BinarySwitchCCReport]
                                      current value: false
2022-07-25T08:01:49.243Z SERIAL « 0x011700498408110410015e86725a738559252027702b2c7a59                (25 bytes)
2022-07-25T08:01:49.248Z SERIAL » [ACK]                                                                   (0x06)
2022-07-25T08:01:49.252Z DRIVER « [Node 008] [REQ] [ApplicationUpdateRequest]
                                    payload: 0x08110410015e86725a738559252027702b2c7a
2022-07-25T08:01:49.257Z CNTRLR « [Node 008] Received updated node info
2022-07-25T08:02:07.684Z SERIAL « 0x011700498408110410015e86725a738559252027702b2c7a59                (25 bytes)
2022-07-25T08:02:07.686Z SERIAL » [ACK]                                                                   (0x06)
2022-07-25T08:02:07.689Z DRIVER « [Node 008] [REQ] [ApplicationUpdateRequest]
                                    payload: 0x08110410015e86725a738559252027702b2c7a
2022-07-25T08:02:07.692Z CNTRLR « [Node 008] Received updated node info

AlCalzone, 2022-07-25:
Please make a [driver log](https://zwave-js.github.io/zwavejs2mqtt/#/troubleshooting/generating-logs?id=driver-logs), loglevel `debug` or `silly` and **attach** it here as a file (drag & drop).
The logfile should include a re-interview of the device.

GAguado, 2022-07-25:
Attached is the log file

Best Regards
Gerardo

AlCalzone, 2022-07-25:
You'll need to do this on the web UI. Github doesn't let you attach stuff to emails.

GAguado, 2022-07-25:
[zwave_js_TKB_TZ56D.log](https://github.com/zwave-js/node-zwave-js/files/9182774/zwave_js_TKB_TZ56D.log)

AlCalzone, 2022-07-26:
> The logfile should include a re-interview of the device.

Your logfile does not.

GAguado, 2022-07-26:
Not easy, at least to me,.. device logs din't show anything after re-interview. I had to go to general LOG and filtering by Zwave JS, please see below:  (there is not any icon to save the log to file)

2022-07-26T16:35:39.772Z CNTRLR   [Node 008] [Scene Actuator Configuration] level[233]: metadata up [Endpoint 0]
                                  dated
2022-07-26T16:35:39.774Z CNTRLR   [Node 008] [Scene Actuator Configuration] dimmingDuration[233]: m [Endpoint 0]
                                  etadata updated
2022-07-26T16:35:39.776Z CNTRLR   [Node 008] [Scene Actuator Configuration] level[234]: metadata up [Endpoint 0]
                                  dated
2022-07-26T16:35:39.778Z CNTRLR   [Node 008] [Scene Actuator Configuration] dimmingDuration[234]: m [Endpoint 0]
                                  etadata updated
2022-07-26T16:35:39.780Z CNTRLR   [Node 008] [Scene Actuator Configuration] level[235]: metadata up [Endpoint 0]
                                  dated
2022-07-26T16:35:39.786Z CNTRLR   [Node 008] [Scene Actuator Configuration] dimmingDuration[235]: m [Endpoint 0]
                                  etadata updated
2022-07-26T16:35:39.787Z CNTRLR   [Node 008] [Scene Actuator Configuration] level[236]: metadata up [Endpoint 0]
                                  dated
2022-07-26T16:35:39.788Z CNTRLR   [Node 008] [Scene Actuator Configuration] dimmingDuration[236]: m [Endpoint 0]
                                  etadata updated
2022-07-26T16:35:39.790Z CNTRLR   [Node 008] [Scene Actuator Configuration] level[237]: metadata up [Endpoint 0]
                                  dated
2022-07-26T16:35:39.792Z CNTRLR   [Node 008] [Scene Actuator Configuration] dimmingDuration[237]: m [Endpoint 0]
                                  etadata updated
2022-07-26T16:35:39.805Z CNTRLR   [Node 008] [Scene Actuator Configuration] level[238]: metadata up [Endpoint 0]
                                  dated
2022-07-26T16:35:39.806Z CNTRLR   [Node 008] [Scene Actuator Configuration] dimmingDuration[238]: m [Endpoint 0]
                                  etadata updated
2022-07-26T16:35:39.808Z CNTRLR   [Node 008] [Scene Actuator Configuration] level[239]: metadata up [Endpoint 0]
                                  dated
2022-07-26T16:35:39.809Z CNTRLR   [Node 008] [Scene Actuator Configuration] dimmingDuration[239]: m [Endpoint 0]
                                  etadata updated
2022-07-26T16:35:39.811Z CNTRLR   [Node 008] [Scene Actuator Configuration] level[240]: metadata up [Endpoint 0]
                                  dated
2022-07-26T16:35:39.812Z CNTRLR   [Node 008] [Scene Actuator Configuration] dimmingDuration[240]: m [Endpoint 0]
                                  etadata updated
2022-07-26T16:35:39.815Z CNTRLR   [Node 008] [Scene Actuator Configuration] level[241]: metadata up [Endpoint 0]
                                  dated
2022-07-26T16:35:39.816Z CNTRLR   [Node 008] [Scene Actuator Configuration] dimmingDuration[241]: m [Endpoint 0]
                                  etadata updated
2022-07-26T16:35:39.818Z CNTRLR   [Node 008] [Scene Actuator Configuration] level[242]: metadata up [Endpoint 0]
                                  dated
2022-07-26T16:35:39.819Z CNTRLR   [Node 008] [Scene Actuator Configuration] dimmingDuration[242]: m [Endpoint 0]
                                  etadata updated
2022-07-26T16:35:39.820Z CNTRLR   [Node 008] [Scene Actuator Configuration] level[243]: metadata up [Endpoint 0]
                                  dated
2022-07-26T16:35:39.822Z CNTRLR   [Node 008] [Scene Actuator Configuration] dimmingDuration[243]: m [Endpoint 0]
                                  etadata updated
2022-07-26T16:35:39.824Z CNTRLR   [Node 008] [Scene Actuator Configuration] level[244]: metadata up [Endpoint 0]
                                  dated
2022-07-26T16:35:39.825Z CNTRLR   [Node 008] [Scene Actuator Configuration] dimmingDuration[244]: m [Endpoint 0]
                                  etadata updated
2022-07-26T16:35:39.827Z CNTRLR   [Node 008] [Scene Actuator Configuration] level[245]: metadata up [Endpoint 0]
                                  dated
2022-07-26T16:35:39.829Z CNTRLR   [Node 008] [Scene Actuator Configuration] dimmingDuration[245]: m [Endpoint 0]
                                  etadata updated
2022-07-26T16:35:39.830Z CNTRLR   [Node 008] [Scene Actuator Configuration] level[246]: metadata up [Endpoint 0]
                                  dated
2022-07-26T16:35:39.832Z CNTRLR   [Node 008] [Scene Actuator Configuration] dimmingDuration[246]: m [Endpoint 0]
                                  etadata updated
2022-07-26T16:35:39.833Z CNTRLR   [Node 008] [Scene Actuator Configuration] level[247]: metadata up [Endpoint 0]
                                  dated
2022-07-26T16:35:39.834Z CNTRLR   [Node 008] [Scene Actuator Configuration] dimmingDuration[247]: m [Endpoint 0]
                                  etadata updated
2022-07-26T16:35:39.836Z CNTRLR   [Node 008] [Scene Actuator Configuration] level[248]: metadata up [Endpoint 0]
                                  dated
2022-07-26T16:35:39.842Z CNTRLR   [Node 008] [Scene Actuator Configuration] dimmingDuration[248]: m [Endpoint 0]
                                  etadata updated
2022-07-26T16:35:39.844Z CNTRLR   [Node 008] [Scene Actuator Configuration] level[249]: metadata up [Endpoint 0]
                                  dated
[... cut ...]
2022-07-26T16:35:39.862Z CNTRLR   [Node 008] [+] [Scene Actuator Configuration] intervie [Endpoint 0] [internal]
                                  wComplete: true
2022-07-26T16:35:39.878Z CNTRLR   [Node 008] Interview stage completed: CommandClasses
2022-07-26T16:35:39.884Z CNTRLR   [Node 008] Interview stage completed: OverwriteConfig
2022-07-26T16:35:39.886Z CNTRLR   [Node 008] Interview completed
2022-07-26T16:35:39.891Z CNTRLR   [Node 008] The node is ready to be used
Starting logging event forwarder at silly level
Stopping logging event forwarder

GAguado, 2022-08-29:
Dear Team,
I noticed that the issue was marked as closed, there is a problem with TKB wall switches, not only one model but at least two. These work flawless with another Zwave controllers but not with HA/Zwave JS
Please let me know how can I help you to fix it, I'm moving to HA but with 25 TKB switches in place I can not
Thanks in advance and best regards
Gerardo

AlCalzone, 2022-08-29:
Please provide the logs I requested above.

GAguado, 2022-08-29:
Hi !
Thanks for your quick action, I thought I sent all LOG files you asked me
for. As newbie with HA I could be wrong, please double check and let me
know which Log you need and I will try it. Nevertheless I guess the fault
is clear, TKB switches (double pad) sense the second (right) pad click
either single or double (on and off) and this sensing function is not
apparently consider on JS driver. There is also a mistake with the model as
HA recognize it but with another model as said in my first message.
Thanks !!
Gerardo

AlCalzone, 2022-08-29:
You just posted the last few lines of an interview (not sure which loglevel). I need:
- the entire thing
- on loglevel "debug"
- attached here as a file (drag & drop into the text box on GitHub)

If you're unsure how to get the log, please ask in the HA forums or on Discord - there should be someone familiar enough with HA to help you.

GAguado, 2022-08-29:
Hi, I send you two logs.
First below is the Zwave JS when clicking on the left pad (on+off) and right pad (on+off) + double click on right (on+off) 
Debug level
Nothing happens when clicking on right pad. An additional message with TKB device diagnostics later. 

2022-08-29T15:45:37.379Z CNTRLR   [Node 008] [Scene Actuator Configuration] level[233]: metadata up [Endpoint 0]
                                  dated
2022-08-29T15:45:37.380Z CNTRLR   [Node 008] [Scene Actuator Configuration] dimmingDuration[233]: m [Endpoint 0]
                                  etadata updated
2022-08-29T15:45:37.382Z CNTRLR   [Node 008] [Scene Actuator Configuration] level[234]: metadata up [Endpoint 0]
                                  dated
2022-08-29T15:45:37.383Z CNTRLR   [Node 008] [Scene Actuator Configuration] dimmingDuration[234]: m [Endpoint 0]
                                  etadata updated
2022-08-29T15:45:37.384Z CNTRLR   [Node 008] [Scene Actuator Configuration] level[235]: metadata up [Endpoint 0]
                                  dated
2022-08-29T15:45:37.385Z CNTRLR   [Node 008] [Scene Actuator Configuration] dimmingDuration[235]: m [Endpoint 0]
                                  etadata updated
2022-08-29T15:45:37.386Z CNTRLR   [Node 008] [Scene Actuator Configuration] level[236]: metadata up [Endpoint 0]
                                  dated
2022-08-29T15:45:37.387Z CNTRLR   [Node 008] [Scene Actuator Configuration] dimmingDuration[236]: m [Endpoint 0]
                                  etadata updated
2022-08-29T15:45:37.388Z CNTRLR   [Node 008] [Scene Actuator Configuration] level[237]: metadata up [Endpoint 0]
                                  dated
2022-08-29T15:45:37.389Z CNTRLR   [Node 008] [Scene Actuator Configuration] dimmingDuration[237]: m [Endpoint 0]
                                  etadata updated
2022-08-29T15:45:37.390Z CNTRLR   [Node 008] [Scene Actuator Configuration] level[238]: metadata up [Endpoint 0]
                                  dated
2022-08-29T15:45:37.391Z CNTRLR   [Node 008] [Scene Actuator Configuration] dimmingDuration[238]: m [Endpoint 0]
                                  etadata updated
2022-08-29T15:45:37.392Z CNTRLR   [Node 008] [Scene Actuator Configuration] level[239]: metadata up [Endpoint 0]
                                  dated
2022-08-29T15:45:37.398Z CNTRLR   [Node 008] [Scene Actuator Configuration] dimmingDuration[239]: m [Endpoint 0]
                                  etadata updated
2022-08-29T15:45:37.400Z CNTRLR   [Node 008] [Scene Actuator Configuration] level[240]: metadata up [Endpoint 0]
                                  dated
2022-08-29T15:45:37.401Z CNTRLR   [Node 008] [Scene Actuator Configuration] dimmingDuration[240]: m [Endpoint 0]
                                  etadata updated
2022-08-29T15:45:37.403Z CNTRLR   [Node 008] [Scene Actuator Configuration] level[241]: metadata up [Endpoint 0]
                                  dated
2022-08-29T15:45:37.405Z CNTRLR   [Node 008] [Scene Actuator Configuration] dimmingDuration[241]: m [Endpoint 0]
                                  etadata updated
2022-08-29T15:45:37.406Z CNTRLR   [Node 008] [Scene Actuator Configuration] level[242]: metadata up [Endpoint 0]
                                  dated
2022-08-29T15:45:37.408Z CNTRLR   [Node 008] [Scene Actuator Configuration] dimmingDuration[242]: m [Endpoint 0]
                                  etadata updated
2022-08-29T15:45:37.409Z CNTRLR   [Node 008] [Scene Actuator Configuration] level[243]: metadata up [Endpoint 0]
                                  dated
2022-08-29T15:45:37.411Z CNTRLR   [Node 008] [Scene Actuator Configuration] dimmingDuration[243]: m [Endpoint 0]
                                  etadata updated
2022-08-29T15:45:37.413Z CNTRLR   [Node 008] [Scene Actuator Configuration] level[244]: metadata up [Endpoint 0]
                                  dated
2022-08-29T15:45:37.415Z CNTRLR   [Node 008] [Scene Actuator Configuration] dimmingDuration[244]: m [Endpoint 0]
                                  etadata updated
2022-08-29T15:45:37.417Z CNTRLR   [Node 008] [Scene Actuator Configuration] level[245]: metadata up [Endpoint 0]
                                  dated
2022-08-29T15:45:37.422Z CNTRLR   [Node 008] [Scene Actuator Configuration] dimmingDuration[245]: m [Endpoint 0]
                                  etadata updated
2022-08-29T15:45:37.424Z CNTRLR   [Node 008] [Scene Actuator Configuration] level[246]: metadata up [Endpoint 0]
                                  dated
2022-08-29T15:45:37.433Z CNTRLR   [Node 008] [Scene Actuator Configuration] dimmingDuration[246]: m [Endpoint 0]
                                  etadata updated
2022-08-29T15:45:37.435Z CNTRLR   [Node 008] [Scene Actuator Configuration] level[247]: metadata up [Endpoint 0]
                                  dated
2022-08-29T15:45:37.436Z CNTRLR   [Node 008] [Scene Actuator Configuration] dimmingDuration[247]: m [Endpoint 0]
                                  etadata updated
2022-08-29T15:45:37.442Z CNTRLR   [Node 008] [Scene Actuator Configuration] level[248]: metadata up [Endpoint 0]
                                  dated
2022-08-29T15:45:37.443Z CNTRLR   [Node 008] [Scene Actuator Configuration] dimmingDuration[248]: m [Endpoint 0]
                                  etadata updated
2022-08-29T15:45:37.445Z CNTRLR   [Node 008] [Scene Actuator Configuration] level[249]: metadata up [Endpoint 0]
[... 10 lines cut ...]
[... cut ...]
2022-08-29T15:45:37.483Z CNTRLR   [Node 008] [+] [Scene Actuator Configuration] intervie [Endpoint 0] [internal]
                                  wComplete: true
2022-08-29T15:45:37.485Z CNTRLR   [Node 008] Interview stage completed: CommandClasses
2022-08-29T15:45:37.487Z CNTRLR   [Node 008] Interview stage completed: OverwriteConfig
2022-08-29T15:45:37.489Z CNTRLR   [Node 008] Interview completed
2022-08-29T15:45:37.490Z CNTRLR   [Node 008] The node is ready to be used
Starting logging event forwarder at debug level
Stopping logging event forwarder

GAguado, 2022-08-29:
[Diagnostics_TKB_TZ36D.log](https://github.com/zwave-js/node-zwave-js/files/9446132/Diagnostics_TKB_TZ36D.log)

GAguado, 2022-08-29:
I hope this is the information you need

AlCalzone, 2022-08-29:
Thats the same stuff you posted above though.

kpine, 2022-08-29:
Follow the instructions in the docs: https://www.home-assistant.io/integrations/zwave_js/#how-to-access-the-z-wave-logs

Set the log level to "debug" as requested. Pay special attention to note in the docs:

> **You need to keep this browser tab open for logging to be active.**

Which means, keep one browser window open on the log window to capture the logs, and a second browser window to perform the interview.

Use the download button in the UI to download the logs as a file.

GAguado, 2022-08-30:
Hi, at last please find it enclosed, thanks for your patience
[Re-interview a Z-Wave Device_TZ56D.txt](https://github.com/zwave-js/node-zwave-js/files/9454965/Re-interview.a.Z-Wave.Device_TZ56D.txt)

AlCalzone, 2022-08-30:
Hooray. I'll check it out tomorrow :)

AlCalzone, 2022-08-31:
So, I'm not sure this device is meant to report the status of all paddles to the controller. It does not support multi channel, so by definition it only appears as a single switch in the Z-Wave network.
From what I could gather from [the manual](https://opensmarthouse.org/zwavedatabase/605/reference/1124375569369.pdf) and [this thread](https://community.smartthings.com/t/tkbhome-tz36d-second-paddle-support/159975/5), the right paddle only controls other associated devices.

For actual multichannel support, you seem to need the TZ37 or TZ57 variants.

GAguado, 2022-08-31:
This popular wall switch TZ56D like the former TZ66D controls only one load
(left pad) and the right pad acts as a button/sensor with entities: right
on, right off, double click right on, and double click right off)
I have many of them working for years with a Zipato controller. The TZ37 or
TZ57 indeed control two loads but the TZ56D and TZ66 control only one.
As recognized as TZ36 HA doesn't show the entities related to the right
pad. I guess the fault is the HA model discovered as it discovers wrongly
TZ36 when it is a TZ56D,.. did you double check the list of models of this
maker? this should be like:
[image: image.png]
Best Regards
Gerardo

GAguado, 2022-08-31:
My fault, screen dump is not showed,.. TZ66D maker 280 product type ID 258 product ID 4128
TZ56D maker 280 product type ID 785  product ID 514 (I don't know whether the Id's are from the maker or Zipato guys, just in case)

AlCalzone, 2022-08-31:
Functionality is not bound to specific models in Z-Wave, it is queried from the device via standardized commands.

Different channels/switches are normally distinguished through the Multi Channel CC, which the switch  you're using doesn't support.

If it was actually supported in Zipato, I can only assume they "hacked" support by adding the controller to the association group of the right paddle and distinguishing the commands it sends somehow. You'll have to try that out though since there's nothing in the manual that gives a hint what that paddle sends.

GAguado, 2022-08-31:
Thanks for your comments but something doesn't match, this wall switch is widely available and fully functional (right pad) with different Zwave controllers. On top I don't understand why this TZ56D is recognized by HA as TZ36D.
Interview shows the ID's I have in Zipato, same:
2022-08-30T16:54:06.051Z CNTRLR   [Node 008] [+] [Manufacturer Specific] manufacturerId: **280**        [Endpoint 0]
2022-08-30T16:54:06.053Z CNTRLR   [Node 008] [Manufacturer Specific] productType: metadata updated  [Endpoint 0]
2022-08-30T16:54:06.058Z CNTRLR   [Node 008] [+] [Manufacturer Specific] productType: **785**           [Endpoint 0]
2022-08-30T16:54:06.061Z CNTRLR   [Node 008] [Manufacturer Specific] productId: metadata updated    [Endpoint 0]
2022-08-30T16:54:06.064Z CNTRLR   [Node 008] [+] [Manufacturer Specific] productId: **514**             [Endpoint 0]
But this is not a TZ36, from the logic point of view product DB is mixed up and therefore driver could be wrong, don't know

GAguado, 2022-08-31:
From eedomus: https://doc.eedomus.com/files/TKB%20New%20wall%20switch%20manual%20(TZ56S,%20TZ56D,%20TZ55S,%20TZ55D%20&%20more).pdf

AlCalzone, 2022-09-01:
> On top I don't understand why this TZ56D is recognized by HA as TZ36D.

The IDs might be incorrect, I give you that. All this changes though is the label and the predefined configuration parameters.

> According this document Multi Channel is supported

Where does it say that?

GAguado, 2022-09-01:
You are right, 'Multi-channel' doesn't not appear in the document I sent you yesterday but I saw it in another which I can not find at this point, sorry for that.
On TKB series document linked yesterday you can read:
Control other Z-WAVE enable devices -> (two “associated” group of 5 each-Groups 1 -> this is something that I readed in another document but I don't understand the technical meaning)
On page 4 it expains how group 2 and group 3 can handle the right pad (single or double click) via associations
I don't know how I can help further, in theory TKB is member of Z-Wave Alliance, more complete documentation should be available somewhere

GAguado, 2022-09-01:
Sorry, my above incomplete sentence shoud be: 
The right switch on the switch allows the user to : Control other Z-WAVE enable devices

GAguado, 2022-09-01:
I had some more time to dig into this and you are right, right pad could be used only it it could be associated to a virtual device and trigger from there its state change. Zipato guys did it, I don't know if with virtual devices or not. 
I'm affraid I will have to be sticked to Zipato unless I change all my TKB switches, I have not the skills to hack anything, any posibility from your side ?
Best Regards
Gerardo

GAguado, 2022-09-01:
What if,.. as Zwave JS does not support device to device associations yet (as far as I know), it could be great to have the possibility to send a command to any ZWave device with the slave zwave device ID on the group X.
This could be a swiss knife in many scenarios,.. what do you think?  this device could be virtual also,..
Gerardo

AlCalzone, 2022-09-01:
> it could be great to have the possibility to send a command to any ZWave device with the slave zwave device ID on the group X.

Something like this does not exist in Z-Wave.

> Zwave JS does not support device to device associations yet

It does. Home Assistant does not have UI for this though, you'll need to temporarily use zwavejs2mqtt, as described [here](https://www.home-assistant.io/integrations/zwave_js#can-i-switch-between-the-official-z-wave-js-add-on-and-zwavejs2mqtt).

> I have not the skills to hack anything

What I meant is:
- add the controller (node ID 1) to groups 2 and 3
- use the right paddle
- check which values change when you dim and toggle and if they are distinguishable from the left paddle

If you make a [driver log](https://zwave-js.github.io/zwavejs2mqtt/#/troubleshooting/generating-logs?id=driver-logs) (level debug) after you added the controller, I can help with that. I just need clear comments like: _at time XX:YY:ZZ I toggled/dimmed the left/right paddle._

AlCalzone, 2022-09-01:
> (you) can say to the master device the ID of the slave you want and this could be a virtual

if the device does not physically exist, this will cause problems and delays, so I'd strongly advise against that.

> how can add the controller node ID1 to groups 2 & 3?

[image]

[image]

Then click Add

GAguado, 2022-09-01:
Thanks a lot, I will try it, first I have to install the mqtt version. 
Late but I found the document in which it is stated that this device support multi:  (ZWave alliance) https://www.dropbox.com/s/4wdc9lt1st7zrmx/ZC08-15070007.pdf?dl=0
Last line of the document

AlCalzone, 2022-09-01:
Switch Multi**level** is just the dimmer functionality.

Distinguishing multiple switches needs Multi**channel** support.

GAguado, 2022-09-08:
Hi !
Sorry for the delay, I couldn't before,..  I installed JS2MQTT and tried to associate wall switch nodes as you suggested.

New Association
**Node Endpoint**
Endpoint 0 (only appears EndPoint 0)

**Group**
OnoffDimmer1 (group2)
  
But I can not leave empty the field Target Node Endpoint (!)  it says 'This field is required'

What am I doing wrong?

Thanks
Gerardo

GAguado, 2022-09-08:
[image]

AlCalzone, 2022-09-08:
> But I can not leave empty the field Target Node Endpoint (!) it says 'This field is required'

You're mixing things up:
Target **Node** needs to be your controller (I guess "Razberry"?)
Target **Endpoint** should stay blank (if it even exists)

GAguado, 2022-09-11:
Hi !  please find the log files, two wall switches: TKB TZ56D and TKB TZ66D (former version of previous with the same functionality)
[Debug TKB TZ56-D (FW2_5).txt](https://github.com/zwave-js/node-zwave-js/files/9542023/Debug.TKB.TZ56-D.FW2_5.txt)
[Debug TKB TZ66-D (FW 1_3).txt](https://github.com/zwave-js/node-zwave-js/files/9542024/Debug.TKB.TZ66-D.FW.1_3.txt)

AlCalzone, 2022-09-12:
Okay, so the left switch's status is found under `Binary Switch`, the right one is using `Basic CC` for single/double taps and `Multilevel Switch CC` for holding.
Holding should be already appearing in Home Assistant as events.
For toggling/tapping I have some bad news: By default, Home Assistant filters Basic CC values, so you won't have entities for them. Also, single/double tap are indistinguishable since they both use the same command.

@martinhjelmare @raman325 Would it be possible to expose Basic CC values in HA? I see devices like this from time to time where part of the functionality is only accessible via Basic CC. Granted, this would have to be rather generic entity without much context.

raman325, 2022-09-12:
We create an entity for the Basic CC, it's just disabled by default. If the user goes to the device page for their device and clicks on show disabled entities, they will see it and can enable it from there.

GAguado, 2022-10-17:
Dear Team,

I guessed you were modifying the code but as time goes I reopen this issue, under disabled entities there is not any entity yet either for double tap or holding tap, please check

Thanks in advance

AlCalzone, 2022-10-17:
I'll need someone with more HA experience to debug this

GAguado, 2022-10-17:
I have not any problem resending all log files (you have them in previous posts) but entities are quite basic ones, look:
[image]

kpine, 2022-10-17:
I asked for the HA device diagnostic file, not a log file.

GAguado, 2022-10-17:
Hi, please find an updated one:

[zwave_js-6f5eef5ac0e4687d5e017d9ccb63e1c3-Dual Paddle Wall Switch-c934f84705b73e3482b3e930c2312da7.json (6).txt](https://github.com/zwave-js/node-zwave-js/files/9803445/zwave_js-6f5eef5ac0e4687d5e017d9ccb63e1c3-Dual.Paddle.Wall.Switch-c934f84705b73e3482b3e930c2312da7.json.6.txt)

kpine, 2022-10-17:
Based on the dump, the driver is not exposing any value IDs for Basic CC, so HA cannot create any entities. HA only creates entities for values it is told about.

@AlCalzone Many devices that report button presses with BasicCC::Set use events instead of values. Is configuring this device to use events not an option?

GAguado, 2022-10-21:
Who is the question addressed to? I have not much infomation about device internals except the user manual in which they (TKB) describes the use of groups:

https://doc.eedomus.com/files/TKB%20New%20wall%20switch%20manual%20(TZ56S,%20TZ56D,%20TZ55S,%20TZ55D%20&%20more).pdf

GAguado, 2022-10-24:
If is me saying that I didn't see any event in HA for this popular TKB wall switch related with either right pad click or double right click

GAguado, 2022-12-02:
Dear Team, any news on this?  I have a similar issue with the Fibaro Button FGPB-101, J-Wave JS only triggers one click when this button has double and triple clicks
Best Regards
Gerardo

69leo, 2023-01-31:
Hello, I am also interested in solving this problem. I have several TZ66D switches. Previously used Z-wave.me and single and double right-click supported the management of associations with other devices. On HA, the right key does not work completely. What is needed to fix this?

69leo, 2023-01-31:
> Судя по дампу, драйвер не предоставляет никаких идентификаторов значений для Basic CC, поэтому HA не может создавать какие-либо сущности. HA создает сущности только для значений, о которых ему говорят.
> 
> @AlCalzoneМногие устройства, которые сообщают о нажатиях кнопок с помощью BasicCC::Set, используют события вместо значений. Настройка этого устройства для использования событий не вариант?

Hello, can you explain in detail how to set this up?

AlCalzone, 2023-01-31:
Sorry for the delay. The linked PR should make it so that the right paddle shows up as Basic Set events in HA. Like I said previously, I don't see a way to distinguish single and double tap though.

closed 2023-01-31 by AlCalzone