zwave-js/zwave-js #4864: Zwave Device not allowing associations

rgerrans, 2022-07-29:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within ZWaveJS2MQTT?

NO, my problem is NOT within ZWaveJS2MQTT

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?

What do you observe?
After adding a Nortek Security & Control LLC | WD500Z-1 | FW: 5.41 to Node-Red, I'm unable to create associations. Single tap actions do not report to the controller but double tap actions do

What did you expect to happen?

Steps to reproduce the behavior:
  1. Go to '...'
  2. Click on '...'
  3. Scroll down to '...'
  4. See error

### Device information

Manufacturer:  Nortek Security & Control LLC 
Model name: WD500Z-1
Node ID in your network:  20

### How are you using `node-zwave-js`?

- [ ] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [ ] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version: 9.6.1
`node-zwave-js` branch:
`zwavejs2mqtt` branch:

### Did you change anything?

no

### Did this work before?

Don't know, this is a new device

### Attach Driver Logfile

[zwave-js.log](https://github.com/zwave-js/node-zwave-js/files/9223693/zwave-js.log)

marcus-j-davies, 2022-07-30:
@AlCalzone,
Just some more context.

The driver reported that Associations are not supported when trying to add them, this seem evident in the discovered CCs.

```
2022-07-29T22:27:02.928Z CNTRLR   finished adding node 20:
                                    basic device class:    Routing Slave
                                    generic device class:  Multilevel Switch
                                    specific device class: Multilevel Scene Switch
                                    supported CCs: 
                                    · Basic (0x20)
                                    · Multilevel Switch (0x26)
                                    · All Switch (0x27)
                                    · Scene Activation (0x2b)
                                    · Scene Actuator Configuration (0x2c)
                                    · Manufacturer Specific (0x72)
                                    · Powerlevel (0x73)
                                    · Configuration (0x70)
                                    · Version (0x86)
                                    controlled CCs: 
2022-07-29T22:27:02.929Z CNTRLR » [Node 020] Assigning SUC return route...
2022-07-29T22:27:02.938Z SERIAL » 0x01050051140fb0            
```                       

However, the configuration file and the devices manual suggests otherwise.
https://devices.zwave-js.io/?jumpTo=0x014f:0x4457:0x3034:0.0
https://www.nortekcontrol.com/wp-content/uploads/WD500Z1_manual.pdf

So I assume it's reported by the device incorrectly, I enforced support for Association with the compatibility flags.

Association and Association Group Info.
```
 "compat": {
        "commandClasses": {
            "add": {
                "0x85": {
                    "isSupported": true,
                    "secure": false
                },
                "0x59": {
                    "isSupported": true,
                    "secure": false
                }
            }
        }
    }
```
But Adding associations... 

[image]

AlCalzone, 2022-08-01:
Did you re-interview after adding the compat flag? The number of groups needs to be picked up first.

marcus-j-davies, 2022-08-01:
Yup, we were able to get passed the not supported error (so added the flags - both of them just in case), hence why I was confused to see the error about the group not existing.

AlCalzone, 2022-08-01:
Groups are queried during the interview, but at that point zwave-js still thought the Association CC was not supported.

rgerrans, 2022-08-01:
@AlCalzone / @marcus-j-davies Here is an updated log file of reinterviewing the node with the patch installed

Let me know what else I can do to help (though I will be out of pocket mid-day tomorrow until next Monday)

[zwave-js.log](https://github.com/zwave-js/node-zwave-js/files/9237840/zwave-js.log)

AlCalzone, 2022-08-02:
I'm afraid this is a case for the manufacturer support. During the interview, the Association CC queries now time out with no response. So either the device does actually not support them, or it is in a weird state and needs to be factory reset (I've seen similar things happen before).

marcus-j-davies, 2022-08-02:
@rgerrans  our patch forcing a query on association groups
```
022-08-01T22:57:25.305Z CNTRLR    [Node 020] Interviewing Association...
2022-08-01T22:57:25.305Z CNTRLR » [Node 020] querying number of association groups...
```
Waiting for the response....... We don't get one.

```
2022-08-01T22:57:26.564Z CNTRLR   [Node 020] Timed out while waiting for a response from the node (ZW0201)
2022-08-01T22:57:26.564Z CNTRLR   [Node 020] Querying association groups timed out, skipping interview...
```

just to reiterate on @AlCalzone, maybe exclude the device,  factory reset the device, then include it again (with our patch)
other than that, despite the manual saying so and our imported config from other sources, the device is not giving as what is to be expected.

Unless you happen to be on an old firmware, that didn't have association support, its a very odd result, the patches are usually for devices that don't advertise support (when in fact they do support such a CC). In this case, it really looks as if it doesn't.

rgerrans, 2022-08-02:
@AlCalzone Thanks. This is an old unit that could be failing as well. Still weird that it sends through double taps but not single, BUT I had not tried a full factory reset just an exclusion/inclusion. I'll give that a try when I'm back home and see if it makes it work. I'll leave this open until I get a chance to do that last test.

Thanks for the support.