zwave-js/zwave-js #4872: Interviewing never finishes for Jasco 46203 dimmer switches

JoshSnider, 2022-08-01:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within ZWaveJS2MQTT?

NO, my problem is NOT within ZWaveJS2MQTT

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

**What causes the bug?**
Interviewing nodes in zwavejs2mqtt

**What do you observe?**
Interview process does not move beyond Node Info.

**What did you expect to happen?**
Interview process should complete and device configuration should be correct and device should become "ready" in home assistant.

**Steps to reproduce the behavior:**
  1. In zwavejs2mqtt, actions -> advanced -> re-interview nodes
  2. See interview status update from protocol info to node info with endless spinner

### Device information

**Manufacturer:** AEON Labs
**Model name:** Z‐Stick Gen5 USB Controller
**Node ID in your network:** 1

**Manufacturer:** Jasco
**Model name:** Enbrighten Z-Wave Plus Smart Dimmer
**Node ID in your network:** 16, 31, 32

### How are you using `node-zwave-js`?

- [X] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [ ] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

**`zwavejs2mqtt` version:** 6.14.1
**commit:** https://github.com/zwave-js/zwavejs2mqtt/commit/2b9e51d1db4a50aa38472b1adc4819274d59af64

### Did you change anything?

no

### Did this work before?

Yes (please describe)

### If yes, where did it work?

Previously I had a single Jasco 46203 device (light switch) paired without security and integrated into homeassistant and it worked.

### Attach Driver Logfile

[zwavejs_2022-07-31.log](https://github.com/zwave-js/node-zwave-js/files/9230459/zwavejs_2022-07-31.log)

AlCalzone, 2022-08-01:
Any chance you also have the zwavejs2mqtt logs for this?
It could be that there is an uncaught error here.

JoshSnider, 2022-08-06:
[zwavejs2mqtt.log](https://github.com/zwave-js/node-zwave-js/files/9275869/zwavejs2mqtt.log)

Sorry for the delay, here's both.

JoshSnider, 2022-08-06:
Also my zwave stick is on USB extension, ~4 feet away from the server

AlCalzone, 2022-08-08:
Ok I can't see whats wrong here. I'll ask Jasco for a sample and debug myself.

AlCalzone, 2022-08-10:
So I got two copies of the device from Jasco. Both added no problem and re-interview finished within a couple of seconds. I tested zwavejs2mqtt `master` branch, the development `zjs-v10` branch and `v6.15.2`.

So I'm really not sure whats going on here.

closed 2022-08-23 by AlCalzone