zwave-js/zwave-js #4876: 3-way add on switches not reporting updates

scuffe82, 2022-08-02:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within ZWaveJS2MQTT?

NO, my problem is NOT within ZWaveJS2MQTT

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?

When using any of my addon (3way) switched the server doesn't receive any data. If i use the main switch the 

What do you observe?

I first noticed the behavior in HA which mimicked a polling issue with my older GE/Jasco 12724 dimmer switches, but to troubleshoot i moved to just working through the zwavejs2mqtt container. I'll attach the full logs below but here are the highlights. I deleted the log files between each test to try and cut down on extra noise.

When i turn on and then off a 3 way light from the primary switch i see:
```
<Snip>
2022-08-02T22:59:09.874Z CNTRLR   [Node 002] Embedded device config loaded
<...>
2022-08-02T22:59:09.918Z CNTRLR » [Node 002] pinging the node...
<...>
2022-08-02T22:59:09.966Z SERIAL » 0x010800130201002501c3                                              (10 bytes)
2022-08-02T22:59:09.966Z DRIVER » [Node 002] [REQ] [SendData]
                                  │ transmit options: 0x25
                                  │ callback id:      1
                                  └─[NoOperationCC]
2022-08-02T22:59:09.977Z SERIAL « [ACK]                                                                   (0x06)
2022-08-02T22:59:09.977Z SERIAL « 0x0104011301e8                                                       (6 bytes)
2022-08-02T22:59:09.978Z SERIAL » [ACK]                                                                   (0x06)
2022-08-02T22:59:09.978Z DRIVER « [RES] [SendData]
                                    was sent: true
2022-08-02T22:59:09.996Z SERIAL « 0x0107001301000003e9                                                 (9 bytes)
2022-08-02T22:59:09.997Z SERIAL » [ACK]                                                                   (0x06)
2022-08-02T22:59:09.997Z DRIVER « [REQ] [SendData]
                                    callback id:     1
                                    transmit status: OK
2022-08-02T22:59:09.999Z CNTRLR   [Node 002] The node is alive.
2022-08-02T22:59:10.007Z CNTRLR   [Node 002] The node is ready to be used
2022-08-02T22:59:10.008Z CNTRLR « [Node 002] ping successful
<...>
022-08-02T22:59:49.421Z DRIVER « [Node 002] [REQ] [ApplicationUpdateRequest]
                                    payload: 0x020a04110126277370867277
2022-08-02T22:59:49.426Z CNTRLR « [Node 002] Received updated node info
2022-08-02T22:59:49.428Z CNTRLR   [Node 002] Node does not send unsolicited updates; refreshing actuator and sen
                                  sor values...
2022-08-02T22:59:49.431Z CNTRLR » [Node 002] requesting current switch state...
2022-08-02T22:59:49.451Z SERIAL » 0x0109001302022602250de9                                            (11 bytes)
2022-08-02T22:59:49.452Z DRIVER » [Node 002] [REQ] [SendData]
                                  │ transmit options: 0x25
                                  │ callback id:      13
                                  └─[MultilevelSwitchCCGet]
2022-08-02T22:59:49.452Z SERIAL « [ACK]                                                                   (0x06)
2022-08-02T22:59:49.459Z SERIAL « 0x0104011301e8                                                       (6 bytes)
2022-08-02T22:59:49.459Z SERIAL » [ACK]                                                                   (0x06)
2022-08-02T22:59:49.459Z DRIVER « [RES] [SendData]
                                    was sent: true
2022-08-02T22:59:49.482Z SERIAL « 0x010700130d000003e5                                                 (9 bytes)
2022-08-02T22:59:49.483Z SERIAL » [ACK]                                                                   (0x06)
2022-08-02T22:59:49.483Z DRIVER « [REQ] [SendData]
                                    callback id:     13
[... 107 lines cut ...]
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version:
`node-zwave-js` branch:
`zwavejs2mqtt` branch:

### Did you change anything?

yes (please describe)

### If yes, what did you change?

This is a fresh setup, The system was not on for a couple years and was just installed fresh and setup.

### Did this work before?

Don't know, this is a new device

### Attach Driver Logfile

[zwavejs_2022-08-02-main-switch.log](https://github.com/zwave-js/node-zwave-js/files/9246783/zwavejs_2022-08-02-main-switch.log)
[zwavejs_2022-08-02-addon-switch.log](https://github.com/zwave-js/node-zwave-js/files/9246784/zwavejs_2022-08-02-addon-switch.log)

scuffe82, 2022-08-03:
Just updated to 1.2 on the z-wave stick no change in behavior.

FW: v1.2
SDK: v6.7.0

AlCalzone, 2022-08-03:
> I fired up an old version of openzwave

Did you just put the stick into OZW and tested or did you do something else too?

> and when preforming the same test both sets of logs reported the same information

Can you check the associations on both devices in OZW and compare them to Z-Wave JS?

I'm not sure what this addon switch is and how it is meant to be operated. Did you get instant feedback in OZW or did it poll the device regularly for its status?
Z-Wave JS doesn't do any regular polling on its own.

AlCalzone, 2022-08-04:
Unfortunately neither debug info includes the associations. You can just share a screenshot of the association UI of both softwares.

And the OZW log excerpt seems to be from the interview, not the operation.

AlCalzone, 2022-08-04:
Lets do the following:
Set up Z-Wave JS debug logging (https://zwave-js.github.io/zwavejs2mqtt/#/troubleshooting/generating-logs?id=driver-logs), but delete old logs.

Switch to OZW, make sure you get the info from the addon switch. Capture a log where you operate the switch and the addon switch - please don't truncate it, just give me the whole thing.

Then switch back to Z-Wave JS, let it interview the switch (or re-interview if it was previously interviewed). Wait until that is done, then operate the switch and the addon switch again. Upload the log of this, again please don't truncate it.

scuffe82, 2022-08-13:
Sorry for the long reply, it was a crazy week. Here are some screenshots and the logs files. Please let me know how else i can help you with this. Thanks
[image]
[image]
[image]
[image]
[image]
[image]
[OpenZWave Control Panel-full.log](https://github.com/zwave-js/node-zwave-js/files/9330710/OpenZWave.Control.Panel-full.log)
[zwavejs_2022-08-13 3.log](https://github.com/zwave-js/node-zwave-js/files/9330714/zwavejs_2022-08-13.3.log)

!

scuffe82, 2022-08-15:
Ok, so OZW is polling by default which is causing the state to get picked up? 

Yes the addon switch doesn't have a zwave radio in it. Thats the one i was talking about just sends a signal to the main switch to do something so it would be up to the main switch to send/report/get polled/etc. 

If you would like a longer log set i can run the test again and leave each app running for like 10 minutes then grab the logs, but if i'm reading your reply correctly the only way that i can make this work would be to either enable polling in zjs or upgrade the physical switches (at almost $50 each won't be happening until they break)

Thanks!

AlCalzone, 2022-08-15:
> If you would like a longer log set i can run the test again and leave each app running for like 10 minutes then grab the logs

Yes please. Probably like this would be best:
- Let the interview finish
- Toggle the main switch
- wait 2 minutes
- Toggle the addon switch
- wait 2 minutes

closed 2022-08-26 by zwave-js-assistant[bot]