zwave-js/zwave-js #4877: Missing device configuration: Ultraloq U-Bolt Z-Wave Smart Lock

ticknert, 2022-08-04:
### Checklist

- [X] It is not in the [configuration DB](https://devices.zwave-js.io/?search=3)

- [X] It was not [merged recently](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Apr+is%3Aclosed) or has a [pending PR](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Aopen+is%3Apr)

### Which device is missing?

Ultraloq U-Bolt Z-Wave Smart Lock

### Manufacturer ID

I don't know

### Product Type

0x0004

### Product ID

0x0001

### Is the device listed on the Z-Wave Alliance website?

https://products.z-wavealliance.org/products/4179

ticknert, 2022-08-04:
Per this document, the Z-Wave SDK version is 7.14.3 https://products.z-wavealliance.org/products/4179/embedpics

ticknert, 2022-08-04:
Also, maybe I have this successfully set up now using the QR code provided with the product. The manufacturer ID is 1106

AlCalzone, 2022-08-04:
We do have this: https://devices.zwave-js.io/?jumpTo=0x0452:0x0004:0x0001:0.0#

closed 2022-08-04 by AlCalzone

ticknert, 2022-08-04:
You're right, thanks! I didn't realize my lock used the same config as the Pro