zwave-js/zwave-js #4879: Ring Keypad v2 (4AK1SZ) seems to lose connection often after being armed

Caros2017, 2022-08-04:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within ZWaveJS2MQTT?

NO, my problem is NOT within ZWaveJS2MQTT

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

**What causes the bug?**
When the keypad is in the state armed for a while and the alarm is triggered, the keypad is losing the connection to the zwave controller very often. The connection light is flashing red and the entry delay is not showing on the panel. It seems that multiple users have this issue [link](https://community.home-assistant.io/t/using-the-ring-alarm-keypad-with-home-assistant/343367/141).

**What do you observe?**
The red networklight is showing red, which indicates that there is a network connection.

**What did you expect to happen?**
The keypad showing an entrance delay and the option to put in my pincode, instead of a flashing red light.

**Steps to reproduce the behavior:**
  1. Arm the keypad
  2. Wait a couple of minutes
  3. Trigger the entry delay
  4. Most of the times the issue occurs now
  

**Additional info:**
There are multiple users experiencing the same problem.
2 times with the Aeotec Gen5+ dongle software version FW: v1.2 / SDK: v6.7.0
1 time with the zozo Gen5 stick

Most users (including myself) report that the network is solid and that is no issue. For example, my device is within a couple of meters and if I test it, I get 10/10 all the time. It's directly connected to the gen5+ stick and the motion sensor works perfectly. 

I have no problems with any other device on my network. Also made the network more quiet, but it doesn't make any difference. One user reports that swapping the device with the vendor doesn't make any difference.

I don't know myself how to troubleshoot this.

### Device information

Manufacturer: Ring
Model name:  Keypad V2
Node ID in your network: 23
Software version: V1.18

### How are you using `node-zwave-js`?

- [X] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [ ] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version:
`zwavejs2mqtt` branch: latest (6.15.1)

### Did you change anything?

no

### Did this work before?

Don't know, this is a new device

### Attach Driver Logfile

It's an S2-encrypted device

marcus-j-davies, 2022-08-04:
Hi @Caros2017 

> It's an S2-encrypted device

Makes no difference, a log file will still indicate what is going on without needing to decrypt traffic.
before anyone can look into what is going on, please could you upload a ZWave JS log file , at a minimum at Debug level.

Thanks

AlCalzone, 2022-08-04:
What he said

Caros2017, 2022-08-04:
@marcus-j-davies @AlCalzone My bad, I misinterpreted the text for making a regular issue. I have just added my logs and ofcourse I changed my pin, since it's there in plain text I see...

AlCalzone, 2022-08-04:
> I changed my pin, since it's there in plain text I see...

Uhhh we should change that.

Caros2017, 2022-08-04:
> > I changed my pin, since it's there in plain text I see...
> 
> Uhhh we should change that.

You can see it in this line, the last one (changed it to random 0110):
```
2022-08-03T11:17:21.100Z DRIVER « [Node 023] [REQ] [ApplicationCommand]
                                  └─[Security2CCMessageEncapsulation]
                                    │ sequence number: 42
                                    └─[SupervisionCCGet]
                                      │ session id:      43
                                      │ request updates: false
                                      └─[EntryControlCCNotification]
                                          sequence number: 106
                                          data type:       2
                                          event type:      2
                                          event data:      0110
```

AlCalzone, 2022-08-04:
So, I found this gist linked in the thread: https://github.com/ImSorryButWho/HomeAssistantNotes/blob/6f05b3f90a979f7118e7c688ce2d01b878227328/RingKeypadV2.md

There seems to be some mixup between hexadecimal and decimal in the automations. The gist correctly mentions that property 10 (hex `0x0a`) is armed stay, but the automation sends 16 (hex `0x10`). Not sure if the automation converts strings to hex somehow?

Also
> For most messages, I find it makes sense to use a property_key of 1 and value of 1, just indicating that the indicator should be turned on

This should probably use a property_key of 2 (binary/boolean value) instead.

AlCalzone, 2022-08-04:
pinging @ImSorryButWho as the blueprint author

AlCalzone, 2022-08-04:
> Can you take a look at this log as well?

This is a collision in S2 state. Roughly at the same time, both Z-Wave JS and the keypad try to send a message and both advance the shared encryption state. This results in both not being able to decrypt the other's message and makes them try to re-establish a shared state.
I'm guessing that the message which could not be decoded was supervised, so the keypad didn't get its expected response.

I have a few ideas on how to improve these situations, but I needed the better testing framework I wrote for the upcoming v10 to even begin simulating this.

Caros2017, 2022-08-04:
> This is a collision in S2 state. Roughly at the same time, both Z-Wave JS and the keypad try to send a message and both advance the shared encryption state. This results in both not being able to [...]

But isn't that the issue we are facing then?

And can you tell me what the `SERIAL « [CAN]` means? This seems every time related to my keypad.

Please ignore my ignorance

ImSorryButWho, 2022-08-04:
> pinging @ImSorryButWho as the blueprint author, see [#4879 (comment)](https://github.com/zwave-js/node-zwave-js/issues/4879#issuecomment-1205365221)

Thanks, I'll freely admit to not actually understanding what the hell I'm doing -- the development process was just to play with things until it worked.  I'll see about re-working it to be actually standards compliant.

Caros2017, 2022-08-04:
> Thanks, I'll freely admit to not actually understanding what the hell I'm doing -- the development process was just to play with things until it worked. I'll see about re-working it to be actually [...]

Do you experience the same issue as described in this issue?

ImSorryButWho, 2022-08-04:
> 

I haven't noticed anything like the behavior being described here, but I can certainly believe there's some environmental difference.  My guess is that the keypad by the door I usually use has the old firmware version without motion sensor reports, so there's not going to be a race between a motion sensor report and the "Entry Delay Started" indicator.  Assuming that's the issue, maybe just add a small delay before sending the indicator?

mundschenk-at, 2022-08-04:
> I haven't noticed anything like the behavior being described here, but I can certainly believe there's some environmental difference. My guess is that the keypad by the door I usually use has the [...]

If I understand @AlCalzone correctly, there are three separate issues: 
1. Some parameters in the blueprint are invalid (100 instead of 99 for multilevel indicators - not sure if that is volume or brightness in this case).
2. The wrong property key is sent in at least one case apparently (maybe a hex/dec mixup).
3. In the logfile, an S2 collision has happened. This is very likely contingent and not something that is directly caused by the automation (a delay somewhere might make it less likely, but I don't understand the situation well enough to judge that).

1. and 2. need to be fixed in the blueprint/automation, 3. is something that probably more related to how the driver handles S2 communication.

mundschenk-at, 2022-08-04:
> I see the following sequences (omitting duplicates):
> 
> * Keypad -> Z-Wave JS: Entry Control (Enter)
>   Z-Wave JS -> Keypad: Indicator (Not armed, multilevel, 100).
> * Keypad -> Z-Wave JS: Entry Control (ArmAway)
>   Z-Wave JS -> Keypad: Indicator (Bypass challenge, multilevel, 100).
> * Keypad -> Z-Wave JS: Entry Control (Enter)
>   Z-Wave JS -> Keypad: Indicator (Exit Delay, 180s)
> * Keypad -> Z-Wave JS: Entry Control (ArmHome)
>   Z-Wave JS -> Keypad: Indicator (Bypass challenge, multilevel, 100).
> 
> Note that 100 is not a valid value for multilevel things in Z-Wave, the maximum is 99 or 255 for last non-zero value. This could be one source of the issue - that the keypad does not receive an expected command.

I don't think that this sequence is complete, @AlCalzone. What @Caros2017 was asking about is around the time the Entry Delay indicator is set after the door opens (around timestamp `2022-08-03T11:01:59.951Z`). 

> There seems to be some mixup between hexadecimal and decimal in the automations. The gist correctly mentions that property 10 (hex `0x0a`) is armed stay, but the automation sends 16 (hex `0x10`). Not sure if the automation converts strings to hex somehow? The property key sent to zwave-js should be the number 10.

Where do you see the hex/dec mixup in the script (or rather: in the log)? Timestamp `2022-08-03T10:37:35.222Z`

The value ID is [constructed like this](https://github.com/home-assistant-libs/zwave-js-server-python/blob/a73ace109d953056cc8f3c6e509e8a2b97808b29/zwave_js_server/model/value.py#L87) in the server lib:

```python
value_id = f"{node.node_id}-{command_class}-{endpoint_}-{property_}"
``` 

It does not look like there is any possible assumption of a hex string anywhere in the code path to me. It seems more likely that there is some race condition or other error in the implicit state machine of the automation and it is accidentally firing `need_bypass` at an inopportune moment?

ImSorryButWho, 2022-08-04:
Right.  I'll fix the 100s to be something more appropriate, but I don't think they're the actual source of this problem.  They work fine on my system.

My interpretation of https://github.com/zwave-js/node-zwave-js/issues/4879#issuecomment-1205376277 is that the keypad and controller are both trying send each other messages (nearly) simultaneously, which causes the S2 state to desynchronize.   The expected flow of events for the case where the alarm is armed and a door opens, which is what we're looking at is:

  - Door Opens
  - Alarm transitions from Armed to Pending
  - Controller sends "Entry Delay Started" indicator to keypad
  - User enters code
  - Alarm disarms
  - Controller sends "Disarmed" indicator to keypad

What I think is happening is the motion sensor in the keypad is seeing the door opening, which causes the keypad to send a motion event to the controller at the same time that the controller is trying to send the "Entry Delay Started" indicator to the keypad.  That explains why I don't have the same problem (my primary keypad is on an older firmware version that doesn't send motion events), and why it's only happening for some people: those whose keypads are in a spot where they can see motion at the same time the alarm is triggered.

A couple of ways to test this:  add a delay before the  entry delay indicator is set (at line 201 of https://github.com/ImSorryButWho/HomeAssistantNotes/blob/main/alarm_keypad_automations.yaml).  Or, try covering the motion sensor with electrical tape.

ImSorryButWho, 2022-08-04:
And, yes, there's no hex/decimal confusion here.  Looking at that log, the user must have tried to arm the system with an open sensor, so we sent a "sensors require bypass" message as expected.  I'm completely confident that the logic there is correct (see the sections at lines 132 and 172 in the blueprint linked above).

mundschenk-at, 2022-08-04:
> What I think is happening is the motion sensor in the keypad is seeing the door opening, which causes the keypad to send a motion event to the controller at the same time that the controller is trying to send the "Entry Delay Started" indicator to the keypad.

OK, yeah, that would describe my placement situation pretty well. I'll try get a clean logfile of the issue and then try again with the delay.

davidcoulson, 2022-08-04:
> Right. I'll fix the 100s to be something more appropriate, but I don't think they're the actual source of this problem. They work fine on my system.

For what it is worth, I have connectivity issues with the Keypad, but I am using my own implementation in Node-Red, not the HA blueprint.

Caros2017, 2022-08-05:
> 
> For what it is worth, I have connectivity issues with the Keypad, but I am using my own implementation in Node-Red, not the HA blueprint.

But do you experience the same issue? Is it possible that in your situation the motion sensor as well as the "entry delay started" triggers at the same time?

> OK, yeah, that would describe my placement situation pretty well. I'll try get a clean logfile of the issue and then try again with the delay.

I am trying a different solution. I just made a flow that disables the motion sensor when the alarm is armed and enables the motion sensor when it is armed. I never prever delays.

AlCalzone, 2022-08-05:
> the user must have tried to arm the system with an open sensor, so we sent a "sensors require bypass" message as expected

Okay, I didn't look into the automation that in-depth. I just assumed from the OP that the alarm was supposed to be armed and not bypassed.

Maybe this is all just caused by the S2 collisions. Can anyone else confirm that [this](https://github.com/zwave-js/node-zwave-js/issues/4879#issuecomment-1205367143) happens for them too?

davidcoulson, 2022-08-05:
> I am trying a different solution. I just made a flow that disables the motion sensor when the alarm is armed and enables the motion sensor when it is armed. I never prever delays.

Are you just disabling the Proximity Detection config setting (126-112-0-15) or something else? I just turned that off permanently via Z-Wave JS so will see if I can still reproduce the problem.

marcus-j-davies, 2022-08-05:
@AlCalzone,

Watching this issue for a while, I have just witnessed the same weirdness, however, I use the Gen 1.
This weirdness is new (but I am rolling V10 Beta 4)

Pin: 12345

**Before** 

 - Start entering my pin (to disarm) -> `Caching` entry control event (so far so good)
 - Press Disarm -> `Disarm all` (with my code)
   - I check the pin entered
      - If valid, set indicator to Disarmed
        - Indicator set (Spoken confirmation)
      
**What I now witness.**

 - Start entering my pin (to disarm) -> `Caching` entry control event (so far so good)
 - Press Disarm -> Nothing happens
   - After a few seconds 
      - The network issue LED is lit up
          - a few more seconds
             - `Disarm all` (with my code)
               - If valid, set indicator to Disarmed
                  - Indicator set (Spoken confirmation)

This was never the case, so guessing something has been skewed with the handling, as this previously was flawless.

Log attached 

Interaction begins from:
`2022-08-05T18:19:09.064Z DRIVER « [Node 018] [REQ] [ApplicationCommand]`

[zwave.log](https://github.com/zwave-js/node-zwave-js/files/9270868/zwave.log)

Caros2017, 2022-08-06:
> This was never the case, so guessing something has been skewed with the handling, as this previously was flawless.

@marcus-j-davies I have definitely not enough knowledge to give you an answer, I am not sure if this is the same problem as the people with V2 keypad are facing. One thing I notice in your logs is dat I see this three times:
```
2022-08-05T18:19:08.967Z DRIVER « [Node 018] [REQ] [ApplicationCommand]
                                  └─[Security2CCMessageEncapsulation] [INVALID]
                                      error: Security2CC_NoSPAN
2022-08-05T18:19:08.968Z CNTRLR » [Node 018] No SPAN is established yet, cannot decode command. Requesting a non
                                  ce...

```

> Are you just disabling the Proximity Detection config setting (126-112-0-15) or something else? I just turned that off permanently via Z-Wave JS so will see if I can still reproduce the problem.

Yes we basically do the same. You turn it of permanently, I only do this once the alarm is armed:
Armed: `{"command_class":"112","endpoint":"0","property":"15","value":"0"}`
Disarmed: `{"command_class":"112","endpoint":"0","property":"15","value":"1"}`

I am really interested in your experiences now that it's turned off.

mundschenk-at, 2022-08-06:
> Maybe this is all just caused by the S2 collisions. Can anyone else confirm that [this](https://github.com/zwave-js/node-zwave-js/issues/4879#issuecomment-1205367143) happens for them too?

Yes. I just experimented a bit with the keypad again and had the red network light go up: 

```
2022-08-06T10:03:14.793Z SERIAL » 0x011a001369139f03c0005ff7aa313b79296255996c691798f8259bea          (28 bytes)
2022-08-06T10:03:14.794Z DRIVER » [Node 105] [REQ] [SendData]
                                  │ transmit options: 0x25
                                  │ callback id:      155
                                  └─[Security2CCMessageEncapsulation]
                                    │ sequence number: 192
                                    └─[IndicatorCCSet]
                                        values:
                                        · indicatorId: 9
                                          propertyId:  1
                                          value:       100
2022-08-06T10:03:14.800Z SERIAL « [ACK]                                                                   (0x06)
2022-08-06T10:03:14.803Z SERIAL « 0x0104011301e8                                                       (6 bytes)
2022-08-06T10:03:14.804Z SERIAL » [ACK]                                                                   (0x06)
2022-08-06T10:03:14.804Z DRIVER « [RES] [SendData]
                                    was sent: true
2022-08-06T10:03:14.825Z SERIAL « 0x0122000400691a9f03d900b4f7a83a437d4d5046e7f1805554b887496cc74d723 (36 bytes)
                                  2bb00a7
2022-08-06T10:03:14.831Z SERIAL » [ACK]                                                                   (0x06)
2022-08-06T10:03:14.832Z DRIVER « [Node 105] [REQ] [ApplicationCommand]
                                  └─[Security2CCMessageEncapsulation] [INVALID]
                                      error: Security2CC_CannotDecode
2022-08-06T10:03:14.833Z CNTRLR » [Node 105] Message authentication failed, cannot decode command. Requesting a 
                                  nonce...
```

marcus-j-davies, 2022-08-06:
I have just tested something....

set `Number of Re-transmissions Attempts` or `Message Retry Attempt Limit` for Gen2 to 0
This seems to have removed the issue for me - not sure why, as its only become necessary, but yet I'm on V10 Beta

AlCalzone, 2022-08-08:
> Yes. I just experimented a bit with the keypad again and had the red network light go up:

Can you share a bit more before and after the snippet?

AlCalzone, 2022-08-08:
> Previously, this was around 1-3s

 @marcus-j-davies yours is definitely a bug - and I assume a new one. This should never happen:
```
                                    └─[SupervisionCCGet]
                                      │ session id:      63
                                      │ request updates: true
                                      └─[SupervisionCCReport]
```

marcus-j-davies, 2022-08-08:
Yup - I guessed something to that affect - Prior to V10, its been flawless.
Let me know if you need anything further (unless you want me to log it separately? )

AlCalzone, 2022-08-08:
All good, I can reproduce this with my new testing framework.

mundschenk-at, 2022-08-08:
> > Yes. I just experimented a bit with the keypad again and had the red network light go up:
> 
> @mundschenk-at Can you share a bit more before and after the snippet? I need to see what leads to the communication issue and how it recovers (or not) afterwards.

Sure!

[Ring-Keypad-V2-collision.log.zip](https://github.com/zwave-js/node-zwave-js/files/9284065/Ring-Keypad-V2-collision.log.zip)

sammyke007, 2022-08-10:
Aha! I was almost returning my keypads because of this. they work great for sometime, but then theystop responding all of a sudden. Red network symbol shows up, the arm/disarm command does get send to HA, but the keypad "freeze's" and doesn't follow the new arming state.

Caros2017, 2022-08-10:
> Aha! I was almost returning my keypads because of this. they work great for sometime, but then they stop responding all of a sudden. Red network symbol shows up, the arm/disarm command does get send to HA, but the keypad "freeze's" and doesn't follow the new arming state. Connection is great so that's def. not the problem.

Have you tried turning off the motion sensor? I did it 5 days ago(when the alarm is armed) and since then no collisions anymore.

sammyke007, 2022-08-10:
> > Aha! I was almost returning my keypads because of this. they work great for sometime, but then they stop responding all of a sudden. Red network symbol shows up, the arm/disarm command does get send to HA, but the keypad "freeze's" and doesn't follow the new arming state. Connection is great so that's def. not the problem.
> 
> Have you tried turning off the motion sensor? I did it 5 days ago(when the alarm is armed) and since then no collisions anymore/yet.

I've disabled them just now on both keypads. I'll report back!

pethans, 2022-08-10:
> A couple of ways to test this: add a delay before the entry delay indicator is set (at line 201 of https://github.com/ImSorryButWho/HomeAssistantNotes/blob/main/alarm_keypad_automations.yaml). Or, try covering the motion sensor with electrical tape.

I was also seeing this problem, so thank you to everyone looking into this. Just my input: since I'm using the motion sensor, I added a 3-second delay in the HA Blueprint when `alarm_pending` is triggered and I haven't had the signal loss / collision problem since — 5 out of last 5 entries and disarms successful (vs approx 1 out of 10 before). I will continue testing with smaller delays and report back if I see the issue again.

AlCalzone, 2022-08-11:
FYI, I can re-create the collisions in simulated tests now without any devices https://github.com/zwave-js/node-zwave-js/pull/4900, so I should hopefully have a solution for some of the more common collision situations soon.

Unfortunately I can't find anything in the specs or elsewhere what the "correct" behavior is when two nodes send roughly at the same time.

sammyke007, 2022-08-11:
Ok, tnx! Will this only be fixed in the node version or also in zwavejs2mqtt?

Caros2017, 2022-08-11:
> FYI, I can re-create the collisions in simulated tests now without any devices #4900, so I should hopefully have a solution for some of the more common collision situations soon.
> 
> Unfortunately I can't find anything in the specs or elsewhere what the "correct" behavior is when two nodes send roughly at the same time, so this will be a best-effort guess.

That's a really nice feature that you can simulate these issues now. 
I am not sure if the Zwave specs are providing a solution for this. I can only find that rapid status changes need a 250ms delay to avoid collisions.

I think this situation should never happen and is the result of using a device with open software. My assumption is this:
Ring is providing complete alarm systems. This keypad is part of it. If you use the ring system, the motion sensor doesn't send Zwave updates, but is only used for lighting up the keypad if you are close.
Now this thing is used in an open environment en some people find the motion sensor option. We all use this and now we are seeing collisions.

I am not sure if you can fix this in ZwaveJS. Probably the solution should be made within the keypad. Which is probably not gonna happen.

I am now just enabling the motion sensor after dearm and disable it after arming.

AlCalzone, 2022-08-11:
> Ok, tnx! Will this only be fixed in the node version or also in zwavejs2mqtt?

All applications use this as the driver, so at some point this fix will be everywhere.

> I can only find that rapid status changes need a 250ms delay to avoid collisions.

That can be part of it. If only only one party sends in a rapid fashion this shouldn't be a problem either. The main problem is that S2 relies on a shared state to perform encryption and decryption. Lets say it starts at `N`. A "normal" flow is:
* A and B are at state N
* A sends command (N), it is now at state N+1
* B receives command (N), it is now at state N+1
* B sends command (N+1), it is now at state N+2
* A receives command (N+1), it is now at state N+2

Or in the case of one party transmitting rapidly:
* A and B are at state N
* A sends command (N), it is now at state N+1
* B receives command (N), it is now at state N+1
* A sends command (N+1), it is now at state N+2
* B receives command (N+1), it is now at state N+2
* ...

As you can see, in a neatly ordered communication, both parties are in sync most of the time, except when messages are in flight. When they are out of sync, the receiving party can just say "hey I didn't get that, here's my half of a new state", and the sender can create a new state and retry the message.

Now what happens here is this:
* A and B are at state N
* B sends command (N), it is now at state N+1
* At the same time A sends command (N), it is now at state N+1
* A receives command (N), but cannot decrypt since it is at state N+1
* A attempts re-synchronization by sending B its half of the new state M
* B receives command (N), but cannot decrypt since it is at state N+1
* B attempts re-synchronization by sending A its half of the new state K
* B receives re-sync attempt from A and discards its state K (is now at M, half-synchronized)
* A receives re-sync attempt from B and does not know what to do with it - it expected a re-transmission of B's command (N)
* chaos

This eventually resolves itself when the next command is sent with a little delay, but at least one of those messages N is going to get lost. By delaying/turning off the motion detection (B's command N) you eliminate the situation altogether.

I plan on preventing this by allowing A to decrypt a received command with state N for a short time after it has sent its own command at state N, but only under very specific circumstances. This way, A doesn't attempt re-synchronization and we should end up with a re-sync flow as it is found in the specs.

sammyke007, 2022-08-11:
Ok, great explanation, tnx! For me the red signal also popped up sometimes when trying to arm, so not only while disarming. PRobably due to the otion detection + exit delay starting. Since yesterday no problems anymore after dsabling the proximity/motion sensor. I'll keep posting results.
I do miss the keypad lighting up function a bit, but it's a lot more important to keep it working for now (WAF ;-) )

sammyke007, 2022-08-13:
Disabling proximity detection / motion sensor indeed seems to solve the problem on this 2 day testing. Keeps working perfectly fine now. I wonder if setting the "Proximity Display Timeout" to eg. 15 seconds instead of 5 seconds also helps. I'll test further.

closed 2022-08-15 by AlCalzone

AlCalzone, 2022-08-15:
Should be fixed in the upcoming v10 release (hopefully)

sammyke007, 2022-08-15:
Great tnx! Zwavejs2mqtt will follow soon?

AlCalzone, 2022-08-15:
Soon-ish. We're still ironing out a few things between the driver, z2m and HA.

davidcoulson, 2022-08-22:
Probably a terrible idea from a security perspective, but has anyone tried pairing the keypad with S0 instead of S2 to manage this issue until it is corrected?

ImSorryButWho, 2022-08-22:
> Probably a terrible idea from a security perspective, but has anyone tried pairing the keypad with S0 instead of S2 to manage this issue until it is corrected?

I experimented with that when I first got one, before ZWaveJS supported S2.  It won't do anything useful (e.g. Entry Control doesn't work) unless it's paired as S2.

AlCalzone, 2022-08-22:
> Probably a terrible idea from a security perspective, but has anyone tried pairing the keypad with S0 instead of S2 to manage this issue until it is corrected?

S0 is not less secure than S2, except for the initial key exchange, which could theoretically be sniffed

sammyke007, 2022-09-01:
Am I correct that todays zwavejs2mqtt release 7.0.0 includes this fix?

AlCalzone, 2022-09-01:
It should. But be aware that there may be some new issues with other devices which we're still investigating.
https://github.com/zwave-js/node-zwave-js/issues/4994

davidcoulson, 2022-09-02:
Has anyone been able to confirm that zwavejs2mqtt 7.0.0 fixes this, or even confirmed it does not? I will test tomorrow morning, but just updated my HA environment.

sammyke007, 2022-09-02:
> Has anyone been able to confirm that zwavejs2mqtt 7.0.0 fixes this, or even confirmed it does not? I will test tomorrow morning, but just updated my HA environment.

I've updated yesterday afternoon and re-enabled motion sensor on 1 of my 2 keypads. Ive armed/disarmed 3 times since then and it seems to be fixed. Longer testing is needed however.

Caros2017, 2022-09-02:
@AlCalzone first of all I want to thank you for trying to fix this issue so fast and all your effor you are putting in to zwave-js.

Even though there seem to be some issues with zwavejs2mqtt 7.0.0 (zwave-js v10), I immediatly upgraded. To try and trigger this issue I have enabled the motion sensor again and purposely set the `[23-112-0-26] Motion Sensor Timeout ` to 1 second. Yesterday I had no issues. Today unfortunately I saw the network light on the keypad showing red and being inresponsive for a while. This happend while trying to arm. In the log at that time I seem to find the same behaviour again:
```
2022-09-02T07:56:36.937Z DRIVER « [Node 023] [REQ] [ApplicationCommand]
                                  └─[Security2CCMessageEncapsulation] [INVALID]
                                      error: Security2CC_CannotDecode
2022-09-02T07:56:36.938Z CNTRLR » [Node 023] Message authentication failed, cannot decode command. Requesting a 
                                  nonce...
```

I have added some more information of the log around it in the text file.

AlCalzone, 2022-09-05:
@Caros2017 the behavior you quoted looks like it is working as intended though:

The driver receives a command that it cannot decode (why ever) and tells the device:
```
2022-09-02T07:56:36.937Z DRIVER « [Node 023] [REQ] [ApplicationCommand]
                                  └─[Security2CCMessageEncapsulation] [INVALID]
                                      error: Security2CC_CannotDecode
2022-09-02T07:56:36.938Z CNTRLR » [Node 023] Message authentication failed, cannot decode command. Requesting a 
                                  nonce...
2022-09-02T07:56:36.944Z SERIAL » 0x011b001317149f022f013194c0dd33297b14141c217ee08fa13405f6d4        (29 bytes)
2022-09-02T07:56:36.944Z DRIVER » [Node 023] [REQ] [SendData]
                                  │ transmit options: 0x05
                                  │ callback id:      246
                                  └─[Security2CCNonceReport]
                                      sequence number:  47
                                      SOS:              true
                                      MOS:              false
                                      receiver entropy: 0x3194c0dd33297b14141c217ee08fa134
```
The device immediately re-transmits the command with a new nonce/EI:
```
2022-09-02T07:56:37.014Z DRIVER « [Node 023] [REQ] [ApplicationCommand]
                                  └─[Security2CCMessageEncapsulation]
                                    │ sequence number: 129
                                    │ extensions:
                                    │ · type: SPAN
                                    │   sender EI: 0x42053c13fe5df0a5911a1a338a2ce475
                                    └─[SupervisionCCGet]
                                      │ session id:      4
                                      │ request updates: false
                                      └─[NotificationCCReport]
                                          notification type:   Home Security
                                          notification status: 255
                                          notification state:  Motion detection
```
The driver confirms receipt of that message:
```
2022-09-02T07:56:37.024Z DRIVER » [Node 023] [REQ] [SendData]
                                  │ transmit options: 0x24
                                  │ callback id:      0
                                  └─[Security2CCMessageEncapsulation]
                                    │ sequence number: 48
                                    └─[SupervisionCCReport]
                                        session id:          4
                                        more updates follow: false
                                        status:              Success
                                        duration:            0s

```

After that I'm not sure what is happening. The device sends motion detection and motion idle reports repeatedly every 0.5 seconds (each one is confirmed) and also this one:
```
2022-09-02T07:56:37.294Z DRIVER « [Node 023] [REQ] [ApplicationCommand]
                                  └─[Security2CCMessageEncapsulation]
                                    │ sequence number: 133
                                    └─[SupervisionCCGet]
                                      │ session id:      8
                                      │ request updates: false
                                      └─[NotificationCCReport]
                                          notification type:   System
                                          notification status: 255
                                          notification state:  System software failure (with failure code)
                                          event parameters:    0xad
```
which seems to mean "Dropped Frame". I don't see any dropped frames though.

sammyke007, 2022-09-06:
For me it's still working perfectly fine with motion detection enabled on my 2 keypads. Tnx a lot!!

sammyke007, 2022-09-13:
One more small bug related to this.
I've enabled the motion sensor again a week ago on one of my two keypads and it has been working fine, except one situation that happened multiple times now on the motion activated keypad only:

When the house is armed and I enter (open/activate a sensor), the entry delay starts. The keypad with the motion sensor enabled doesn't start the entry delay procedure (red bar countdown/up). I can still use the keypad however to disarm the house, but someone else doesn't know the alarm is armed and the entry delay is running.
I haven't had the time to find the logs however. Am I the only one?

KidA001, 2023-12-21:
This issue persists unless I disable motion on the keypad, then it doesn't happen. I'm on the newest version of ZWaveJS as a AddOn in HomeAssistant. Is anyone else still experiencing this?

sammyke007, 2023-12-21:
I've had the motion sensor disabled for months and it has been running perfectly without that. Funny enough I've enabled the motion sensor again since yesterday to test if this is fixed. Apparently it isn't? I'll report back

joostlitjens, 2024-05-28:
> I've had the motion sensor disabled for months and it has been running perfectly without that. Funny enough I've enabled the motion sensor again since yesterday to test if this is fixed. Apparently it isn't? I'll report back

I'm new too the table, what are your findings after enabling motion?

sammyke007, 2024-05-28:
Still occasional red antenna's while motion sensor is enabled...

joostlitjens, 2024-05-28:
> Still occasional red antenna's while motion sensor is enabled...

Thank you for the quick response, im also going too disable motion, for a higher WAF😝

hazardousmedic, 2025-04-14:
Has anyone found a way around this, other than just disabling the motion sensor? It's clearly still broken on some level via z-wave. I can replicate this regularly, and it's been a pain.

If I disable the motion sensor I fumble in the dark without a lit keypad at 5am after a night shift to disarm the damn thing. If I leave it on it sees the door open and the motion event seems to conflict with the countdown start and it locks up dropping the connection for a while.

[8 more replies]