zwave-js/zwave-js #4889: [question] POPP Z-Weather strange humidity value

opensource-alt, 2022-08-08:


AlCalzone, 2022-08-08:
It also reports 100 for the illuminance and pressure. This can't be a coincidence. I'd contact the manufacturer in your case.

opensource-alt, 2022-08-08:
100 for pressure was purely coincidental. Illuminance is at 100% all day and change at night (the device change the report frequency below 20%). The only strange value is the humidity (it never changes, always 100%, an impossible value...)

This is a used purchase. The previous owner told me that it had no strange value. He didn't use the same software though (and I can't verify his statement about the values with the other software)

[zweater-debuginfo.txt](https://github.com/zwave-js/node-zwave-js/files/9283451/zweater-debuginfo.txt)

AlCalzone, 2022-08-08:
That's not a driver log. If you want me to look at it, please read the link in my previous post again.

opensource-alt, 2022-08-09:
Sorry for the missing log file...
I filtered logs to the Popp Z-Weather node only (node ID 34) as there's many devices in my ZWave network (25 + controller). Tell me if it's more relevant to remove this filter.
I'm really unsure this issue could be related to ZWaveJS mishandling of this device but as the previous owner told me he didn't have those strange values...

[zwavejs_2022-08-09.log](https://github.com/zwave-js/node-zwave-js/files/9288779/zwavejs_2022-08-09.log)

AlCalzone, 2022-08-09:
The sensor sends this value. Here's a breakdown of the command that is received:
```
010d00 040022 05 3105 05 01 64 b10034
├────┘ ├────┘ ├┘ ├──┘ ├┘ ├┘ ├┘ ├────┘ 
│      │      │  │    │  │  │  └ Protocol related
│      │      │  │    │  │  └ value 100
│      │      │  │    │  └ 0 decimal digits, scale 0 (%), 1 byte
│      │      │  │    └ Humidity
│      │      │  └ Multilevel Sensor Report
│      │      └ Length 5
│      └ Incoming packet from node 34
└ Protocol related
```

closed 2022-08-12 by zwave-js-assistant[bot]