zwave-js/zwave-js #4917: [Bug]: Adding Zooz ZSE44 keeps failing (incomplete/multiple nodes added, all not working)

alexruffell, 2022-08-15:


alexruffell, 2022-08-20:
@AlCalzone Yes I realize that I was likely adding issues as I was unaware that Smart Start would include it without any user action... Anyhow today I reset one of my Zooz ZSE44 sensors and then let it sit there. I did not touch or do anything.

First I got:
[image]

Then I got this:
[image]

In the snapshot above you can see it failed adding 118, and then ended up adding the sensor as 119. This is the exact behavior I've been seeing with all the ZSE44 sensors I've added. I believe I've seen this before with other 700 series devices but I am not certain.

AlCalzone, 2022-08-22:
I've actually never seen a device be added twice. I guess you interrupted the previous inclusion attempt under ID 118 by resetting it, and when you left it alone it properly added as 119.

closed 2022-09-09 by zwave-js-assistant[bot]