zwave-js/zwave-js #4935: CentraLite Systems, Inc/Lowe's Iris Smart Plug 3210-L

RichJeanes, 2022-08-19:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within ZWaveJS2MQTT?

NO, my problem is NOT within ZWaveJS2MQTT

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?
Attempting to include the device

What do you observe?
Inclusion fails.

What did you expect to happen?
For the device to be included as a repeater in the network

Steps to reproduce the behavior:
  1. Click "+ Add Device"
  2. Plug the outlet in and toggle the outlet on and off using the button on the front of the device

### Device information

Manufacturer: CentraLite Systems, Inc
Model name: Smart Plug 3210-L
Device is a Zigbee outlet with a Z-wave repeater. Zigbee control and reporting works without issue (Zigbee2MQTT).

### How are you using `node-zwave-js`?

- [ ] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [X] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version: 0.1.65

### Did you change anything?

no

### Did this work before?

Yes (please describe)

### If yes, where did it work?

Device worked previously with Samsung SmartThings hub.

### Attach Driver Logfile

[zwave_js.log](https://github.com/zwave-js/node-zwave-js/files/9384639/zwave_js.log)

AlCalzone, 2022-08-22:
> status = Failed

Usually means there is a connectivity issue during the initial communication between the stick and the node.
We don't get more insight what it is, but the stick location is a frequent cause of this.

Start by following these steps: https://zwave-js.github.io/node-zwave-js/#/troubleshooting/connectivity-issues?id=general-troubleshooting

If that doesn't help, you can try these things:
- bring the device closer
- power cycle both the device and the stick (restart is not enough)

RichJeanes, 2022-08-22:
My Z-Wave stick is on an extension, elevated away from my other equipment. I tried adding the device while it was within two feet of the stick after power cycling, no change.
[image]
[zwave_js(1).log](https://github.com/zwave-js/node-zwave-js/files/9396929/zwave_js.1.log)
Since opening this issue, I have successfully added a different device farther away from the stick and on the outside of my house, so I don't think 
What are the odds that this is just an issue with my device? It's admittedly been several years since I used it with my SmartThings hub. Honestly not worth a ton of effort to get it working. I only have the one and not a ton of Z-Wave devices that need a repeater anyway.

AlCalzone, 2022-08-23:
> What are the odds that this is just an issue with my device? 

Really can't tell. The communication that leads to the "failed" status is unfortunately hidden from us and can only be observed with a Zniffer (separate stick with special firmware and a Windows software to capture network traffic).

Two more things you could try, then I'm out of ideas:
- start an **exclusion**, then actuate the device as if you wanted to include it. Then retry the inclusion
- factory reset it, then retry the inclusion

dshokouhi, 2022-09-06:
I have this device myself. Although I have not paired the zwave portion of the plug the steps don't seem right. The steps shown are to pair the zigbee part. IIRC in order to pair the zwave extender we need to hit the power button very fast about 8 times to get it to pulse like a heartbeat to indicate the zwave portion is pairing. It is very tricky to get the zwave part to pair in my own experience so I usually don't pair it.

For reference here is a forum post from a different product that also mentions the pairing differences in zwave vs zigbee for this device: https://community.hubitat.com/t/iris-3210-l-integration-issue/34938/3

One more from a different site for the same device: https://community.smartthings.com/t/release-iris-smartplug-z-wave-repeater-with-diagnostics/111183/99

RichJeanes, 2022-09-07:
@dshokouhi Wow. That worked. Paired in seconds, first try, with no issues. Thanks for chiming in!

closed 2022-09-07 by RichJeanes