zwave-js/zwave-js #4940: [feat] Automatically reset security classes when re-interviewing an S2 authenticated device after firmware upgrade

cmamatey, 2022-08-22:


jmgiaever, 2022-08-22:
Can manufacture id, product id etc change on a fw update @AlCalzone? Would this cause the cache entry to be invalidated and require a fresh re-interview?

cmamatey, 2022-08-22:
Screenshot of issue:
[image]

Screenshot of normal display (before upgrade or after Reset security classes):
[image]

cmamatey, 2022-08-22:
I have the log ready.
It is ~96k lines long, so I assume I need to attach as a file.
Before I do, is there anything that would be contained in the log that I'm going to want to redact for privacy/security reasons?

FYI it has happened every time for this particular device for both upgrades and downgrades.
That said, it is currently my only device configured with S2 authentication, and I would absolutely believe that S2 is not configured correctly on my system.
It's also just as likely that my controller (Nortek HUSBZB-1) doesn't support it properly; I was surprised when I was originally able to add the device to my network with S2 authentication.

jmgiaever, 2022-08-22:
You can encrypt it as a zip file and attach it here, and send the encryption key in an email to @AlCalzone . You'll find his email on his profile.

cmamatey, 2022-08-22:
[zwavejs_2022-08-22.zip](https://github.com/zwave-js/node-zwave-js/files/9397766/zwavejs_2022-08-22.zip)

The attached log file includes the following 4 key events:
1. Change logging to Debug
2. Firmware upgrade of Node 15 including interview with bad results
3. Additional interview with bad results
4. Final interview with Reset security classes checked, resulting in successful interview

The text "[Node 015] The node is ready to be used" in the log seems to be a pretty good indicator of where each event ends.

Emailing the encryption key to @AlCalzone now.

AlCalzone, 2022-08-23:
I **think** the interview after the firmware update was just started too early. While the driver did wait for the specified wait time of 30 seconds, the device responded a bit strange after that. A few seconds later, things were better.

In addition you ran into a bug that will be fixed in the v10 release: https://github.com/zwave-js/node-zwave-js/pull/4923

closed 2022-08-23 by AlCalzone