zwave-js/zwave-js #4953: [Bug]: Firmware updates time out if the target device is not directly connected to the controller

bradreardon, 2022-08-27:


AlCalzone, 2022-08-27:
@bradreardon This seems like you were having connectivity issues before changing sticks. Unlike most other communication, firmware updates are driven by the devices. After zwave-js initializes the upgrade, the devices request each fragment in sequence and zwave-js answers with the requested fragment.

When these requests stop for more than 30s, zwave-js considers the upgrade aborted/timed out. It seems like your devices were having trouble requesting the next fragment after a while (or the responses got lost).

We have a couple of write-ups related to detecting and fixing connectivity issues in the documentation. It might be worth working through those before attempting further firmware update troubleshooting:
1. Controller placement: https://zwave-js.github.io/node-zwave-js/#/troubleshooting/connectivity-issues?id=general-troubleshooting
2. Making sure the network isn't flooded: https://zwave-js.github.io/node-zwave-js/#/troubleshooting/network-health?id=optimizing-the-reporting-configuration
3. Measuring the connection strength: https://zwave-js.github.io/node-zwave-js/#/troubleshooting/network-health?id=testing-the-connection-strength - you'll find the health check on the network map in zwavejs2mqtt when you click a node

bradreardon, 2022-08-31:
Hey @AlCalzone, thanks for the information. That's helpful in understanding how these firmware updates might be failing. It sounds like I do have a bit more investigation to do on my end.

As for the issue of controller placement, I did try using a USB extension cord to connect my 500-series controller in the past, but strangely I saw worse connectivity with the extension cord in place. I'll give it another shot with the new controller and see if that helps reduce interference and get everything connected directly to the controller.

The note about network flooding is also helpful -- I do have a device that's sending power usage reports for 2 sockets on a 15-20 second interval currently, so I'll have to look into whether or not that node reporting power usage is correlated at all to the firmware updates failing.

Finally, for the network health suggestion -- I have tried using this tool before, but didn't know how to make sense of the information within it, so I appreciate you sending that page over.

I have a busy week so I'm not sure I'll be able to play around with this again until the weekend, but I'll be sure to report back with my findings. Thanks for the help!

bradreardon, 2022-09-04:
Apologies: accidentally posted a duplicate response! Here's my resolution:

--

After doing some tests with my controller connected via a USB extension cord, I've confirmed that all nodes in my network are able to maintain a direct connection to the controller. I was able to complete firmware updates for the last two switches that I was experiencing issues with as a result.

Still quite interesting that the update fails for cases where the node is not directly connected! Thanks for the help in troubleshooting!

closed 2022-09-04 by bradreardon

jmgiaever, 2022-09-04:
My guess is that it's caused by the interference from the USB, which in turn garbles up the radio data, and not because of a bad connection (between the nodes) per se. 

I once tried to stack a PoE HAT with a Z-wave Razberry controller board on top, onto my Pi. It couldn't even communicate with a device that was just next to the controller, cause the signals got garbled.