zwave-js/zwave-js #4955: Missing device configuration: Minoston MP31ZP

InSearchOfTh1ngs, 2022-08-27:
### Checklist

- [X] It is not in the [configuration DB](https://devices.zwave-js.io/?search=3)

- [X] It was not [merged recently](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Apr+is%3Aclosed) or has a [pending PR](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Aopen+is%3Apr)

### Which device is missing?

Minoston MP31ZP

### Manufacturer ID

786

### Product Type

65280

### Product ID

65294

### Firmware Version

FW 1.0; SDK 7.13.10

### Is the device listed on the Z-Wave Alliance website?

https://products.z-wavealliance.org/products/4626?selectedFrequencyId=-1

InSearchOfTh1ngs, 2022-08-27:
Seems like this device is showing up as the MP21ZP not the MP31ZP

AlCalzone, 2022-08-27:
See https://github.com/zwave-js/node-zwave-js/pull/4882 and https://github.com/zwave-js/node-zwave-js/pull/4894
It seems like Minoston reused the MP21ZP firmware, making the MP31ZP internally the same.

closed 2022-08-27 by AlCalzone