zwave-js/zwave-js #4962: [Bug]:

AaronAutomation, 2022-08-29:
### Checklist

- [X]  I am not using Home Assistant. Or: a developer has told me to come here.
- [X] I have checked the troubleshooting section and my problem is not described there.
- [X] I have read the changelog and my problem is not mentioned there.

### Deploy method

Docker

### Zwavejs2Mqtt version

6.15.2

### ZwaveJS version

9.6.2

### Describe the bug

GE/Jasco - Direct Wire Indoor/Outdoor Smart Switch - 14285
FW: v5.55
SDK: v6.81.1

After updating to the newest version of Zwavejs2Mqtt, changing the state of the Binary Switch v2 for any GE/Jasco devices (model and version above) will occasionally result in the state reverting back to the previous state after exactly 1 second. This happens one to two times in a row after not being changed for a few hours, before correctly moving to the target value. I first noticed this happening in Home Assistant, but quickly ruled out HA as the issue by changing the state directly in the Zwavejs2Mqtt dashboard. This doesn't affect any other devices. 
[image]

### To Reproduce

Change the target value from on -> off or off -> on (depending on which state is active)

### Expected behavior

When changing the target value from on -> off, or off -> on, the value should stay and not revert back.

AaronAutomation, 2022-08-30:
[zwavejs_2022-08-30.log](https://github.com/zwave-js/node-zwave-js/files/9449812/zwavejs_2022-08-30.log)

Here is driver log of Node 5's manual binary state target value being updated from off -> on, then reverting back to off less than a second later.

AlCalzone, 2022-08-30:
The issue here is that the device did not understand the SET command. Driver version 10 has changed a few things in this regard, you may want to try that version. @robertsLando should be able to tell you how to test it.

robertsLando, 2022-08-30:
Just try to use `test` tag instead of `latest` if you are using docker as you mentioned in the issue

AaronAutomation, 2022-08-30:
Okay thanks, updated my stack and running 10.0.1. I'll report back with any updates/logs in the next few days.

AaronAutomation, 2022-09-06:
Since updating to the test tag the issue has not occurred FYI

robertsLando, 2022-09-06:
try to switch to `master` tag. That's the most updated now

AlCalzone, 2022-09-06:
I'll close then. Feel free to reopen if it happens again.

closed 2022-09-06 by AlCalzone