zwave-js/zwave-js #4974: Fibaro Roller Shutter 3 tilt not working

JakubMacoun, 2022-08-30:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within ZWaveJS2MQTT?

NO, my problem is NOT within ZWaveJS2MQTT

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

When I try to set my blinds tilt (open/close/set position), nothing happens. opening or closing blinds it self works fine, just no tilting of the slates for me.
In HA this can be achieved by calling one of services `cover.open_cover`, `cover.close_cover` or `cover.set_cover_position` on the tilt entity. Calling `cover.set_cover_tilt_position` does not work (as it never did).
Device parameter 151 (operating mode) is set to `Venetian blind`.

Downgrading to HA addon verson 0.1.65 solved the problem for now. Upgrading to 0.1.67 did not solve the issue.

### Device information

Manufacturer: Fibaro
Model name: Roller Shutter 3 (FGR223)
Node ID in your network: 033

### How are you using `node-zwave-js`?

- [ ] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [X] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version: 10.0.1
HA addon version: 0.1.66/67

### Did you change anything?

no

### Did this work before?

Yes (please describe)

### If yes, where did it work?

It works on HA version 0.1.65 (I guess that's Z-Wave JS 10.0.0)

### Attach Driver Logfile

[zwave.txt](https://github.com/zwave-js/node-zwave-js/files/9454788/zwave.txt)

AlCalzone, 2022-08-30:
While the logfile is indeed a driver log, it is not on **loglevel debug** like requested in the issue template.

JakubMacoun, 2022-08-31:
All right, I'll try to reproduce with newer version and correct log level later today.

AlCalzone, 2022-08-31:
FYI: https://github.com/home-assistant/core/issues/77466#issuecomment-1232810702

JakubMacoun, 2022-08-31:
Upgrading to 0.1.68 and re-interviewing the devices solved the issue. Thanks.

closed 2022-08-31 by JakubMacoun