zwave-js/zwave-js #4980: Themostats are not responding to mode change: Heat  <-> Turn Off (From version 1.66 - 1.68)

oivindklausen, 2022-08-31:
Thermostats not responding to (or slow/unstable response) Mode change: Heat  <-> Turn Off

AlCalzone, 2022-08-31:
Please make a [driver log](https://zwave-js.github.io/zwavejs2mqtt/#/troubleshooting/generating-logs?id=driver-logs), loglevel `silly` and attach it here as a file.
The log should contain a re-interview and the non-working interaction.

oivindklausen, 2022-08-31:
Pleas find attached log for 0.1.68. Tried to re-interview Node 16, but this time it did not complete (Still running after 5minutes).

 Home Assistant 2022.8.7
Supervisor 2022.08.5
 Operating System 8.5
[zwave_js_v0168.log](https://github.com/zwave-js/node-zwave-js/files/9464718/zwave_js_v0168.log)

 Frontend 20220802.0 - latest

mcowell, 2022-09-02:
Just want to add I'm seeing the same issue on Home Assistant

oivindklausen, 2022-09-03:
@AlCalzone . This issue seems related to  issue [https://github.com/zwave-js/node-zwave-js/issues/4957] and  what  @jl-678 see.  
It is like the GUI (Thermostat Card)  is dead and never changes to reflect  thermostat changes. I am able to switch the TF016 Switch on and off.

AlCalzone, 2022-09-05:
> Tried to re-interview Node 16, but this time it did not complete

It didn't even start. There's no occurence of the word "interview" in that log.

oivindklausen, 2022-09-05:
Please see attached log when trying to reinterview Node 016.

 [Node 016] Error during node interview: Node #16 (Endpoint 2) does not support
                                   the command GetSupportedSensor! (ZW0302)

[zwave_js(5).log](https://github.com/zwave-js/node-zwave-js/files/9491184/zwave_js.5.log)

AlCalzone, 2022-09-06:
Please re-test with v10.0.4. This version is included in:
zwave_js addon: 0.1.70
or
zwavejs2mqtt: 7.1.0

AlCalzone, 2022-09-07:
Fixed in Home Assistant 2022.9.0

closed 2022-09-07 by AlCalzone