zwave-js/zwave-js #4994: META Issue: Communication problems on v10.x

AlCalzone, 2022-09-01:
It looks like there were some changes in v10, which break communication with some devices. We're still in the process of figuring things out and need some information to do so.

### If devices don't work (fully or partially) after upgrading, _first re-interview them_.

If they still don't work, please **share the following information** (attached as files):
1. a node dump
2. a driver log (loglevel "**silly**") of a re-interview
3. a driver log (loglevel "**silly**") of the actions that don't work
4. What does not work?

Comments that just say "me too" or similar and don't provide information will be deleted.

bckmrk, 2022-09-01:
Hi, I have many devices not working after this update and attached is log No. 2 and 3 requested above. Where do i create a node dump? Logs above is specifically one device, Node 012 in my setup. 

2. [zwave_js-silly-Node012_re-interview.log](https://github.com/zwave-js/node-zwave-js/files/9470218/zwave_js-silly-Node012_re-interview.log)
3. [zwave_js-silly-Node012_lightsOn_NotWorking.log](https://github.com/zwave-js/node-zwave-js/files/9470219/zwave_js-silly-Node012_lightsOn_NotWorking.log)

4. The status of my dimmers updates in HA when I press the switches on the wall, but nothing happens with the lamps if I change the state of the switches on my phone or laptop. The switches in HA does not change either.

AlCalzone, 2022-09-01:
Your log shows that the dimmer accepted the command and sent an update after roughly one second.
```
# Turn on:
2022-09-01T13:23:32.106Z DRIVER » [Node 012] [REQ] [SendData]
                                  │ transmit options: 0x25
                                  │ callback id:      50
                                  └─[SecurityCCCommandEncapsulation]
                                    │ nonce id: 238
                                    └─[MultiChannelCCCommandEncapsulation]
                                      │ source:      0
                                      │ destination: 1
                                      └─[MultilevelSwitchCCSet]
                                          target value: 255
                                          duration:     default

...

# Value changed to 99
2022-09-01T13:23:33.266Z CNTRLR   [Node 012] [~] [Multilevel Switch] currentValue: 0 => 99          [Endpoint 1]
2022-09-01T13:23:33.271Z CNTRLR   [Node 012] Scheduled poll canceled because expected value was received
2022-09-01T13:23:33.273Z DRIVER « [Node 012] [REQ] [ApplicationCommand]
                                  └─[SecurityCCCommandEncapsulation]
                                    │ sequenced: false
                                    └─[MultiChannelCCCommandEncapsulation]
                                      │ source:      1
                                      │ destination: 1
                                      └─[MultilevelSwitchCCReport]
                                          current value: 99
```

(some power usage updates are there aswell)

mirkochip88, 2022-09-01:
Qubino Flush 1D relay, node dead and not responding. Interviewing fail.
[zwave_js.log](https://github.com/zwave-js/node-zwave-js/files/9473117/zwave_js.log)
[zwave_js-6e431c384e81d642f9ea95e419f54ab6-Flush 1D relay-fbf0b7ebeb27bbac5eec9a483e422e80.json.txt](https://github.com/zwave-js/node-zwave-js/files/9473119/zwave_js-6e431c384e81d642f9ea95e419f54ab6-Flush.1D.relay-fbf0b7ebeb27bbac5eec9a483e422e80.json.txt)

AlCalzone, 2022-09-02:
> Qubino Flush 1D relay, node dead and not responding

This claims it supports Multi Channel CC but does not respond to it at all. Interview gets aborted because this is considered a critical interview part.
Because it's Qubino, try excluding without reset and then include again. Their devices sometimes behave strangely when you change a config parameter that changes endpoints, but then don't re-include like the manual says.

bckmrk, 2022-09-02:
> Your log shows that the dimmer accepted the command and sent an update after roughly one second.
> 
> ```
> # Turn on command sent:
> 2022-09-01T13:23:32.106Z DRIVER » [Node 012] [REQ] [SendData]
>                                   │ transmit options: 0x25
>                                   │ callback id:      50
>                                   └─[SecurityCCCommandEncapsulation]
>                                     │ nonce id: 238
>                                     └─[MultiChannelCCCommandEncapsulation]
>                                       │ source:      0
>                                       │ destination: 1
>                                       └─[MultilevelSwitchCCSet]
>                                           target value: 255
>                                           duration:     default
> 
> ...
> 
> # Value changed to 99
> 2022-09-01T13:23:33.266Z CNTRLR   [Node 012] [~] [Multilevel Switch] currentValue: 0 => 99          [Endpoint 1]
> 2022-09-01T13:23:33.271Z CNTRLR   [Node 012] Scheduled poll canceled because expected value was received
> 2022-09-01T13:23:33.273Z DRIVER « [Node 012] [REQ] [ApplicationCommand]
>                                   └─[SecurityCCCommandEncapsulation]
>                                     │ sequenced: false
>                                     └─[MultiChannelCCCommandEncapsulation]
>                                       │ source:      1
>                                       │ destination: 1
>                                       └─[MultilevelSwitchCCReport]
>                                           current value: 99
> ```
> 
> (some power usage updates are there aswell).
> 
> Are you sure this is the correct log?

Yes I'm sure it's the correct log. However I'm not sure the lamp didn't respond as intended... I wasn't at home doing this log so couldn't check... will redo log's later today.

bckmrk, 2022-09-02:
Hmm, it's a little bit awkward to confess this, but re-interviewing every node that failed after the update helped with the problem. Now they seem to work perfectly again...

AlCalzone, 2022-09-02:
Thanks for the feedback though.

mirkochip88, 2022-09-02:
> This claims it supports Multi Channel CC but does not respond to it at all. Interview gets aborted because this is considered a critical interview part. Because it's Qubino, try excluding without [...]

It works, thanks!

ekzor, 2022-09-02:
Hi there, I'm experiencing an issue where my Zooz ZSE19 siren will no longer play sound clips ever since I updated to the latest version of zwavejs2mqtt. Sound clips for this device are played through a configuration parameter being set to an integer. The sounds won't play anymore and it looks like the config parameter is basically stuck; it won't change no matter what I do.

Here are my node debug info (is that a node dump?), as well as Silly-level logs of the interview and the command that doesn't work.
- [zwavejs2mqtt-node_dump.txt](https://github.com/zwave-js/node-zwave-js/files/9481072/zwavejs2mqtt-node_dump.txt)
- [zwave_js-interview.log](https://github.com/zwave-js/node-zwave-js/files/9481036/zwave_js-interview.log)
- [zwave_js-not_working.log](https://github.com/zwave-js/node-zwave-js/files/9481042/zwave_js-not_working.log)

Versions:
- zwavejs2mqtt 7.0.0
- zwavejs 10.0.3

Things I've tried:

1. Healing
2. Reinterviewing
3. Restarting the zwave container
4. Excluding / Reincluding
5. Reinterviewing again
6. Restarting again

Other things to note:

- I can see in the node's properties that `112-0-3` (the sound effect config parameter) is supposedly writeable, so I don't know why this is happening.
- The actual security siren functionality (i.e. the loud piercing wailing noise) and temperature/humidity reporting works just fine, i.e. I can turn the siren on and off and see values for the other stuff, so the node _is_ responsive. It's just setting the config parameter that doesn't seem to want to work.
- I haven't changed any of my settings or security keys, and my other S2-included devices (including other parts of this one) are still functional after the update.
- zwavejs2mqtt reports that the api call is successful although it's clear from the zwavejs driver logs that it actually isn't, but that's a different problem altogether :P

Things I'm about to try:

- Excluding / Reincluding without security (although ideally I wouldn't have to do this...)

Appreciate any help or advice. Thank you!

skynet-network, 2022-09-03:
I have mostly fibaro dimmer and sunricher dimmer 

Going back to a 0.1.4  backup and unplugg "me zwave" stick removed from vm and add to vm rebooted host machine then all my zwave router was available and responded

bckmrk, 2022-09-03:
My happiness was short-lived... now almost all of my z-wave devices are not working. Some commands pass through, but it takes a long, long time...  

[zwave_js_OnOff.log](https://github.com/zwave-js/node-zwave-js/files/9483075/zwave_js_OnOff.log)
[zwave_js_ReInterview.log](https://
[zwave_js-nodedump.json.txt](https://github.com/zwave-js/node-zwave-js/files/9483080/zwave_js-nodedump.json.txt)
github.com/zwave-js/node-zwave-js/files/9483076/zwave_js_ReInterview.log)

Tried to heal the network and reinterview nodes again without any change. Suggestions?

gralin, 2022-09-03:
I'm having issues with Zooz Open Close XS Sensor (ZSE41). I was doing the same test using latest HA addon 0.1.68 (zwave-js v10) and 0.1.65 (last zwave-js v9). In both tests I started with factory reset of the sensor and then tried to add it. The interview finished when using v9 but not in case of v10.

[zwave_js-37b187458f723b1a394110dcea8ea050-Node 20-ab82391c377dceca05b7f0bc778af64a.json.txt](https://github.com/zwave-js/node-zwave-js/files/9483425/zwave_js-37b187458f723b1a394110dcea8ea050-Node.20-ab82391c377dceca05b7f0bc778af64a.json.txt)
[log.txt](https://github.com/zwave-js/node-zwave-js/files/9483426/log.txt)

epower53, 2022-09-04:
I'm struggling with a Zooz Zen16 configured as a garage door opener. Both relays 1 and 2 are configured for garage door mode (parameters 16-112-0-2 and 16-112-0-3 set to a value of 3). Working just fine before upgrading to ZwaveJS2mqtt version 7.0.0. Now, when I send a command to open or close either garage door, I hear a quick click from the relays, but it's too fast for the opener to register it as a physical button press and the doors do not open / close. Looking at the relay states in the ZwaveJS2mqtt GUI and watching topics using MQTT Explorer, it looks as though there's some serious funkiness going on. 

Manually changing the individual relay states from true to false or vice versa shows up in the targetValue and currentValue subtopics for the appropriate relays, however no action was actually taken by the device, aside from the quick click I hear. However, if I toggle the master device state for the whole Zen16 from false to true, both relays 1 and 2 fire simultaneously, for the correct momentary contact duration, and the garage doors are toggled. I'd prefer to have control over each door individually, as I did with the 6.X.X branch of ZwaveJS2mqtt and 9.X.X branch of Z-wave JS.

Current versions are ZwaveJS2mqtt 7.0.0, Z-wave JS 10.0.3. Here are the logs
[nodes_dump.txt](https://github.com/zwave-js/node-zwave-js/files/9483805/nodes_dump.txt)
[interview log.log](https://github.com/zwave-js/node-zwave-js/files/9483806/interview.log.log)
[driver level log.log](https://github.com/zwave-js/node-zwave-js/files/9483807/driver.level.log.log)

AlCalzone, 2022-09-05:
@ekzor the ZSE19 doesn't seem to accept supervised commands correctly. The v10 version started using those extensively to increase reliability through S2. I've notified Zooz and will disable Supervision for that device as a short term fix.

@zwave-js-bot add compat flag to `0x027a:0x000c:0x0003:0.0`
```jsonc
commandClasses: {
  remove: {
    // Supervised commands always result in a status of "Fail"
    "0x6c": {
      endpoints: "*"
    }
  }
}
```

AlCalzone, 2022-09-05:
@bckmrk Your stick is hanging and does not respond to commands at all. It might need to be re-plugged.

> Error during node interview: Timeout while waiting for an ACK from the controller (ZW0200)
> 2022-09-03T17:21:48.947Z CNTRLR   Failed to execute controller command after 2/3 attempts. Scheduling next try i
                                  n 1100 ms.

AlCalzone, 2022-09-05:
@gralin that log is not on loglevel "debug". Please make another one.

gralin, 2022-09-05:
@AlCalzone I'm sorry, I have selected `silly` level in the Z-Wave JS addon configuration in HA but now when you pointed this out I noticed that after restarting, the log level of the addon is back to `info`. Now when I selected 'debug' instead of `silly` the setting survived the restart so I will perform the tests again and submit the proper logs. Is is enough when I copy the logs from the Log tab of the addon page? The start of the log seems to be truncated but I don't know where to get the complete ones.

AlCalzone, 2022-09-05:
@gralin This works: https://www.home-assistant.io/integrations/zwave_js#how-to-access-the-z-wave-logs

bckmrk, 2022-09-05:
> 

Is that usually needed after an update? I have never done that before. Will test it.

bckmrk, 2022-09-05:
> @bckmrk Your stick is hanging and does not respond to commands at all. It might need to be re-plugged.

Is that usually needed after an update? Never done that before. Will test it.

AlCalzone, 2022-09-05:
> Is that usually needed after an update?

It isn't usually, but it happens sometimes. Maybe the stick was still blocked by something else.

AlCalzone, 2022-09-05:
@epower53 I'm seeing lots of connectivity issues in your log. 48 duplicate incoming commands during the interview, RSSI values close to -90 dBm (which isn't much more than the background noise), several instances where routing suddenly uses 3 hops instead of 0, timed out responses.

This hints at a bad connection between the node and the controller, in both directions. I suggest you check out and follow this:
- https://zwave-js.github.io/node-zwave-js/#/troubleshooting/connectivity-issues?id=general-troubleshooting
- https://zwave-js.github.io/node-zwave-js/#/troubleshooting/network-health?id=testing-the-connection-strength (in zwavejs2mqtt on the network map after clicking a node)

The previous version probably works because all the information is already there, but the issue here is primarily in the real world.

epower53, 2022-09-05:
Sorry to disagree with you @AlCalzone, but I'm going to have to. This device has been installed and working for over 2 years on both Hubitat and then Z-Wave JS 9.X.X. It's unsurprising the direct connection statistics are poor - there's a great deal of distance and a significant amount of wood, plaster, and concrete between the device and my controller. Usually this device runs through at least one relay, sometimes two, after my mesh has settled in. I know you can only base on opinion on what you see in the logs, so I'm providing this extra info so you understand that this device NEVER misbehaved until I tried 10.0.3. My guess is that it's another S2 issue. It's paired S2 unauthenticated, and you mentioned that the ZSE19 was not playing nice with supervision packets. It's something in the code (either Zooz's or Z-wave JS's).

AlCalzone, 2022-09-05:
Can you share an interview log from 9.x then, so I can compare where the differences are?

epower53, 2022-09-05:
Happy to... here's an interview just completed using 9.6.2

[zwavejs_2022-09-05.log](https://github.com/zwave-js/node-zwave-js/files/9491115/zwavejs_2022-09-05.log)

AlCalzone, 2022-09-05:
The connectivity issues are also there on 9.x, so timeouts because of them are more like a game of chance. If they happen at the wrong time you'll see isses after re-interviewing on 9.x too.

However the issue you're experiencing is a different one. The Zen16 does simply not respond when asked which commands each endpoint supports securely. 9.x (incorrectly) didn't do anything with this info anyways and simply used the information from the root device. This means this worked out most of the time by accident.

10.x assumes that the endpoint's commands are not secure instead - which is probably not the best choice if the device is secure. I can work around that, but ultimately it's the device's fault.

epower53, 2022-09-05:
Zooz has been good offering support from an end customer's perspective  - hopefully they'll be willing to dig in and fix some of the issues the 10.X code branch is exposing.

labob246, 2022-09-05:
I am also having a Zooz Zen16 issue.
[interview.log](https://github.com/zwave-js/node-zwave-js/files/9491711/interview.log)
[Actions that dont work.txt](https://github.com/zwave-js/node-zwave-js/files/9491713/Actions.that.dont.work.txt)

I do not know how to do a node dump, but the node in question is 15 and all nodes are directly connected to the stick.
In the Actions that don't work.txt file, you will see that the device can be turned on and off but there is no individual control of relays 1, 2 or 3.  In the file I turn on the device (relay 0), try turning on relay 3, try turning off relay 3 and then turn off the device.  The device (relay 0) controls the on/off of 1 or 2 or 3 when they should be able to be turned on/off individually, which doesn't work.  It stopped working with this latest version:  
zwavejs2mqtt: 7.0.0
zwave-js: 10.0.3

gralin, 2022-09-05:
@AlCalzone coming back to you with `debug` logs. Same test as before: starting with factory reset Zooz Open Close XS Sensor (ZSE41) with firmware 1.10 and trying to add it to HA (I was using it successfully with HA for some time now). After adding the device I close and open the door once.

**Test 1**
Z-Wave JS 0.1.64
Driver Version: 9.4.0
Server Version: 1.20.0
Result: ✔️ **Device added to HA**
[zwave_js_adding_device_suceeded.log](https://github.com/zwave-js/node-zwave-js/files/9491984/zwave_js_adding_device_suceeded.log)
[zwave_js-success_node_diagnostics.json.txt](https://github.com/zwave-js/node-zwave-js/files/9491985/zwave_js-success_node_diagnostics.json.txt)

**Test 2**
Z-Wave JS 0.1.69
Driver Version: 10.0.3
Server Version: 1.22.1
Result: ❌ **Failed to add device to HA**
[zwave_js_adding_device_failed.log](https://github.com/zwave-js/node-zwave-js/files/9491993/zwave_js_adding_device_failed.log)
[zwave_js-failed_node_diagnostics.json.txt](https://github.com/zwave-js/node-zwave-js/files/9491994/zwave_js-failed_node_diagnostics.json.txt)

I have registered with Zooz and received latest firmware 1.30. Should I try to upgrade and do the test again?

AlCalzone, 2022-09-06:
@gralin I found something in your log that aborts the interview. However this is triggered by the device behaving strangely, which **could** be the result of a low battery:
```
2022-09-05T20:44:59.029Z DRIVER « [Node 021] [REQ] [BridgeApplicationCommand]
                                  │ RSSI: -91 dBm
                                  └─[Security2CCMessageEncapsulation]
                                    │ sequence number: 35
                                    └─[BatteryCCReport]
                                        level:  0
                                        is low: true
```

AlCalzone, 2022-09-06:
@labob246 yours is the same issue as epower's

bckmrk, 2022-09-06:
> > Is that usually needed after an update?
> 
> It isn't usually, but it happens sometimes. Maybe the stick was still blocked by something else.

Updated to 0.1.69, re-interviewed all Z-wave devices, re-plugged the stick. Now it's working. A little bit time-consuming, but thanks for your work and assistance. Greatly appreciated. :-)

gralin, 2022-09-06:
@AlCalzone thank you for analyzing my logs ❤️ 

>I found something buggy in your log that aborts the interview. However this is triggered by the device behaving strangely, which could be the result of a low battery

I saw that too and that's why... I replaced the battery with a new one before the tests! The old one was due to be replaced anyway, but maybe I was out of luck and the new battery was faulty which I didn't consider. Especially because I was able to add the device (with that battery) using old version of the driver, but not with new one.

Today I have updated Z-Wave JS addon in HA and performed the tests again using driver 10.0.4. This time I managed to add the device! 🎉 Here is the log:  [zwave_js.log](https://github.com/zwave-js/node-zwave-js/files/9500539/zwave_js.log) If I am not mistaken, I have replaced the battery again after this becuase I saw in HA that reported battery level is 0%.

Finally I have upgraded the firmware in ZSE41 to latest 1.30 and re-added the device. No problems in this case either. HEre is the log: [zwave_js (1).log](https://github.com/zwave-js/node-zwave-js/files/9500549/zwave_js.1.log)

m4dm4rtig4n, 2022-09-06:
Hello, since update to 7.X.X, all device in zwavejs2mqtt as disappeared :/
While in the logs I have just passed information about the nodes
I have try re-interviewed all Z-wave devices, reboot compute, But the node list is still empty.

Any idea why ?

AlCalzone, 2022-09-06:
@m4dm4rtig4n please read the first comment in this thread and supply the information I ask for there.

m4dm4rtig4n, 2022-09-06:
Start log available here :
https://pastebin.com/ckqpbJTA

I have found this error :
2022-09-06T21:54:36.280Z DRIVER   Could not check for config updates: Failed to download package information! (ZW0343)

jcustenborder, 2022-09-09:
I found this via this thread. [Missing Door State entity when adding Ecolink DWZWAVE25 sensors](https://community.home-assistant.io/t/missing-door-state-entity-when-adding-ecolink-dwzwave25-sensors/456395)

I have a DWZWAVE25 that I added a while ago and a few that are new. The new nodes are missing the state entity. I did notice that there is a difference in the nodes. I have attached a good and a bad node for comparison

[bad.txt](https://github.com/zwave-js/node-zwave-js/files/9531847/bad.txt)
[good.txt](https://github.com/zwave-js/node-zwave-js/files/9531848/good.txt)
[zwavejs_current.log](https://github.com/zwave-js/node-zwave-js/files/9531849/zwavejs_current.log)

gralin, 2022-09-09:
@jcustenborder I had exactly same effect but with different door sensor (Zooz). Did you try updating zwave-js to latest version and re-pairing those devices? It helped in my case eventually (I also needed to replace the battery but that was just a coincidence).

AlCalzone, 2022-09-09:
> re-pairing those devices

is usually not necessary. Re-interviewing is far easier.

AlCalzone, 2022-09-09:
@jcustenborder yours is a duplicate of https://github.com/zwave-js/node-zwave-js/issues/5027

jcustenborder, 2022-09-09:
@gralin Sorry I forgot to add I'm running `zwavejs/zwavejs2mqtt:7.1.0` of the docker container.

jcustenborder, 2022-09-09:
@AlCalzone  Thank you! Is there any information that would be helpful to add to #5027

AlCalzone, 2022-09-09:
Nope, the bug is known, I just need to fix it.

AlCalzone, 2022-09-09:
@epower53 Zooz have told me that the Zen16 bug is known and a firmware with a fix exists. You might want to reach out to them to get the update.

epower53, 2022-09-09:
Thanks @AlCalzone. Is this a different f/w than the latest release? I upgraded to the latest release, 1.04, last month. This is the version I was on when the bad behavior exposed by Z-wave JS 10.0 showed up.

AlCalzone, 2022-09-09:
Not sure actually, I guess you could just ask them and mention that the maintainer of Z-Wave JS told you about the update.

epower53, 2022-09-09:
Will do, thanks

epower53, 2022-09-12:
Zooz support provided me with a newer (beta?) firmware to try on my Zen16. I'll apply it sometime later this week and re-try Z-wave JS 10.0.X and report back.

finity69x2, 2022-09-13:
Is there a fix for this forthcoming?

It seems that we can't run v9.x of zwavejs along with v2022.9.x of HA. 

So an update of either one has poor results.

zenturacp, 2022-09-13:
> Is there a fix for this forthcoming?
> 
> 
> It seems that we can't run v9.x of zwavejs along with v2022.9.x of HA. 
> 
> 
> So an update of either one has poor results.

No you need to wait until there is a fix :-(

finity69x2, 2022-09-14:
I thought this was the one related to this not working on V10. Or did I not read the issue title or posts here correctly?

I could create another one but I don't think it will help if it's the same as this issue.

I was just wondering if this issue was close to being fixed yet.

AlCalzone, 2022-09-15:
@finity69x2 see the very first post. This issue is meant to collect information about what does not work and why. Just saying "it doesn't work" is not helpful.

zenturacp, 2022-09-18:
> It looks like there were some changes in v10, which break communication with some devices. We're still in the process of figuring things out and need some information to do so.
>
>
>
> ### If devices [...]

Do you need more tests? I can if needed make the tests with my fibaro units.. If you still havnt found the issue?

AlCalzone, 2022-09-19:
> If you still havnt found the issue?

Which one? There were several, some related, some unrelated. This thread is meant to bundle all this investigation into one, so I don't have to look at 10 duplicate issues every day.

I'm not sure what is and what isn't working for you, or if your issues (if you had/have any) were already fixed.

Anyways, not a lot has surfaced in the past week, so unless someone reports back here with something new, I'll close this thread soon. The remaining issues I identified here seem to be related to S2 polishing and have existed before v10.

Danymyte, 2022-09-19:
With the upgrade my Aeon Labs Garage Door Controller ZW062 is not working anymore. Trying to open or close the garage door results in no action at all. Tried to reinterview but without sucesss so far.

Just struggling how to collect all the data needed for bug fixing, can I get some advice?

Thanks a lot!

TheDK, 2022-09-19:
> Do you need more tests? I can if needed make the tests with my fibaro units.. If you still havnt found the issue?

I just checked the following devices against the current version with todays release of Z-Wave JS to MQTT:

- Fibaro Single / Double Switch
- Fibaro Dimmer 2
- Fibaro Roller Shutter 3
- Fibaro Relay
- Fibaro Multisensor (Motion Eye)
- Steinel Outdoor PIR

All seem to be working just fine.

zenturacp, 2022-09-19:
> I just checked the following devices against the current version with todays release of Z-Wave JS to MQTT:
>
> * Fibaro Dimmer 2
>
> All seem to be working just fine.

Okay - i'm using the Z-Wave JS addon in Homeassistant - and that breaks - is the change that went into Z-Wave JS to MQTT specfic for that? If that works then its great ..

AlCalzone, 2022-09-19:
@zenturacp the official addon was also bumped to contain the fix: https://github.com/home-assistant/addons/blob/master/zwave_js/CHANGELOG.md#0171

TheDK, 2022-09-19:
Just FYI: No interview necessary on my part, everything just kept working as ever (I checked switches, states, meter).

blommanx, 2022-09-19:
Hello, when upgrading to zwavejs 0.1.71 some of my Fibaro devices stops working (I didn't check all devices, but two Fibaro devices were the obvious ones).
They report the wrong state in Home Assistant and they don't respond to commands.
Reinterview on 71 did not help, rolling back to 0.1.65 is only way to fix it (had to reinterview on .65 and restart for them to come back).
Attached is logs from the Double Relay device
[zwave_js-diag-Double Switch 2.json.txt](https://github.com/zwave-js/node-zwave-js/files/9601231/zwave_js-58cd637ccb995fa3e52c9b69be6d0056-Double.Switch.2-e9ffc8d72c4dbc32ecd061585eaa963b.json.txt)
[zwave_js-interview.log](https://github.com/zwave-js/node-zwave-js/files/9601236/zwave_js.log)

[14 more replies]