zwave-js/zwave-js #4999: Termostat Heatit Z-TRM3: No entity created when the device is added

cobch, 2022-09-02:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within ZWaveJS2MQTT?

NO, my problem is NOT within ZWaveJS2MQTT

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

I am using Aeotec Z-Stick 7 with ZwaveJs addon on Home Assistant. I managed to add the device but not any entities is created and the node is described as not ready.
I have tried to heal and reinterview the device but it does not help. To exclude and include again but same problem, no entities coming with the device.
[image]

### Device information

Manufacturer: Heatit
Model name: Z-TRM3
Node ID in your network: <4>

### How are you using `node-zwave-js`?

- [ ] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [X] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version: zwave-js: 0.1.68

### Did you change anything?

no

### Did this work before?

Don't know, this is a new device

### Attach Driver Logfile

│ source node id:   1
                                  │ transmit options: 0x24
                                  │ callback id:      0
                                  └─[Security2CCMessageEncapsulation]
                                    │ sequence number: 1
                                    └─[MultiChannelCCCommandEncapsulation]
                                      │ source:      0
                                      │ destination: 2
                                      └─[SupervisionCCReport]
                                          session id:          20
                                          more updates follow: false
                                          status:              Success
                                          duration:            0s
2022-09-02T12:20:37.853Z SERIAL « [ACK]                                                                   (0x06)
2022-09-02T12:20:37.855Z SERIAL « 0x010401a90152                                                       (6 bytes)
2022-09-02T12:20:37.856Z SERIAL » [ACK]                                                                   (0x06)
2022-09-02T12:20:37.858Z DRIVER « [RES] [SendDataBridge]
                                    was sent: true
2022-09-02T12:20:37.888Z SERIAL « 0x012300a80001041a9f03e40086615a51573e6dba4c5e2ab4fe60bdeb74a43e772 (37 bytes)
                                  c7700a96e
2022-09-02T12:20:37.891Z SERIAL » [ACK]                                                                   (0x06)
2022-09-02T12:20:37.897Z CNTRLR   [Node 004] [Multilevel Sensor] Air temperature: metadata updated  [Endpoint 3]
2022-09-02T12:20:37.898Z CNTRLR   [Node 004] [~] [Multilevel Sensor] Air temperature: 0 => 0        [Endpoint 3]
2022-09-02T12:20:37.900Z DRIVER « [Node 004] [REQ] [BridgeApplicationCommand]
                                  │ RSSI: -87 dBm
                                  └─[Security2CCMessageEncapsulation]
                                    │ sequence number: 228
                                    └─[MultiChannelCCCommandEncapsulation]
                                      │ source:      3
                                      │ destination: 0
                                      └─[SupervisionCCGet]
                                        │ session id:      21
                                        │ request updates: false
                                        └─[MultilevelSensorCCReport]
                                            type:  Air temperature
                                            scale: Celsius
                                            value: 0
2022-09-02T12:20:37.913Z SERIAL » 0x012100a90104159f03020094b187239f34689212953deee4c6d500bd240000000 (35 bytes)
                                  00013
2022-09-02T12:20:37.915Z DRIVER » [Node 004] [REQ] [SendDataBridge]
                                  │ source node id:   1
                                  │ transmit options: 0x24
                                  │ callback id:      0
                                  └─[Security2CCMessageEncapsulation]
                                    │ sequence number: 2
                                    └─[MultiChannelCCCommandEncapsulation]
                                      │ source:      0
                                      │ destination: 3
                                      └─[SupervisionCCReport]
                                          session id:          21
                                          more updates follow: false
                                          status:              Success
                                          duration:            0s
2022-09-02T12:20:37.921Z SERIAL « [ACK]                                                                   (0x06)
2022-09-02T12:20:37.923Z SERIAL « 0x010401a90152                                                       (6 bytes)
2022-09-02T12:20:37.926Z SERIAL » [ACK]                                                                   (0x06)
2022-09-02T12:20:37.927Z DRIVER « [RES] [SendDataBridge]
                                    was sent: true
2022-09-02T12:20:37.957Z SERIAL « 0x012300a80001041a9f03e500f5ccdebbc25b515d7dbd837188000a1ac6df6785c (37 bytes)
                                  fc300aa2c
2022-09-02T12:20:37.961Z SERIAL » [ACK]                                                                   (0x06)
2022-09-02T12:20:37.965Z CNTRLR   [Node 004] [Multilevel Sensor] Air temperature: metadata updated  [Endpoint 4]
2022-09-02T12:20:37.966Z CNTRLR   [Node 004] [~] [Multilevel Sensor] Air temperature: 24.3 => 24.3  [Endpoint 4]
2022-09-02T12:20:37.969Z DRIVER « [Node 004] [REQ] [BridgeApplicationCommand]
                                  │ RSSI: -86 dBm
                                  └─[Security2CCMessageEncapsulation]
                                    │ sequence number: 229
                                    └─[MultiChannelCCCommandEncapsulation]
                                      │ source:      4
                                      │ destination: 0
                                      └─[SupervisionCCGet]
                                        │ session id:      22
                                        │ request updates: false
                                        └─[MultilevelSensorCCReport]
                                            type:  Air temperature
                                            scale: Celsius
                                            value: 24.3
2022-09-02T12:20:37.986Z SERIAL » 0x012100a90104159f0303008fdf04cf8a1901c37d6ab5263513d71195240000000 (35 bytes)
                                  000e7
2022-09-02T12:20:37.988Z DRIVER » [Node 004] [REQ] [SendDataBridge]
                                  │ source node id:   1
                                  │ transmit options: 0x24
                                  │ callback id:      0
                                  └─[Security2CCMessageEncapsulation]
                                    │ sequence number: 3
                                    └─[MultiChannelCCCommandEncapsulation]
                                      │ source:      0
                                      │ destination: 4
                                      └─[SupervisionCCReport]
                                          session id:          22
                                          more updates follow: false
                                          status:              Success
                                          duration:            0s
2022-09-02T12:20:37.994Z SERIAL « [ACK]                                                                   (0x06)
2022-09-02T12:20:37.996Z SERIAL « 0x010401a90152                                                       (6 bytes)
2022-09-02T12:20:37.998Z SERIAL » [ACK]                                                                   (0x06)
2022-09-02T12:20:37.999Z DRIVER « [RES] [SendDataBridge]
                                    was sent: true
Starting logging event forwarder at debug level
Stopping logging event forwarder

AlCalzone, 2022-09-02:
@cobch please make a log (loglevel debug) of the **entire** interview as described [here](https://www.home-assistant.io/integrations/zwave_js#how-to-access-the-z-wave-logs).
When done, download the log and **attach** it here by dragging and dropping into the textbox.

cobch, 2022-09-02:
Sorry, I am not really good with Zwave since I started 2 days ago and nothing seems to work (-:
Here come the log when I pressed the button to increase and decrease the temperatur in the termostat.
[zwave_js.log](https://github.com/zwave-js/node-zwave-js/files/9478041/zwave_js.log)

AlCalzone, 2022-09-02:
That is not the interview, those are just temperature measurements from the thermostat. Click on the 3 dots next to "Configure Device" and select Re-Interview **while** you have the log window open.

cobch, 2022-09-02:
Here it comes. So sorry about the wrong log before...
[zwave_js (re-interview).log](https://github.com/zwave-js/node-zwave-js/files/9479201/zwave_js.re-interview.log)

AlCalzone, 2022-09-02:
Thanks, found the problem!

cobch, 2022-09-02:
Great to know but you keep the secret totally (-: Is it something I should fix by myself?

perosb, 2022-09-02:
I was just about to report the same issue.
Not sure it's the same. My log is attached.

`[Node 124] Error during node interview: Node #124 (Endpoint 2) does not support the command GetSupportedSensor (ZW0302)`
Let me know if I should open a new issue or provide diferent logs.

[trm3.log](https://github.com/zwave-js/node-zwave-js/files/9479931/trm3.log)

AlCalzone, 2022-09-02:
Yeah I'll fix it on Monday and release a new version.

suneworld, 2022-09-05:
I am experiencing the same problem. I have upgraded to 0.1.69 but it did not fix the issue.
I would at the same time say that I am very greatful for all the fantastic work on this addon and that I am absolutely waiting patiently for a fix :)

perosb, 2022-09-05:
It's not released yet.
However, it worked for me in a version using nodejs-zwave < 10.

suneworld, 2022-09-05:
> It's not released yet. However, it worked for me in a version using nodejs-zwave < 10.

It is possible to downgrade? If yes, how?

danielfinans, 2022-09-06:
I am having the same problem as well. Let me know if there is any additional logs needed. Looking forward to testing if new version release will fix this.

suneworld, 2022-09-06:
A rollback to version  0.1.65 fixed it for me. Looking forward to a fix in the current version :)

AlCalzone, 2022-09-06:
Yeah I haven't released the fix yet because a few other bugs were identified that I also want to eliminate in one go. Otherwise this will just cause tons of comments like "still does not work after updating".

AlCalzone, 2022-09-06:
Try 0.1.70 please. Again a re-interview may be required to get devices to work.

perosb, 2022-09-06:
10.0.4 works like a charm!

Thanks 🙏

cobch, 2022-09-06:
I get 9 entities after upgraded to latest version so I guess it is working too (-:
Should I close the issue?

AlCalzone, 2022-09-06:
Fine be me. If you don't mind, another interview log now that it is working would be nice.

cobch, 2022-09-06:
I am again with my work but will do it as soon as I can (-:

AlCalzone, 2022-09-07:
Thanks. Not 100% in terms of handling S2-related collisions, but that's a separate issue.

closed 2022-09-07 by AlCalzone

cobch, 2022-09-07:
> [trm3.log](https://github.com/zwave-js/node-zwave-js/files/9498034/trm3.log)
> My log attached

How many entities could you see? I have only 9 and do not find how to remote shut on/off heating floor )-:

cobch, 2022-09-08:
@perosb How does it look on your side? I have not the on/off controller entity.
[image]

perosb, 2022-09-08:
@cobch Not on the device itself but if you add a thermostat card to your views
[image]

if you click the "Floor thermostat" entity you get a dropdown with control.
[image]

cobch, 2022-09-15:
Thanks/Tack!