zwave-js/zwave-js #5024: AeonStick getting error on Windows 11

CristianMori, 2022-09-07:
Hello. I am running zwave-js server on a win 10 machine, and everything works fine.
I am moving to a windows 11 and when running zwave-server I get the following output

00:14:50.880 DRIVER   ███████╗ ██╗    ██╗  █████╗  ██╗   ██╗ ███████╗             ██╗ ███████╗
                      ╚══███╔╝ ██║    ██║ ██╔══██╗ ██║   ██║ ██╔════╝             ██║ ██╔════╝
                        ███╔╝  ██║ █╗ ██║ ███████║ ██║   ██║ █████╗   █████╗      ██║ ███████╗
                       ███╔╝   ██║███╗██║ ██╔══██║ ╚██╗ ██╔╝ ██╔══╝   ╚════╝ ██   ██║ ╚════██║
                      ███████╗ ╚███╔███╔╝ ██║  ██║  ╚████╔╝  ███████╗        ╚█████╔╝ ███████║
                      ╚══════╝  ╚══╝╚══╝  ╚═╝  ╚═╝   ╚═══╝   ╚══════╝         ╚════╝  ╚══════╝
00:14:50.885 DRIVER   version 10.0.4
00:14:50.886 DRIVER
00:14:50.887 DRIVER   starting driver...
00:14:50.905 DRIVER   opening serial port COM6
00:14:50.918 DRIVER   serial port opened
00:14:50.920 SERIAL » [NAK]                                                                   (0x15)
00:14:50.923 DRIVER   loading configuration...
00:14:50.928 CONFIG   version 10.0.4
00:14:51.793 DRIVER   beginning interview...
00:14:51.795 DRIVER   added request handler for AddNodeToNetwork (0x4a)...
                      1 registered
00:14:51.796 DRIVER   added request handler for RemoveNodeFromNetwork (0x4b)...
                      1 registered
00:14:51.797 DRIVER   added request handler for ReplaceFailedNode (0x63)...
                      1 registered
00:14:51.798 CNTRLR   querying controller IDs...
00:14:51.946 SERIAL » 0x01030020dc                                                         (5 bytes)
00:14:51.950 DRIVER » [REQ] [GetControllerId]
00:14:51.953 SERIAL « [ACK]                                                                   (0x06)
00:14:51.960 SERIAL « 0x01080120fa1cf99e0156                                              (10 bytes)
00:14:51.962 SERIAL » [ACK]                                                                   (0x06)
00:14:51.965 DRIVER « [RES] [GetControllerId]
                        home ID:     0xfa1cf99e
                        own node ID: 1
00:14:51.968 DRIVER « [RES] [GetControllerId]                                           [unexpected]
                        home ID:     0xfa1cf99e
                        own node ID: 1
00:14:51.969 DRIVER   unexpected response, discarding...
00:14:52.965 CNTRLR   Failed to execute controller command after 1/3 attempts. Scheduling next try i
                      n 100 ms.
00:14:53.071 SERIAL » 0x01030020dc                                                         (5 bytes)
00:14:53.074 DRIVER » [REQ] [GetControllerId]
00:14:53.079 SERIAL « [ACK]                                                                   (0x06)
00:14:53.088 SERIAL « 0x01080120fa1cf99e0156                                              (10 bytes)
00:14:53.091 SERIAL » [ACK]                                                                   (0x06)
00:14:53.094 DRIVER « [RES] [GetControllerId]
                        home ID:     0xfa1cf99e
                        own node ID: 1
00:14:53.101 DRIVER « [RES] [GetControllerId]                                           [unexpected]
                        home ID:     0xfa1cf99e
                        own node ID: 1
00:14:53.103 DRIVER   unexpected response, discarding...
00:14:54.095 CNTRLR   Failed to execute controller command after 2/3 attempts. Scheduling next try i
                      n 1100 ms.
00:14:55.209 SERIAL » 0x01030020dc                                                         (5 bytes)
00:14:55.210 DRIVER » [REQ] [GetControllerId]
00:14:55.213 SERIAL « [ACK]                                                                   (0x06)
00:14:55.216 SERIAL « 0x01080120fa1cf99e0156                                              (10 bytes)
00:14:55.218 SERIAL » [ACK]                                                                   (0x06)
00:14:55.221 DRIVER « [RES] [GetControllerId]
                        home ID:     0xfa1cf99e
                        own node ID: 1
00:14:55.224 DRIVER « [RES] [GetControllerId]                                           [unexpected]
                        home ID:     0xfa1cf99e
                        own node ID: 1
00:14:55.225 DRIVER   unexpected response, discarding...
00:14:56.247 DRIVER   Failed to initialize the driver: ZWaveError: Timeout while waiting for an ACK
                      from the controller (ZW0200)
                          at Driver.sendMessage (C:\Users\Cristian\node_modules\zwave-js\src\lib\dri
                      ver\Driver.ts:3895:23)
                          at ZWaveController.identify (C:\Users\Cristian\node_modules\zwave-js\src\l
                      ib\controller\Controller.ts:713:33)
                          at Driver.initializeControllerAndNodes (C:\Users\Cristian\node_modules\zwa
                      ve-js\src\lib\driver\Driver.ts:1186:26)
                          at Immediate.<anonymous> (C:\Users\Cristian\node_modules\zwave-js\src\lib\
                      driver\Driver.ts:993:16)

I tried with two different zwave stick a gen5 and an old S2. Both works fine on win 10 and have the same issue on windows 11. (same node version)

AlCalzone, 2022-09-07:
To be honest, I'm not quite sure why this is erroring. Can you share the first couple of lines from Windows 10, so I can compare?

CristianMori, 2022-09-07:
Sure thing. Here it is

02:58:13.551 DRIVER   ███████╗ ██╗    ██╗  █████╗  ██╗   ██╗ ███████╗             ██╗ ███████╗
                      ╚══███╔╝ ██║    ██║ ██╔══██╗ ██║   ██║ ██╔════╝             ██║ ██╔════╝
                        ███╔╝  ██║ █╗ ██║ ███████║ ██║   ██║ █████╗   █████╗      ██║ ███████╗
                       ███╔╝   ██║███╗██║ ██╔══██║ ╚██╗ ██╔╝ ██╔══╝   ╚════╝ ██   ██║ ╚════██║
                      ███████╗ ╚███╔███╔╝ ██║  ██║  ╚████╔╝  ███████╗        ╚█████╔╝ ███████║
                      ╚══════╝  ╚══╝╚══╝  ╚═╝  ╚═╝   ╚═══╝   ╚══════╝         ╚════╝  ╚══════╝
02:58:13.554 DRIVER   version 10.0.4
02:58:13.555 DRIVER
02:58:13.555 DRIVER   starting driver...
02:58:13.566 DRIVER   opening serial port COM3
02:58:14.578 DRIVER   serial port opened
02:58:14.579 SERIAL » [NAK]                                                                   (0x15)
02:58:14.582 DRIVER   loading configuration...
02:58:14.585 CONFIG   version 10.0.4
02:58:15.515 DRIVER   beginning interview...
02:58:15.517 DRIVER   added request handler for AddNodeToNetwork (0x4a)...
                      1 registered
02:58:15.518 DRIVER   added request handler for RemoveNodeFromNetwork (0x4b)...
                      1 registered
02:58:15.519 DRIVER   added request handler for ReplaceFailedNode (0x63)...
                      1 registered
02:58:15.519 CNTRLR   querying controller IDs...
02:58:15.615 SERIAL » 0x01030020dc                                                         (5 bytes)
02:58:15.617 DRIVER » [REQ] [GetControllerId]
02:58:15.620 SERIAL « [ACK]                                                                   (0x06)
02:58:15.622 SERIAL « 0x01080120fa1cf99e0156                                              (10 bytes)
02:58:15.624 SERIAL » [ACK]                                                                   (0x06)
02:58:15.625 DRIVER « [RES] [GetControllerId]
                        home ID:     0xfa1cf99e
                        own node ID: 1
02:58:15.632 CNTRLR   received controller IDs:
                        home ID:     0xfa1cf99e
                        own node ID: 1
02:58:15.633 CNTRLR   querying API capabilities...
02:58:15.638 SERIAL » 0x01030007fb                                                         (5 bytes)
02:58:15.638 DRIVER » [REQ] [GetSerialApiCapabilities]
02:58:15.639 SERIAL « [ACK]                                                                   (0x06)
02:58:15.644 SERIAL « 0x012b0107010000860101005afe81ff884f1f0000fb9f7da067000080008086000 (45 bytes)
                      000e87300000e0000600000fa
02:58:15.645 SERIAL » [ACK]                                                                   (0x06)
02:58:15.652 DRIVER « [RES] [GetSerialApiCapabilities]
                        payload: 0x010000860101005afe81ff884f1f0000fb9f7da067000080008086000000e8730
                      0000e0000600000
02:58:15.656 CNTRLR   received API capabilities:
                        firmware version:    1.0
                        manufacturer ID:     0x86
                        product type:        0x0101
                        product ID:          0x5a
                        supported functions:
                        · GetSerialApiInitData (0x02)
                        · SetApplicationNodeInformation (0x03)
                        · ApplicationCommand (0x04)
                        · GetControllerCapabilities (0x05)
                        · SetSerialApiTimeouts (0x06)
                        · GetSerialApiCapabilities (0x07)
                        · SoftReset (0x08)
                        · GetProtocolVersion (0x09)
                        · SetRFReceiveMode (0x10)
                        · UNKNOWN_FUNC_SET_SLEEP_MODE (0x11)
                        · FUNC_ID_ZW_SEND_NODE_INFORMATION (0x12)
                        · SendData (0x13)
                        · SendDataMulticast (0x14)
                        · GetControllerVersion (0x15)
                        · SendDataAbort (0x16)
                        · FUNC_ID_ZW_R_F_POWER_LEVEL_SET (0x17)
                        · UNKNOWN_FUNC_SEND_DATA_META (0x18)
                        · FUNC_ID_ZW_GET_RANDOM (0x1c)
                        · GetControllerId (0x20)
                        · UNKNOWN_FUNC_MEMORY_GET_BYTE (0x21)
                        · UNKNOWN_FUNC_MEMORY_PUT_BYTE (0x22)
                        · UNKNOWN_FUNC_MEMORY_GET_BUFFER (0x23)
                        · UNKNOWN_FUNC_MEMORY_PUT_BUFFER (0x24)
                        · UNKNOWN_FUNC_FlashAutoProgSet (0x27)
                        · GetNVMId (0x29)
                        · ExtNVMReadLongBuffer (0x2a)
                        · ExtNVMWriteLongBuffer (0x2b)
                        · ExtNVMReadLongByte (0x2c)
                        · ExtExtWriteLongByte (0x2d)
                        · GetNodeProtocolInfo (0x41)
                        · HardReset (0x42)`

AlCalzone, 2022-09-07:
Are you running the server natively using Node.js or somehow packaged with Docker or whatever?
If natively, which version of Node.js (`node -v`)?

CristianMori, 2022-09-07:
Natively. 16.17.0

AlCalzone, 2022-09-07:
Ok, what's the output of this when you're in the server dir?
```
npm ls --all
```

CristianMori, 2022-09-08:
on Windows 11
+-- @zwave-js/server@1.22.1
| +-- @homebridge/ciao@1.1.5
| | +-- debug@4.3.4
| | | `-- ms@2.1.2
| | +-- fast-deep-equal@3.1.3
| | +-- source-map-support@0.5.21 deduped
| | `-- tslib@2.4.0
| +-- minimist@1.2.6
| +-- ws@8.8.1
| | +-- UNMET OPTIONAL DEPENDENCY bufferutil@^4.0.1
| | `-- UNMET OPTIONAL DEPENDENCY utf-8-validate@^5.0.2
| `-- zwave-js@10.0.4 deduped
`-- zwave-js@10.0.4
  +-- @alcalzone/jsonl-db@2.5.3
  | +-- @alcalzone/proper-lockfile@4.1.3-0
  | | +-- graceful-fs@4.2.10 deduped
  | | +-- retry@0.12.0 deduped
  | | `-- signal-exit@3.0.7 deduped
  | +-- alcalzone-shared@4.0.3 deduped
  | `-- fs-extra@10.1.0 deduped
  +-- @alcalzone/pak@0.8.1
  | +-- axios@0.26.1
  | | `-- follow-redirects@1.15.1 deduped
  | +-- execa@5.1.1 deduped
  | `-- fs-extra@10.1.0 deduped
  +-- @sentry/integrations@7.12.1
  | +-- @sentry/types@7.12.1
  | +-- @sentry/utils@7.12.1
  | | +-- @sentry/types@7.12.1 deduped
  | | `-- tslib@1.14.1 deduped
  | +-- localforage@1.10.0
  | | `-- lie@3.1.1
  | |   `-- immediate@3.0.6
  | `-- tslib@1.14.1
  +-- @sentry/node@7.12.1
  | +-- @sentry/core@7.12.1
  | | +-- @sentry/hub@7.12.1 deduped
  | | +-- @sentry/types@7.12.1 deduped
  | | +-- @sentry/utils@7.12.1 deduped
  | | `-- tslib@1.14.1 deduped
  | +-- @sentry/hub@7.12.1
  | | +-- @sentry/types@7.12.1 deduped
  | | +-- @sentry/utils@7.12.1 deduped
  | | `-- tslib@1.14.1 deduped
  | +-- @sentry/types@7.12.1 deduped
  | +-- @sentry/utils@7.12.1 deduped
  | +-- cookie@0.4.2
  | +-- https-proxy-agent@5.0.1
  | | +-- agent-base@6.0.2
  | | | `-- debug@4.3.4 deduped
  | | `-- debug@4.3.4 deduped
  | +-- lru_map@0.3.3
  | `-- tslib@1.14.1 deduped
  +-- @zwave-js/cc@10.0.4
  | +-- @zwave-js/core@10.0.4 deduped
  | +-- @zwave-js/host@10.0.4 deduped
  | +-- @zwave-js/serial@10.0.4 deduped
  | +-- @zwave-js/shared@10.0.4 deduped
  | +-- alcalzone-shared@4.0.3 deduped
  | +-- ansi-colors@4.1.3 deduped
  | `-- reflect-metadata@0.1.13 deduped
  +-- @zwave-js/config@10.0.4
  | +-- @zwave-js/core@10.0.4 deduped
  | +-- @zwave-js/shared@10.0.4 deduped
  | +-- alcalzone-shared@4.0.3 deduped
  | +-- ansi-colors@4.1.3 deduped
  | +-- fs-extra@10.1.0 deduped
  | +-- json-logic-js@2.0.2
  | +-- json5@2.2.1
[... 152 lines cut ...]
  | | | +-- color@3.2.1
  | | | | +-- color-convert@1.9.3
  | | | | | `-- color-name@1.1.3
  | | | | `-- color-string@1.9.1
  | | | |   +-- color-name@1.1.3 deduped
  | | | |   `-- simple-swizzle@0.2.2
  | | | |     `-- is-arrayish@0.3.2
  | | | `-- text-hex@1.0.0
  | | +-- enabled@2.0.0
  | | `-- kuler@2.0.0
  | +-- async@3.2.4
  | +-- is-stream@2.0.1 deduped
  | +-- logform@2.4.2 deduped
  | +-- one-time@1.0.0
  | | `-- fn.name@1.1.0
  | +-- readable-stream@3.6.0
  | | +-- inherits@2.0.4
  | | +-- string_decoder@1.3.0
  | | | `-- safe-buffer@5.2.1
  | | `-- util-deprecate@1.0.2
  | +-- safe-stable-stringify@2.3.1
  | +-- stack-trace@0.0.10
  | +-- triple-beam@1.3.0 deduped
  | `-- winston-transport@4.5.0 deduped
  `-- xstate@4.33.5

CristianMori, 2022-09-08:
Output from Windows 10 machine
[win10.txt](https://github.com/zwave-js/node-zwave-js/files/9523388/win10.txt)

AlCalzone, 2022-09-08:
That doesn't look wrong either...
The stick's ACK definitely gets received, but this info somehow doesn't find its way to the state machine. Aside from actually debugging on a Windows 11 machine (I only have 10), I have honestly no idea how to proceed.

CristianMori, 2022-09-09:
It looks like somehow the same message is processed twice
00:14:53.094 DRIVER « [RES] [GetControllerId]
                        home ID:     0xfa1cf99e
                        own node ID: 1
00:14:53.101 DRIVER « [RES] [GetControllerId]                                           [unexpected]
                        home ID:     0xfa1cf99e
                        own node ID: 1
00:14:53.103 DRIVER   unexpected response, discarding...

If you need me to run any test code on that setup, happy to. In the meanwhile I moved to an RPI and had no problems

AlCalzone, 2022-09-09:
It's not processed twice, just logged twice (once on receipt, then again because it is unexpected).

ptorrent, 2023-01-09:
Hello,

Same error here on windows 10 

To reproduce:

1)driver.controller.beginInclusion()
2)Disconnect the USB
3)Reconnect the USB

Even if I remove the usb key ZW200 and replug it.

Thanks for your help !

jriker1, 2023-02-02:
I posted an issue in the HA forum like this and was referenced to this issue.

https://community.home-assistant.io/t/failed-to-execute-controller-command-after-1-3-attempts-scheduling-next-try/528787/9

Apparently the same except or me running on Windows 2022 Server.

Log: https://www.dropbox.com/s/12rio3prde2y7ik/zwavejs_2023-02-01.log?dl=0

StreetCube, 2023-04-27:
I also had this error, I'm gonna check it out tomorrow, and get back to you with the information

StreetCube, 2023-04-28:
It seems to be still wrong for me, same issue.

bostjane, 2023-05-03:
I haven't encountered this issue since I started using Zwave JS UI, that is few months now. However, latest version 8.14.2 produces this error on two different Windows 10 machines, both have different USB sticks, different ZWave networks.

I have no clue what else could do the difference. As a cause for error I would exclude Windows Updates, other system components and also Zwave store folder and saved stuff in it as simply running previous version (8.14.1) in the same location makes the problem disappear.

Could some extra logging info be added to track the error in details?

AlCalzone, 2023-05-03:
I just tried again. I can reproduce. Sometimes...

Attempt 1 was fine.

Attempt 2:
```
12:27:14.359 CNTRLR   querying controller IDs...
12:27:14.417 SERIAL » 0x01030020dc                                                         (5 bytes)
12:27:14.419 DRIVER » [REQ] [GetControllerId]
12:27:14.425 SERIAL « [ACK]                                                                   (0x06)
12:27:14.427 SERIAL « 0x01080120deab0b21028b                                              (10 bytes)
12:27:14.428 SERIAL » [ACK]                                                                   (0x06)
12:27:14.430 DRIVER « [RES] [GetControllerId]
                        home ID:     0xdeab0b21
                        own node ID: 2
12:27:14.432 DRIVER « [RES] [GetControllerId]                                           [unexpected]
                        home ID:     0xdeab0b21
                        own node ID: 2
12:27:14.433 DRIVER   unexpected response, discarding...
12:27:15.437 CNTRLR   Failed to execute controller command after 1/3 attempts. Scheduling next try i
                      n 100 ms.
12:27:15.546 SERIAL » 0x01030020dc                                                         (5 bytes)
12:27:15.547 DRIVER » [REQ] [GetControllerId]
12:27:15.552 SERIAL « [ACK]                                                                   (0x06)
12:27:15.553 SERIAL « 0x01080120deab0b21028b                                              (10 bytes)
12:27:15.554 SERIAL » [ACK]                                                                   (0x06)
12:27:15.555 DRIVER « [RES] [GetControllerId]
                        home ID:     0xdeab0b21
                        own node ID: 2
12:27:15.559 CNTRLR   received controller IDs:
                        home ID:     0xdeab0b21
                        own node ID: 2
12:27:15.559 CNTRLR   querying API capabilities...
```
here it works after the retry.

Attempt 3 without issues at first:
```
12:27:27.896 CNTRLR   querying controller IDs...
12:27:27.953 SERIAL » 0x01030020dc                                                         (5 bytes)
12:27:27.955 DRIVER » [REQ] [GetControllerId]
12:27:27.961 SERIAL « [ACK]                                                                   (0x06)
12:27:27.963 SERIAL « 0x01080120deab0b21028b                                              (10 bytes)
12:27:27.964 SERIAL » [ACK]                                                                   (0x06)
12:27:27.966 DRIVER « [RES] [GetControllerId]
                        home ID:     0xdeab0b21
                        own node ID: 2
12:27:27.970 CNTRLR   received controller IDs:
                        home ID:     0xdeab0b21
                        own node ID: 2
12:27:27.970 CNTRLR   querying API capabilities...
```
but a later command shows the same issue
```
12:27:28.462 DRIVER » [REQ] [SerialAPISetup]
                        command: GetPowerlevel16Bit
                        payload: 0x13
...
12:27:28.474 DRIVER « [RES] [SerialAPISetup]
                        command:               GetPowerlevel16Bit
                        normal powerlevel:     0.0 dBm
                        output power at 0 dBm: 3.3 dBm
12:27:28.475 DRIVER « [RES] [SerialAPISetup]                                            [unexpected]
                        command:               GetPowerlevel16Bit
                        normal powerlevel:     0.0 dBm
                        output power at 0 dBm: 3.3 dBm
12:27:28.475 DRIVER   unexpected response, discarding...
```

AlCalzone, 2023-05-03:
Good news, I think I finally figured out what causes this. A fix (hopefully) is up and will be merged soon.

PR #5737 (2023-05-03): fix: decouple reading the serialport from handling the data
fixes: #5024

It seems that serialport v10 on Windows behaves differently than on Linux and older serialport versions on Windows - possibly lagging the event loop or something similar. After the upgrade, the state machine was sometimes still in the old state and hasn't progressed yet. This caused some message tests to use the old state and fail / consider the message unsolicited/unexpected.

After delaying the data handling using `setImmediate`, I can no longer reproduce the issue - so I hope this is actually fixed now.
closed 2023-05-03 by zwave-js-bot