zwave-js/zwave-js #5025: S2: Ignore Security2CCNonceGet when outgoing command with Sender EI is pending

AlCalzone, 2022-09-07:
```
2022-09-06T14:32:02.080Z DRIVER « [Node 124] [REQ] [ApplicationCommand]
                                  └─[Security2CCNonceReport]
                                      sequence number:  63
                                      SOS:              true
                                      MOS:              false
                                      receiver entropy: 0x6f5765e3b482c803beedb6c1afd94d62
2022-09-06T14:32:02.083Z CNTRLR   [Node 124] failed to decode the message, retrying with SPAN extension...
2022-09-06T14:32:02.093Z SERIAL » 0x012900137c229f030d01124104f992956bacc50eb4d2a5999eba66a37efb933fd (43 bytes)
                                  8b8aeee6be0457125ca4c
2022-09-06T14:32:02.095Z DRIVER » [Node 124] [REQ] [SendData]
                                  │ transmit options: 0x25
                                  │ callback id:      202
                                  └─[Security2CCMessageEncapsulation]
                                    │ sequence number: 13
                                    │ extensions:
                                    │ · type: SPAN
                                    │   sender EI: 0x04f992956bacc50eb4d2a5999eba66a3
                                    └─[ConfigurationCCInfoGet]
                                        parameter #: 7
2022-09-06T14:32:02.098Z SERIAL « 0x01090004007c039f014053                                            (11 bytes)
2022-09-06T14:32:02.101Z SERIAL » [ACK]                                                                   (0x06)
2022-09-06T14:32:02.101Z SERIAL « [CAN]                                                                   (0x18)
2022-09-06T14:32:02.103Z CNTRLR   Failed to execute controller command after 1/3 attempts. Scheduling next try i
                                  n 100 ms.
2022-09-06T14:32:02.107Z DRIVER « [Node 124] [REQ] [ApplicationCommand]
                                  └─[Security2CCNonceGet]
                                      sequence number: 64
2022-09-06T14:32:02.205Z SERIAL » 0x012900137c229f030d01124104f992956bacc50eb4d2a5999eba66a37efb933fd (43 bytes)
                                  8b8aeee6be0457125ca4c
2022-09-06T14:32:02.207Z DRIVER » [Node 124] [REQ] [SendData]
                                  │ transmit options: 0x25
                                  │ callback id:      202
                                  └─[Security2CCMessageEncapsulation]
                                    │ sequence number: 13
                                    │ extensions:
                                    │ · type: SPAN
                                    │   sender EI: 0x04f992956bacc50eb4d2a5999eba66a3
                                    └─[ConfigurationCCInfoGet]
                                        parameter #: 7
2022-09-06T14:32:02.209Z SERIAL « [ACK]                                                                   (0x06)
2022-09-06T14:32:02.216Z SERIAL « 0x0104011301e8                                                       (6 bytes)
2022-09-06T14:32:02.217Z SERIAL » [ACK]                                                                   (0x06)
2022-09-06T14:32:02.219Z DRIVER « [RES] [SendData]
                                    was sent: true
2022-09-06T14:32:02.235Z SERIAL « 0x01070013ca00000322                                                 (9 bytes)
2022-09-06T14:32:02.236Z SERIAL » [ACK]                                                                   (0x06)
2022-09-06T14:32:02.238Z DRIVER « [REQ] [SendData]
                                    callback id:     202
                                    transmit status: OK
2022-09-06T14:32:02.245Z SERIAL » 0x011b00137c149f020e0185f64adcf6901ae503ea6eeccf9b6dcd05cb20        (29 bytes)
2022-09-06T14:32:02.247Z DRIVER » [Node 124] [REQ] [SendData]
                                  │ transmit options: 0x05
                                  │ callback id:      203
                                  └─[Security2CCNonceReport]
                                      sequence number:  14
                                      SOS:              true
                                      MOS:              false
                                      receiver entropy: 0x85f64adcf6901ae503ea6eeccf9b6dcd
```

gminco, 2022-09-08:
I'm not sure if it's the same issue, but just in case it helps.
I'm running v0.46.1 and while trying to add a couple of ZSE41 (door sensors) I keep running into failed interviews (re-interviewing too) while performing the inclusion right next to the controller (a ZST10-700).

I've added these devices before (I have 20+) and while I had to re-interview, or "exclude / factory reset / include", it has been been this bad (I literally tried 20 times today).
The only way I could end up adding the devices was using S0.

```
2022-09-07T14:03:31.886Z SERIAL Â» [ACK]                                                                   (0x06)
2022-09-07T14:03:31.887Z SERIAL Â« [CAN]                                                                   (0x18)
2022-09-07T14:03:31.891Z CNTRLR   Failed to execute controller command after 1/3 attempts. Scheduling next try i
                                  n 100 ms.
```

[zwave_js.log](https://github.com/zwave-js/node-zwave-js/files/9523040/zwave_js.log)

AlCalzone, 2022-09-08:
Similar, but not entirely the same --> https://github.com/zwave-js/node-zwave-js/issues/5028