zwave-js/zwave-js #5027: Creating metadata for multiple notification events of the same notification type overrides the old metadata instead of extending

AlCalzone, 2022-09-07:
For example, this should have states for 22 and 23:
```
{"k":"{\"nodeId\":7,\"commandClass\":113,\"endpoint\":0,\"property\":\"Access Control\",\"propertyKey\":\"Door state\"}","v":{"type":"number","readable":true,"writeable":false,"min":0,"max":255,"label":"Door state","states":{"23":"Window/door is closed"},"ccSpecific":{"notificationType":6}}}
```

jdiegmueller, 2022-09-08:
I think I am running in to this (or a variant of it) using node-zwave-js v10.0.4 with zwavejs2mqtt v7.1.0 and Homeassistant 2022.7.7.

I noticed one of my Aeotec Door Sensor 7 Pros (ZWA012) suddenly grew a motion sensor entity in Home Assistant and was no longer reporting contact open/closed for the door.  I re-interviewed it but but when it coms back it seems the entity is now "sensor.outside_garage_entry_door_sensor_access_control_door_state" as opposed to "binary_sensor..outside_garage_entry_door_sensor_access_control_window_door_is_open".  Furthermore, Homeassistant is seeing some weird states in the new Homeassisant sensor:
* When closed: "Window/door is closed"
* When open: "22"

Attaching zwavejs log covering a new re-interview followed by a couple of open/close events.  Node is 185.

[node185.log](https://github.com/zwave-js/node-zwave-js/files/9526865/node185.log)

Verbalinsurection, 2022-09-09:
Hi,
I have the similar issue. If this can help, here my zwavejs log, zwavetomqtt log and zwave dump when I add the device and use
https://gist.github.com/Verbalinsurection/c07007f3f189bf22a1d3045e60462d6b

I have posted full decription [here](https://github.com/zwave-js/node-zwave-js/discussions/5033) before finding this issue with help of kpine.

PR #5037 (2022-09-09): fix: merge notification metadata for variables with multiple states
fixes: #5027
closed 2022-09-09 by AlCalzone