zwave-js/zwave-js #5029: Network Died and Can't Recreate

wraith1385, 2022-09-09:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within ZWaveJS2MQTT?

NO, my problem is NOT within ZWaveJS2MQTT

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

Woke up a few days ago and none of my z-wave devices were responding. Thought it was a bad stick so I ordered a replacement and am getting the same results.

 

My setup is a rasberry pi 4 with zwavejs2mqtt running in a docker container and then HomeAssistant points to that. At this point I have tried 3 different zooz zst10 700 sticks as well as an aeotec z-stick 7 plus. I have upgraded the firmware on all of the sticks to 7.17.2. I have factory reset the sticks, blown away the docker container and recreated it, switched to try and use the zwavejs addon in HA, and switched to try the zwavejs2mqtt addon in HA.

 

During all of this I thought I had an nvm backup of the original stick, but apparently not. However, I have factory reset a few different zwave devices in my house (a switch, a sensor, a plug, and the garage door controller) to try to begin recreating the network.

 

No matter which configuration I try, the sticks don't see anything in both inclusion and exclusion mode. I've turned logging up to silly and all I see are the API calls to start and end the inclusion or exclusion modes. 

### Device information

Have tried both the Aeotec z-stick 7 plus as well as a few zooz zst10 700's.  Firmware on all of them was upgraded to 7.17.2.  

### How are you using `node-zwave-js`?

- [X] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [X] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version:
`node-zwave-js` branch: 10.0.4
`zwavejs2mqtt` branch: latest (7.1.0)
HA zwave_js integration: 0.1.70

Don't run both zwavejs2mqtt and the HA zwavejs integration at the same time, but during the troubleshooting process I was trying everything.

### Did you change anything?

no

### Did this work before?

Yes (please describe)

### If yes, where did it work?

House had a successful zwave mesh with over 20 mixed devices (locks, switches, outlets, etc)
[zwavejs2mqtt-store 2.zip](https://github.com/zwave-js/node-zwave-js/files/9531472/zwavejs2mqtt-store.2.zip)

### Attach Driver Logfile

[zwavejs_2022-09-09.log](https://github.com/zwave-js/node-zwave-js/files/9531464/zwavejs_2022-09-09.log)
[zwavejs_2022-09-08.log](https://github.com/zwave-js/node-zwave-js/files/9531467/zwavejs_2022-09-08.log)

AlCalzone, 2022-09-09:
Where are you located? USA? EU?
Is your stick on an extension cord? https://zwave-js.github.io/node-zwave-js/#/troubleshooting/connectivity-issues?id=general-troubleshooting

wraith1385, 2022-09-09:
Located in USA
Stick is on an extension cord.  See photo
[image]

I thought I did an NVM backup, but it must not have ever completed so that's all gone, unless it can somehow be recreated from the json files in the store folder.

AlCalzone, 2022-09-09:
No unfortunately it can't be recreated. You could try moving the stick a bit further away from the other radios, maybe bring one device closer for testing. Other than that I don't see anything wrong.

wraith1385, 2022-09-09:
I held the stick out farther away and brought one of the sensors right next to it and got the same results.  Also some of the switches and outlets I was testing with are only 5 or so feet away.  It's honestly like my house has turned into some kind of z-wave dead zone.

kpine, 2022-09-09:
What about the controller's region and power levels? Are they set correctly?

[image]

AlCalzone, 2022-09-09:
According to the log, +9.9dBm TX power, region USA. So there should be at least some signal.

wraith1385, 2022-09-10:
Yeah, this is what the GUI shows
[image]

kpine, 2022-09-10:
I don't know if it matters, or what the impact is, but those power levels are nowhere close to the recommended ones for a 700-series stick.

AlCalzone, 2022-09-10:
In theory, the powerlevel setting here should make the stick "shout louder", and the 2nd value is for calibrating what the actual power is. Different sticks seem to have different defaults here.

@wraith1385 You could try going to the combination of values that @kpine's screenshot shows. Maybe that changes things, maybe not.

Elsewhere I've tested the settings for my EU Z-Stick 7 and they had no noticeable effect. But that might be a different issue.

wraith1385, 2022-09-10:
Changed the values to those in the screenshot.  Same results.

AlCalzone, 2022-09-10:
At this point I'm afraid you need a Zniffer (UZB3 flashed with special firmware) to inspect the RF traffic and figure out if it's the controller not receiving or the devices not sending.

wraith1385, 2022-09-10:
Ok, I've just ordered a UZB3.  Can someone on here help me with what I'm looking for when I get it and flash it?  I've never delved this deep into z-wave before.

AlCalzone, 2022-09-10:
Here's a description how you can convert it:
https://community.silabs.com/s/article/z-wave-500-converting-a-uzb3-controller-to-a-zniffer?language=en_US

The necessary tools are linked there somewhere, but if you can't find them, I should also have them somewhere.

AlCalzone, 2022-09-10:
Apparently, the Z-Wave Programmer is part of the Z/IP Gateway SDK, which can be downloaded through Simplicity Studio:

> zw_programmer
> 
> A programming tool with source distributed in Z/IP Gateway SDK. It communicates with Z-Wave controller via UART interface, and can be used for Z-Wave controller OTW update. Please install the latest Z/IP Gateway SDK 7.18 from Simplicity Studio v5.
> 
> After installing Z/IP Gateway package on Linux, the zw_programmer is typically located at /usr/local/bin
> 
> The source is available in Z/IP Gateway SDK package:
> C:\SiliconLabs\SimplicityStudio\v5\developer\sdks\zwave\zip_gateway\v7.18.0\Source\zipgateway-7.18.00-Source.tar\zipgateway-7.18.00-Source\usr\local\systools\zw_programmer

kpine, 2022-09-10:
`zw_programmer` is a command line tool for doing OTW updates of the 700-series controllers and NVM backups and restores. It won't work for flashing the Zniffer firmware on a UZB3. It's also painful to compile and install (I wish they'd just make it an available download for everyone, by far the easiest way to update a 700 stick!). You need the "Z-Wave Programmer" (or also called PC Programmer) windows application to do that.

I find this guide to be the easiest to follow: https://bishopfox.com/blog/set-up-zniffer-for-z-wave

Silicon Labs is constantly re-designing their website, but the [link](https://www.silabs.com/development-tools/wireless/z-wave/embedded-development-kit?tab=overview) there for the downloads is still valid. The downloads can be easy to miss, don't go to the "Software & Tools" tab, stay on the Overview tab. Scroll down to the right and look for the "Z-Wave Protocol SDK Downloads" section. You'll need to create a Silicon Labs account to download them. You need both Z-Wave Programmer and Zniffer downloads. You won't need Simplicity Studio at all if you use these.

The Zniffer download from above includes the zniffer firmware. The Zniffer downloaded with Simplicity Studio **does not**
 include the UZB3 firmware. Simplicity Studio does have a newer version of the Zniffer application though.

For anyone else that stumbles upon this comment, I did find a community post with a `sniffer_ZW050x_USBVCP.hex` file download. In case the Zniffer download linked above gets lost in another Silicon Labs website redesign, you may still be able to get the hex file [here](https://community.silabs.com/s/question/0D58Y00008fKLuySAG/to-convert-a-uzb-controller-to-a-zniffer-where-to-download-the-firmware-image-snifferzw050xusbvcphex?language=en_US). I did confirm it's the same exact file in the old Zniffer download.

wraith1385, 2022-09-18:
Ok, I got my zniffer flashed and ready.  Can someone direct me as to what to test and what I'm looking for?

wraith1385, 2022-09-18:
I am so unbelievably confused.  All of a sudden items are working.  I've attached the trace file.  Can you take a look when you get a chance and tell me if anything looks wrong?

[trace.zip](https://github.com/zwave-js/node-zwave-js/files/9594921/trace.zip)

wraith1385, 2022-09-27:
My network has now been stable for over a week.  No idea what changed.

closed 2022-10-15 by zwave-js-assistant[bot]