zwave-js/zwave-js #5035: Missing device configuration: Fakro ZWS230

Sennevds, 2022-09-09:
### Checklist

- [X] It is not in the [configuration DB](https://devices.zwave-js.io/?search=3)

- [X] It was not [merged recently](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Apr+is%3Aclosed) or has a [pending PR](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Aopen+is%3Apr)

### Which device is missing?

Fakro ZWS230

### Manufacturer ID

0x0085

### Product Type

0x0002

### Product ID

0x0116

### Firmware Version

1.2

### Is the device listed on the Z-Wave Alliance website?

https://products.z-wavealliance.org/products/2477

### Do you have a manual?

https://manual.zwave.eu/backend/make.php?lang=en&sku=FAK_ZWS230&cert=ZC08-11020009

AlCalzone, 2022-09-10:
@zwave-js-bot import config 2477 from zwa

AlCalzone, 2022-09-10:
@Sennevds the link you posted is for the ZWS12, which we have a file for.

Sennevds, 2022-09-10:
Sorry about that. After checking the manual and product it looks like it doesn't even use the ZWS230 but is a separate product:
DEF Z-Wave flat roof window.
I will close this issue and create a new one once I have found the Z-Wave specifications

closed 2022-09-10 by Sennevds