zwave-js/zwave-js #5041: [Node 005] [REQ] [ApplicationCommand]   [MultilevelSwitchCCSupportedReport] [INVALID]

Sennevds, 2022-09-10:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within ZWaveJS2MQTT?

NO, my problem is NOT within ZWaveJS2MQTT

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?
Added a new device (Fakro DEG Z-Wave) and even after reinterviewing I get following log:
[Node 005] [REQ] [ApplicationCommand]
                                  └─[MultilevelSwitchCCSupportedReport] [INVALID]
What do you observe?
The shutter is recognized but I can only set the value I don't have the open or close buttons like other shutters

What did you expect to happen?
See the buttons for open and close

### Device information

Manufacturer: Fakro
Model name: DEG Z-Wave (0x0085-0x0002-0x0116 not found on the z-wave alliance website I asked Fakro for more info)
Node ID in your network: 005

### How are you using `node-zwave-js`?

- [X] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [ ] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Did you change anything?

no

### Did this work before?

Don't know, this is a new device

### Attach Driver Logfile

[zwavejs_current.log](https://github.com/zwave-js/node-zwave-js/files/9540523/zwavejs_current.log)

AlCalzone, 2022-09-10:
Interesting - although the secondary switch type field is deprecated, I've never actually seen a device that omits the byte for it. The driver did enforce it to exist, but didn't use it, so I've now relaxed that check.

closed 2022-09-10 by AlCalzone

Sennevds, 2022-09-10:
Damn you're fast 😀. Thank you for the fix!

PR #5042 (2022-09-10): fix: do not enforce Multilevel Switch CC secondary switch type field to exist
we don't use or support it anyways
fixes: #5041