zwave-js/zwave-js #5045: Problem updating firmware for GE 14287 / ZW4002

avocadosalsa, 2022-09-11:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within ZWaveJS2MQTT?

NO, my problem is NOT within ZWaveJS2MQTT

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?
I am unable to successfully update the firmware on my GE 14287 / ZW4002 from 5.22 to 5.24.  [Others on the Home Assistant forum are having the same issue.](https://community.home-assistant.io/t/ota-firmware-update-fails-for-ge-jasco-14287-dimmer-switch/458649)

What do you observe?
It appears the firmware is sent to the device but step after, where it re-interviews the device, fails after 5 attempts.  At this point (as the re-interview occurs) the device turns off and acts as if it has been factory reset.  The node is flagged as dead and I have to re-include it before it can be accessed again.  After re-including the node, the firmware is still 5.22.

What did you expect to happen?
The firmware to update successfully.

Steps to reproduce the behavior:
  1. Go to node
  2. Go to 'OTA Updates'
  3. Click update
  4. Wait for update to finish
  5. Observe node is no longer accessible

### Device information

Manufacturer: GE
Model name: 14287 / ZW4002
Node ID in your network: 038

### How are you using `node-zwave-js`?

- [X] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [ ] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version:
`node-zwave-js` branch: 10.0.4
`zwavejs2mqtt` branch: 7.1.0

### Did you change anything?

no

### Did this work before?

Don't know, this is a new device

### Attach Driver Logfile

[zwavejs_2022-09-11.log](https://github.com/zwave-js/node-zwave-js/files/9541785/zwavejs_2022-09-11.log)
Firmware update begins at T04:09:07

AlCalzone, 2022-09-11:
I'll pull that update for now and let Jasco know.

kpine, 2022-09-11:
FWIW, I updated this device manually (before the OTA service existed) with the file from GitHub, and have had no issues at all. I wonder what the difference could be. The file I used is identical to the one the service links to.

I did have a problem with a different device (14292 / ZW4003, 0x0063-0x4952-0x3037) which I updated via the OTA service (using HA). The device upgraded successfully but cannot pass Health Checks, 0/10 pings for every test for a 0 score. Yet normal operation and manual pings were fine. The Health Check was working OK before the update. I could also not exclude the device. I factory reset it, re-included it as a replaced failed node, and it works fine now, passing Health Check with high scores. I do have zniffer and driver logs of before and after, but not sure they reveal anything besides confirming that the device was acting bad.

avocadosalsa, 2022-09-12:
In an attempt to rule out a multi-hop situation, I physically moved one of the problematic switches to be near the controller.  I then healed that node and checked that it had a direct connection to the controller.  Updating the firmware (grabbed the file from GitHub and manually updated) resulted in the same failure.

FYI, the date code and the problematic switch is 1738.

AlCalzone, 2025-05-09:
We can't do anything about this.