zwave-js/zwave-js #5075: Isses with floor sensor for thermostat

pittbull, 2022-09-15:
I've installed 5 thermostats ([Namron](https://www.elektroimportoren.no/docs/lib/4512744-Brukerveiledning-5.pdf)) in my house and I am seeing issues with one in particular. 

These thermostats have two sensors: room and floor, snd can operate with only the floor sensor active or floor+room. 

The one with problems is the one running in floor-only mode. It seems to have issues with mapping the floor temperature to current temperatur and when adjusting the set point the thermostats only kicks in when passing the room temperature (even though running with floor only). 

I have tried multiple resets and inclusion without a new result. 

How can I isolate the problem?

AlCalzone, 2022-09-15:
Please make a [driver log](https://zwave-js.github.io/zwavejs2mqtt/#/troubleshooting/generating-logs?id=driver-logs), loglevel `debug` and attach it here as a file.

The logfile should highlight the interaction that isn't working as expected.

pittbull, 2022-09-15:
Thanks. The node in question is number 78.

[zwavejs2mqtt_2022-09-15.log](https://github.com/zwave-js/node-zwave-js/files/9577860/zwavejs2mqtt_2022-09-15.log)

AlCalzone, 2022-09-15:
Wrong log. 
[image]

pittbull, 2022-09-16:
My bad.

[zwavejs_2022-09-16.log](https://github.com/zwave-js/node-zwave-js/files/9581048/zwavejs_2022-09-16.log)

pittbull, 2022-09-19:
Well, I am hard pressed at supplying what you need as I cannot point to a specific. The problem, as I observe it, is that the current temperature indicator is not being supplied from the floor sensor in the thermostat, but rather the room sensor.

Any suggestions on how I can gather what you need?

AlCalzone, 2022-09-19:
Do you mean on the actual thermostat itself? Or in Home Assistant / Z2M / ...?
The first we have no influence over.

pittbull, 2022-09-19:
I mean in HA/Z2M and how I can capture the information you require.

AlCalzone, 2022-09-19:
Looking at your last screenshot, it seems like both measurements reach the driver and z2m displays them both, correct?
Which would make this an issue for [Home Assistant](https://github.com/home-assistant/core/issues), not this repository.

I'm not sure though what they can do about this, since the floor sensor gets reported on a different endpoint, which is a different logical device according to the specification.

pittbull, 2022-09-19:
I see. Thanks for the feedback. I will try with HA.

kpine, 2022-09-20:
If there's a configuration parameter to determine which temperature sensor to read from (from the manual, appears that would be parameter 10?), the integration can be customized to read from a specific sensor and endpoint. That's how Heat-It thermostats are done. https://github.com/home-assistant/core/blob/1021c90bb8723fc827efd49abd472a8817d1ec9a/homeassistant/components/zwave_js/discovery.py#L399-L48

pittbull, 2022-09-20:
> If there's a configuration parameter to determine which temperature sensor to read from (from the manual, appears that would be parameter 10?), the integration can be customized to read from a [...]

Interesting, thanks for sharing. @AlCalzone any comments?

AlCalzone, 2022-09-20:
If that param exists, this is still an issue for the HA Core repo, but with a possible solution.

closed 2022-09-30 by zwave-js-assistant[bot]