zwave-js/zwave-js #5082: Jasco GE Z-Wave 12724 / ZW3005 Status no completely updated when switch is pressed

jcat2, 2022-09-18:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

YES, BUT a Home Assistant developer has told me to come here

### Is your problem within ZWaveJS2MQTT?

NO, my problem is NOT within ZWaveJS2MQTT

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?
Manually pressing a Jasco GE Z-Wave 12724 / ZW3005 dimmer.

What do you observe?
I have multiple Jasco GE Z-Wave 12724 / ZW3005 dimmers in my house. These dimmer have a unique feature than when they are power off they gracefully take their time to reach the off state. From full power on this can take up to 3 seconds. While these dimmers are powering down they report their current power level when requested by a controller. In the case of Z-Wave JS when the switch is powered off the node in question send a status update and the controller requests the updated status. Seeing the device is still in transition the power level it is at when requested is reported. The driver never requests another status so an application may get a stale value. So in an application like Home Assistant the light status is still showing on even though the dimmer has transitioned to an off state.

What did you expect to happen?
When a Jasco GE Z-Wave 12724 / ZW3005 dimmer switch is pressed and the controller has received such a request it will request the power level and continue to check it until it stabilizes or so many tried have been attempted. An other option would to be to have the option of programming a delay before the controller probes the device in order of the dimmer to go thru the required transition.

Steps to reproduce the behavior:
  1. Set a Jasco GE Z-Wave 12724 / ZW3005 to full power
  2. Press the dimmer off manually
  3. Monitor the driver activity
  4. See error

See Home Assistant Bug Report for additional information https://github.com/home-assistant/core/issues/78396.

### Device information

Manufacturer: Jasco GE 
Model name: Z-Wave 12724 / ZW3005 
Node ID in your network: 8

### How are you using `node-zwave-js`?

- [ ] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [X] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

Driver Version:
10.0.4
Server Version:
1.22.1

### Did you change anything?

no

### Did this work before?

Yes (please describe)

### If yes, where did it work?

This did work in Home Assistant 2022.8.7 but i don't know what version of Z-Wave JS was used thend

### Attach Driver Logfile

[Full On Press Off Press On.log](https://github.com/zwave-js/node-zwave-js/files/9595271/Full.On.Press.Off.Press.On.log)

This log was performed under the following conditions.

The dimmer was set to full power
The dimmer was commanded off via a manual push off
The status of the dimmer was manually requested in Home Assistant
The dimmer was commanded on via a manual push on
The status of the dimmer was manually requested in Home Assistant

AlCalzone, 2022-09-19:
Not sure how this was supposed to be working before, but Z-Wave JS has never done any repeated polling nor has it been refreshing until a status has stabilized.

What happens when you manually control the switches here is that they send a (very generic) command. For older generation devices, this command is used as a trigger to refresh the status once. 
The switches seem to send that command immediately, so when the status is refreshed, it is very early in the dimming process.
We have a compat flag to delay the refresh, but it seems like this delay is actually user-configurable, so a fixed delay in the config file will work for some users only and introduce an unnecessary delay for others.

Any chance you're able to make a log on an older version where it did work?

jcat2, 2022-09-20:
I may be able to go back to Home Assistant 2022.8.7. I'll get another SD card and see if I can get some logs of what was going on in that version. While I'm waiting for SD card I'll see if I can set up a compat flag to change the delay of the refresh for that device.

closed 2022-10-01 by zwave-js-assistant[bot]