zwave-js/zwave-js #5094: No automatic updates

roblomq, 2022-09-21:
For some reason there are no automatic updates made on for example energy and temperature reports. 

I have several battery and net powered devices (mostly Fibaro) and when I refresh a meter value manually a MQTT message is send (I see the update in MQTT Explorer) but even when the settings of these devices are configured (triple checked) to sendo frequent updates, there is nothing happening. Only when done manually of press on the save button in the configuration menu. 

Strangely enough the only Neo Coolcam device I have does send regular updates all other Fibaro devices don't.
What could this be?

AlCalzone, 2022-09-21:
Try re-interviewing to have the lifeline association set up automatically. Maybe that went wrong last time.

roblomq, 2022-09-21:
Tried that already buy to make sure I did it again. Same result, after re-interviewing the values are updated once but no updates after that.
Last active time is more than 4 hours ago while for example energy report interval is on 10 min.

AlCalzone, 2022-09-21:
Please make a [driver log](https://zwave-js.github.io/zwavejs2mqtt/#/troubleshooting/generating-logs?id=driver-logs), loglevel `debug` and attach it here as a file (drag & drop). Please start the log before doing another re-interview and keep it running for 20 minutes or so so we should see at least one report.

roblomq, 2022-09-21:
[zwavejs2mqtt_2022-09-21.log.txt](https://github.com/zwave-js/node-zwave-js/files/9620686/zwavejs2mqtt_2022-09-21.log.txt)
Is this one sufficient or do you need more log files?

robertsLando, 2022-09-22:
That is application log, nont driver log, check here: https://zwave-js.github.io/zwave-js-ui/#/troubleshooting/generating-logs?id=driver-logs

roblomq, 2022-09-22:
[zwavejs_2022-09-22.log.txt](https://github.com/zwave-js/node-zwave-js/files/9628850/zwavejs_2022-09-22.log.txt)
This one then, sorry for not looking good enough to the instructions..

AlCalzone, 2022-09-23:
Ok, so the interview of the thermostat (39) looks okay, the association is set up, but no messages are received after the interview. Did you do anything on the thermostat that should have triggered a report, like turning heating on/off?

On second look I do see a single Meter CC received from node 39, but this one uses a non-existent command (0x06), so it does not get parsed. Would be interesting to see if that is a fluke (i.e. a bit flipped due to RF noise) or if the device actually sends invalid reports.

roblomq, 2022-09-23:
Thank you for your time so far.
I did not do anything on the thermostat but if I'm right there are several sensors that should send a report, like lux, floor and air temperature, power and energy usage.
If I turn the heat on there are also no updates send for the above sensors. Should I make a new log or do I need to do something else?

AlCalzone, 2022-09-23:
There are a couple of reports in the log after the interview though.
* Node 34, temperature
* Node 2, Binary sensor
* Node 5, kWh, W, V, A
* Node 2, Binary sensor again
* Node 2, Binary sensor again
* Node 3, kWh
* Node 39 (the strange one above)
* Node 2, Binary sensor again
* Node 14, W
* Node 2, Binary sensor again
* Node 36, W
* Node 40, kWh
* Node 34, temperature
* Node 35, kWh
* Node 5, kWh, W, V, A
etc. etc.

If you're missing updates from a specific sensor/meter, I can look for them, but I'll need to know which node and what you're missing when. I'll need a new log then where you do something that should guaranteed result in a report.

roblomq, 2022-09-23:
Let's start with focussing on node 39, this is a Heltun HE-HT01 Thermostat, documentation with a full parameter list can be found [here](https://drive.google.com/file/d/1dNGNnWmmXbKfB-OBE4jAytLGRheSSln4/view)
Here the log of only this node 39.
[zwavejs_2022-09-23.log](https://github.com/zwave-js/node-zwave-js/files/9636879/zwavejs_2022-09-23.log)
I've turned the thermostat on so this must force some parameter updates. After that (on the end of the logfile) i've refreshed the parameter list (one by one per category) through the refresh buttons in zwavejs2mqtt. This causes that the values are updated in domoticz so there are mqtt messages send.

AlCalzone, 2022-09-25:
All I see in relation to Node 39 are pings after driver startup. No commands controlling it, no automatic reports of any kind.

But there are a lot of serial commands missing the human-readable explanation. Do you have any filters active for the log? If so, please disable them and try again. I don't care about the additional noise, that's better than missing log entries.

roblomq, 2022-09-25:
I only put a filter on the node number. Should I remove that and send a new one?

AlCalzone, 2022-09-25:
Yes please. That doesn't seem to work as intended

roblomq, 2022-09-25:
[zwavejs_2022-09-25.log.txt](https://github.com/zwave-js/node-zwave-js/files/9640867/zwavejs_2022-09-25.log.txt)
This is a log with no filter on node number and on debug level. I don't know if there are more ways of filtering than the one in the zwave settings so  I think this is the most unfiltered log possible.

AlCalzone, 2022-09-25:
There's one incoming report for Node 39 in there (power usage stayed the same at 0.8 W):
```
2022-09-25T13:27:44.701Z CNTRLR   [Node 039] [~] [Meter] value[66049]: 0.8 => 0.8                   [Endpoint 1]
2022-09-25T13:27:44.710Z SERIAL » [ACK]                                                                   (0x06)
2022-09-25T13:27:44.718Z DRIVER « [Node 039] [REQ] [ApplicationCommand]
                                  └─[MultiChannelCCCommandEncapsulation]
                                    │ source:      1
                                    │ destination: 0
                                    └─[MeterCCReport]
                                        type:       Electric
                                        scale:      W
                                        rate type:  Consumed
                                        value:      0.8
                                        time delta: 0 seconds
```

roblomq, 2022-09-25:
That's strange because I tuned it on, I wish is was 0,8W 😄. But besides that, where are the other reports of the temperature and lux for example? But I assume this is something of the device itself right? So nothing missing in Zwave2MQTT?
I have contacted Heltun already about this.
The first device I had made a high beeping sound so got a new one butt this also beeps.

AlCalzone, 2022-09-25:
There were also several multi sensor reports, I just focused on that one.

roblomq, 2022-11-05:
After a firmware update of the Heltun HE-HT01 Thermostat the updates are automatic now, so this is solved now.
Only my GreenWave Reality Multi-socket PowerNode (6-plug) isn't sending regular updates but this seems a known issue.

closed 2022-11-05 by roblomq