zwave-js/zwave-js #5095: FGSD002 Not reporting Smoke Alarm

Rouzax, 2022-09-21:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within ZWaveJS2MQTT?

NO, my problem is NOT within ZWaveJS2MQTT

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?
The Fibaro Smoke Sensor is pickup up correctly but when triggering a Smoke Alarm test on it, I do not see any topics coming in and last seen state is also not updated
[image]

What do you observe?
No updates come in on ZwaveJS UI

What did you expect to happen?
Expect data to come in about Alarm

Steps to reproduce the behavior:
  1. Go to ZwaveJS UI and open Device
  2. On Physical Fibaro Device trigger Self Test

Before I used OpenZwave in Domoticz and triggering a Self test would give me Alarms coming in.

### Device information

Manufacturer: Fibaro
Model name: FGSD002 
Node ID in your network: 

### How are you using `node-zwave-js`?

- [X] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [ ] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version: 10.2.0
`node-zwave-js` branch: 10.2.0
`zwavejs2mqtt` branch:

zwave-js-ui: 8.0.0
zwave-js: 10.2.0
home id: 3244133030
home hex: 0xc15d8aa6

### Did you change anything?

no

### Did this work before?

Don't know, this is a new device

### Attach Driver Logfile

[zwavejs_2022-09-21.log](https://github.com/zwave-js/node-zwave-js/files/9617329/zwavejs_2022-09-21.log)

1. Did a device wakeup
2. Initiated self test
3. did a device wakeup

AlCalzone, 2022-09-21:
Can you add node 1 to the sensor's association group 4 and try again?
[image]

If it works then, we'll have to edit the config file for the controller to be auto-assigned there.

Rouzax, 2022-09-21:
Could you give me a bit more direction? Pretty new to this.

You mean, add this to the Smoke Sensor?
[image]

Rouzax, 2022-09-21:
So I tried to do the above but nothing is added to the `Groups` even after waking the device up to let it receive its config.

Rouzax, 2022-09-21:
Do get this error eventually not sure if related
[image]

Rouzax, 2022-09-21:
> That's what I meant, but if the association already exists, it won't change anything. You'll have to reach out to the Fibaro support then. If the device doesn't send notifications, Z-Wave JS cannot show them.
> 
> > Do get this error eventually not sure if related
> > Nah, that happens because an automation or something tries to set the `targetValue` to a string, when it must be `true` or `false`.

Not sure if the correct smoke alarm association exist as there are 2 Smoke Alarm.

In my setup I have OpenZwave running in Domoticz and am in the process to migrate to ZwaveJS so I can have either running at a time to test and with OpenZwave and the same controller and Fibaro device I do get the alarms in.

Could it be that you expect a binary value while Fibaro is sending out Text?
My notification would have actual text coming from the smoke sensor with the state.

Rouzax, 2022-09-21:
These are the devices that are shown in OpenZwave for this device
[image]

The Alarm Text is device 392
The switch that is triggered with Alarm is 348

None of these seem to come in on ZwaveJS

Rouzax, 2022-09-21:
It looks like it is not a Text directly but a code which is translated by Domoticz

AlCalzone, 2022-09-21:
> Could it be that you expect a binary value while Fibaro is sending out Text?

You're misunderstanding. The text is coming from an application trying to control the device. Z-Wave JS complains that the application tries to set the value to a string, but it needs to be a boolean.

> so I can have either running at a time to test and with OpenZwave and the same controller and Fibaro device I do get the alarms in.

Can you make a logfile with OZW that shows the alarm command being received? In your previous log, the device didn't send anything.

Rouzax, 2022-09-21:
> > Could it be that you expect a binary value while Fibaro is sending out Text?
> 
> You're misunderstanding. The text is coming from an application trying to control the device. Z-Wave JS complains that the application tries to set the value to a string, but it needs to be a boolean.

I understand now 😃

> 
> > so I can have either running at a time to test and with OpenZwave and the same controller and Fibaro device I do get the alarms in.
> 
> Can you make a logfile with OZW that shows the alarm command being received? In your previous log, the device didn't send anything.

Would you happen to know how to do that?

AlCalzone, 2022-09-21:
I don't use Domoticz, but I know that OZW has an option to write a file (I think OZW_Log.txt) somewhere, which contains the serial communication.

Rouzax, 2022-09-21:
Thanks, I'll have a look

Rouzax, 2022-09-22:
Found it,
Here is the Log filtered to only include the Smoke Sensor which is Node 037
At 08:47:05.497 I Woke up the device
At 08:47:05.829 I triggered the Alarm

[OZW_Log_Node37.txt](https://github.com/zwave-js/node-zwave-js/files/9623135/OZW_Log_Node37.txt)

If you prefer the entire log let me know

AlCalzone, 2022-09-22:
Ok it definitely sends the command that wasn't included in the zwave-js logs. Can you compare the associations between OZW and zwave-js to see if there are differences?

Rouzax, 2022-09-22:
I think you mean this
[image]

AlCalzone, 2022-09-22:
Ok unless you edited them in the meantime, they are identical.
Can you try again with zwave-js, but disable any filtering on the logs? Not that the message was just filtered out.

Rouzax, 2022-09-22:
Will try that later today, have some work stuff I need to do first 😄

Rouzax, 2022-09-22:
Looking at MQTT Explorere I now also see things coming in there.
[image]

zwave/Game_Room/Server_Sensor/113/0/Smoke_Alarm/Alarm_status
```json
{"time":1663838247612,"value":0,"nodeName":"Server Sensor","nodeLocation":"Game Room"}
{"time":1663838235827,"value":3,"nodeName":"Server Sensor","nodeLocation":"Game Room"}
```

AlCalzone, 2022-09-22:
Looks like its working then. Probably a fluke before?

Rouzax, 2022-09-22:
> Looks like its working then. Probably a fluke before?

I don't know, but yes most likely 😄 
Did see that 1 of the Smoke Alarms I had, only had Lifeline to Node_1 and needed to add the other 2 myself.

Other thing I'm seeing is a discrepancy between the 2 of the same Smoke Sensors
[image]

AlCalzone, 2022-09-22:
> needed to add the other 2 myself.

Did that change anything in terms of reports? If so, I'll need to edit the config file.

> Top device only had 6 HA Attributes while the bottom has 9

Might be a difference in device firmware. In any case, that list is the legacy HA MQTT discovery, which is not maintained:
https://zwave-js.github.io/zwave-js-ui/#/homeassistant/homeassistant-mqtt?id=home-assistant-using-mqtt-discovery

Rouzax, 2022-09-22:
> > needed to add the other 2 myself.
> 
> Did that change anything in terms of reports? If so, I'll need to edit the config file.

Not sure as the one I did the testing with already had 3 to begin with.

> > Top device only had 6 HA Attributes while the bottom has 9
> 
> Might be a difference in device firmware. In any case, that list is the legacy HA MQTT discovery, which is not maintained: https://zwave-js.github.io/zwave-js-ui/#/homeassistant/homeassistant-mqtt?id=home-assistant-using-mqtt-discovery

Not sure what this would mean for Domoticz as I think they use this.

AlCalzone, 2022-09-22:
> Not sure what this would mean for Domoticz as I think they use this.

I think long term they should move over to the websocket API, but their dev once said they'd keep using the MQTT api because they already have a system in place for it and it works.
I guess that's fair, but we can't make any guarantees that everything will work over that API indefinitely.

Rouzax, 2022-09-22:
Thanks for your help so far, I'm closing the issue as I think it is working and it was just a fluke.

closed 2022-09-22 by Rouzax