zwave-js/zwave-js #5109: Missing device configuration: Honeywell TH6320ZW2007

sormy, 2022-09-25:
### Checklist

- [X] It is not in the [configuration DB](https://devices.zwave-js.io/?search=3)

- [X] It was not [merged recently](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Apr+is%3Aclosed) or has a [pending PR](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Aopen+is%3Apr)

### Which device is missing?

Honeywell TH6320ZW2007

### Manufacturer ID

0x041b

### Product Type

0x0011

### Product ID

0x0009

### Firmware Version

3.0.2

### Is the device listed on the Z-Wave Alliance website?

nope

### Do you have a manual?

https://www.resideo.com/us/en/products/air/thermostats/programmable-thermostats/z-wave-t6-pro-programmable-thermostat-with-smartstart-th6320zw2007-u/#support

It looks like an improved version of th6320zw2003.

I have noticed that th6320zw2007 supports S2 Authenticated and Z-Wave Plus: Version 2, has a pin code on the back of thermostat. th6320zw2003 doesn't have s2 authenticated support and supports only Z-Wave Plus: Version 1.

Functionally th6320zw2007 works the same as th6320zw2003 in home assistant but "configure device" in HA shows a bunch of properties without labels.

I have looked on raw values for  th6320zw2007 (45 parameters) and compared with th6320zw2003 (42 parameters). It seems like most of them are the same parameters and in the same position as on th6320zw2003 but th6320zw2007 has extra 3 parameters.

DanielWinks, 2022-11-29:
Looks like @sormy added a commit to this almost 2 months back, with the necessary config files.

@AlCalzone Is there anything else needed to get this merged? I also have this same thermostat and would like to see the config option text set on it. If there's anything else needed I'd be happy to help out to get this merged.

Thanks.

DanielWinks, 2022-11-29:
Here's a PR for this: https://github.com/zwave-js/node-zwave-js/pull/5230

AlCalzone, 2022-12-05:
You'll have to wait on @blhoward2 to review the PR. I'm on parental leave until the end of 2022.

AlCalzone, 2023-01-31:
fixed in https://github.com/zwave-js/node-zwave-js/pull/5143

closed 2023-01-31 by AlCalzone