zwave-js/zwave-js #5110: Auto firmware update of iblinds v3 fails with status Error_InvalidFirmwareTarget

decairn, 2022-09-25:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within Z-Wave JS UI (formerly ZwaveJS2MQTT)?

NO, my problem is NOT within Z-Wave JS UI

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?
Unknown

What do you observe?
Tried update of firmware through HA UI, device ends up on original firmware

What did you expect to happen?
Device successful in updating to latest firmware

Steps to reproduce the behavior:
  1. Go to '...' - HA settings, updates
  2. Click on '...' - the device you want, then install
  3. Scroll down to '...'
  4. See error - no error in HA (I'll raise an issue with them on that separately), zwavejs logs shows the failure messa

### Device information

Manufacturer: iblinds
Model name: v3
Node ID in your network: 30, 31, 34 tried
[nodes_dump.zip](https://github.com/zwave-js/node-zwave-js/files/9640909/nodes_dump.zip)

### How are you using `node-zwave-js`?

- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (latest)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (dev)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [ ] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [X] Other _(please describe)_

### Which branches or versions?

version:
zwavejs2mqtt: 7.2.0
zwave-js: 10.1.0
HA 2022.9.6

Using the zwave2mqtt addon in HA and not the zwave_js integration directly

### Did you change anything?

no

### Did this work before?

Don't know, this is a new device

### If yes, where did it work?

N/A - New function / integration with iblinds auto updates

### Attach Driver Logfile

Nodes 30 and 31 are INFO level, node 34 I did a restart with DEBUG

[zwavejs_current.zip](https://github.com/zwave-js/node-zwave-js/files/9640913/zwavejs_current.zip)

AlCalzone, 2022-09-25:
@chancehiblinds I'm afraid you need to take a look at this.
The process seems fine to me, nothing stands out:

It starts:
```
2022-09-25T13:14:27.272Z DRIVER » [Node 034] [REQ] [SendData]
                                  │ transmit options: 0x25
                                  │ callback id:      110
                                  └─[FirmwareUpdateMetaDataCCMetaDataGet]

2022-09-25T13:14:28.563Z DRIVER « [Node 034] [REQ] [ApplicationCommand]
                                  └─[FirmwareUpdateMetaDataCCMetaDataReport]
                                      manufacturer id:         647
                                      firmware id:             257
                                      checksum:                0
                                      firmware upgradable:     true
                                      max fragment size:       40
                                      additional firmware IDs: []
                                      hardware version:        1
                                      continues to function:   unknown
                                      supports activation:     unknown
```

the device is instructed to update target 0
```
2022-09-25T13:14:28.689Z DRIVER » [Node 034] [REQ] [SendData]
                                  │ transmit options: 0x25
                                  │ callback id:      111
                                  └─[FirmwareUpdateMetaDataCCRequestGet]
                                      manufacturer id: 0x0287
                                      firmware id:     0x0101
                                      checksum:        0x750d
                                      firmware target: 0
                                      fragment size:   40
                                      activation:      false

2022-09-25T13:14:28.742Z DRIVER « [Node 034] [REQ] [ApplicationCommand]
                                  └─[FirmwareUpdateMetaDataCCRequestReport]
                                      status: OK
```

it requests the fragments
```
2022-09-25T13:14:28.874Z DRIVER « [Node 034] [REQ] [ApplicationCommand]
                                  └─[FirmwareUpdateMetaDataCCGet]
                                      total # of reports: 1
                                      report number:      1
2022-09-25T13:14:28.885Z CNTRLR » [Node 034] Sending firmware fragment 1 / 5201

...

2022-09-25T13:23:43.563Z DRIVER « [Node 034] [REQ] [ApplicationCommand]
                                  └─[FirmwareUpdateMetaDataCCGet]
                                      total # of reports: 1
                                      report number:      5201
2022-09-25T13:23:43.566Z CNTRLR » [Node 034] Sending firmware fragment 5201 / 5201

2022-09-25T13:23:43.580Z DRIVER » [Node 034] [REQ] [SendData]
                                  │ transmit options: 0x25
                                  │ callback id:      226
                                  └─[FirmwareUpdateMetaDataCCReport]
                                      report #: 5201
                                      is last:  true
```

but then returns the error "invalid firmware target"
```
2022-09-25T13:23:46.844Z SERIAL « 0x010d00040022057a07040002b5001f                                    (15 bytes)
2022-09-25T13:23:46.849Z SERIAL » [ACK]                                                                   (0x06)
2022-09-25T13:23:46.853Z DRIVER « [Node 034] [REQ] [ApplicationCommand]
                                  └─[FirmwareUpdateMetaDataCCStatusReport]
                                      status:    Error_InvalidFirmwareTarget
                                      wait time: 2 seconds
2022-09-25T13:23:46.859Z CNTRLR « [Node 034] Firmware update failed with status Error_InvalidFirmwareTarget
```

asucrews, 2022-09-28:
Hello, I have a different Manufacturer, and different device type but having the same issues. Would you like me to start a new issues?

Aeotec Door Sensor 7

[zwavejs_2022-09-27.log](https://github.com/zwave-js/node-zwave-js/files/9661264/zwavejs_2022-09-27.log)

AlCalzone, 2022-09-28:
Yeah let's keep that separate please.

closed 2022-10-07 by AlCalzone

decairn, 2022-10-07:
Confirmed this is now working for both versions of 3.x controller.  Thank you!