zwave-js/zwave-js #5113: Missing device configuration: Mco Home Technology Co. LTD MH-DS221

itzniru, 2022-09-25:
### Checklist

- [X] It is not in the [configuration DB](https://devices.zwave-js.io/?search=3)

- [X] It was not [merged recently](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Apr+is%3Aclosed) or has a [pending PR](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Aopen+is%3Apr)

### Which device is missing?

Mco Home Technology Co. LTD MH-DS221

### Manufacturer ID

351

### Product Type

0x2210

### Product ID

0x7102

### Firmware Version

2.0.0

### Do you have a manual?

I made a config file:

https://github.com/itzniru/mh-ds221

My first one so probably missing something
[MH-DS221.pdf](https://github.com/zwave-js/node-zwave-js/files/9641773/MH-DS221.pdf)

itzniru, 2022-09-25:
{
	"manufacturer": "McoHome Technology Co., Ltd.",
	"manufacturerId": "0x015f",
	"label": "MH-DS221",
	"description": "Micro Switch",
	"devices": [
		{
			"productType": "0xd221",
			"productId": "0x7102"
		}
	],
	"firmwareVersion": {
		"min": "0.0",
		"max": "255.255"
	},
	"paramInformation": [
		{			
		    "#": "1", 
			"label": "Behavior if power fails",
			"description": "If the state of the switch is saved after a power failure",
			"valueSize": 1,
			"minValue": 0,
			"maxValue": 1,
			"defaultValue": 0,
			"allowManualEntry": false,
			"options": [
				{
					"label": "Not saved, switch will be off",
					"value": 0
				},
				{
					"label": "Saved, will return to previous state",
					"value": 1
				}
			]
		},
		{
			"#": "2",
			"label": "External switch type",
			"valueSize": 1,
			"minValue": 0,
			"maxValue": 3,
			"defaultValue": 0,
			"allowManualEntry": false,
			"options": [
				{
					"label": "Momentary",
					"value": 0
				},
				{
					"label": "Toggle",
					"value": 1
		    }
			]
		},
		{
			"#": "3",
			"label": "External switch behavior",
			"valueSize": 1,
			"minValue": 0,
			"maxValue": 1,
			"defaultValue": 1,
			"allowManualEntry": false,
			"options": [
				{
					"label": "Without specified on/off position",
					"value": 0
				},
				{
					"label": "With specified on/off position",
					"value": 1
				}
			]
		},
		{
			"#": "4",
			"label": "Behavior when timer is active",
			"description": "If manual operation is allowed when the timer function is active",
			"valueSize": 1,
			"minValue": 0,
			"maxValue": 1,
			"defaultValue": 1,
			"allowManualEntry": false,
			"options": [
				{
					"label": "Forbid manual operation",
					"value": 0
				},
				{
					"label": "Allow manual operation",
					"value": 1
				}
			]
		},
		{
			"#": "5",
			"label": "Timer function",
			"description": "Timer function to automaticly turn off the Load",
			"unit": "100ms",
			"valueSize": 2,
			"minValue": 0,
			"maxValue": 32767,
			"defaultValue": 0
		},
    {
			"#": "8",
			"label": "Report options",
			"description": "Sets report options to association group 2",
			"valueSize": 1,
			"minValue": 0,
			"maxValue": 255,
			"defaultValue": 0,
			"allowManualEntry": false,
			"options": [
				{
					"label": "Report both on/off",
					"value": 0
				},
				{
					"label": "Report on only",
					"value": 1
				},
				{
					"label": "Report off only",
					"value": 2
				},
				{
					"label": "Reporting is disabled",
					"value": 255
				}
			]
		},
		{
			"#": "20",
			"label": "Local operation",
			"description": "If local operation of the switch is allowed",
			"valueSize": 1,
			"minValue": 0,
			"maxValue": 1,
			"defaultValue": 0,
			"allowManualEntry": false,
			"options": [
				{
					"label": "Local operation is enabled",
					"value": 0
				},
				{
					"label": "Locked, local operation is disabled",
					"value": 1
				}
			]
		},
		
		{
			"#": "21",
			"label": "Remote operation",
      			"description": "If remote operation of the switch is allowed",
			"valueSize": 1,
			"minValue": 0,
			"maxValue": 1,
			"defaultValue": 0,
			"allowManualEntry": false,
			"options": [
				{
					"label": "Remote operation is enabled",
					"value": 0
				},
				{
					"label": "Locked, remote operation is disabled",
					"value": 1
				}
			]
		},
		{
			"#": "255",
			"label": "Factory reset",
			"valueSize": 1,
			"minValue": 0,
			"maxValue":85,
			"defaultValue": 0,
			"unsigned": true,
			"writeOnly": true,
			"allowManualEntry": false,
			"options": [
				{
					"label": "Idle",
					"value": 0
				},
				{
					"label": "Factory reset",
					"value": 85
				}
			]
		}
	]
}

itzniru, 2022-09-26:
{
	"manufacturer": "McoHome Technology Co., Ltd.",
	"manufacturerId": "0x015f",
	"label": "MH-DS221",
	"description": "Micro Switch",
	"devices": [
		{
			"productType": "0xd221",
			"productId": "0x7102"
		}
	],
	"firmwareVersion": {
		"min": "0.0",
		"max": "255.255"
	},
	"paramInformation": [
		{			
		    "#": "1", 
			"label": "Behavior if power fails",
			"description": "If the state of the switch is saved after a power failure",
			"valueSize": 1,
			"minValue": 0,
			"maxValue": 1,
			"defaultValue": 0,
			"unsigned": true,
			"allowManualEntry": false,
			"options": [
				{
					"label": "Not saved, switch will be off",
					"value": 0
				},
				{
					"label": "Saved, will return to previous state",
					"value": 1
				}
			]
		},
		{
			"#": "2",
			"label": "External switch type",
			"valueSize": 1,
			"minValue": 0,
			"maxValue": 1,
			"defaultValue": 0,
			"unsigned": true,
			"allowManualEntry": false,
			"options": [
				{
					"label": "Momentary",
					"value": 0
				},
				{
					"label": "Toggle",
					"value": 1
		    }
			]
		},
		{
			"#": "3",
			"label": "2-State Switch",
			"valueSize": 1,
			"minValue": 0,
			"maxValue": 1,
			"defaultValue": 1,
			"unsigned": true,
			"allowManualEntry": false,
			"options": [
				{
					"label": "Without specified on/off position",
					"value": 0
				},
				{
					"label": "With specified on/off position",
					"value": 1
				}
			]
		},
		{
			"#": "4",
			"label": "Behavior when timer is active",
			"description": "If manual operation is allowed when the timer function is active",
			"valueSize": 1,
			"minValue": 0,
			"maxValue": 1,
			"defaultValue": 1,
		   "unsigned": true,
			"allowManualEntry": false,
			"options": [
				{
					"label": "Forbid manual operation",
					"value": 0
				},
				{
					"label": "Allow manual operation",
					"value": 1
				}
			]
		},
		{
			"#": "5",
			"label": "Timer function",
			"description": "Timer function to automaticly turn off the Load",
			"unit": "100ms",
			"valueSize": 2,
			"minValue": 0,
			"maxValue": 32767,
			"defaultValue": 0,
			"unsigned": true
		},
    {
			"#": "8",
			"label": "Report options",
			"description": "Sets report options to association group 2",
			"valueSize": 1,
			"minValue": 0,
			"maxValue": 255,
			"defaultValue": 0,
			"unsigned": true,
			"allowManualEntry": false,
			"options": [
				{
					"label": "Report both on/off",
					"value": 0
				},
				{
					"label": "Report on only",
					"value": 1
				},
				{
					"label": "Report off only",
					"value": 2
				},
				{
					"label": "Reporting is disabled",
					"value": 255
				}
			]
		},
		{
			"#": "20",
			"label": "Local operation",
			"description": "If local operation of the switch is allowed",
			"valueSize": 1,
			"minValue": 0,
			"maxValue": 1,
			"defaultValue": 0,
			"unsigned": true,
			"allowManualEntry": false,
			"options": [
				{
					"label": "Local operation is enabled",
					"value": 0
				},
				{
					"label": "Locked, local operation is disabled",
					"value": 1
				}
			]
		},
		
		{
			"#": "21",
			"label": "Remote operation",
        "description": "If remote operation of the switch is allowed",
			"valueSize": 1,
			"minValue": 0,
			"maxValue": 1,
			"defaultValue": 0,
			"unsigned": true,
			"allowManualEntry": false,
			"options": [
				{
					"label": "Remote operation is enabled",
					"value": 0
				},
				{
					"label": "Locked, remote operation is disabled",
					"value": 1
				}
			]
		},
		{
			"#": "255",
			"label": "Factory reset",
			"valueSize": 1,
			"minValue": 0,
			"maxValue":85,
			"defaultValue": 0,
			"unsigned": true,
			"writeOnly": true,
			"allowManualEntry": false,
			"options": [
				{
					"label": "Idle",
					"value": 0
				},
				{
					"label": "Factory reset",
					"value": 85
				}
			]
		}
	]
}

itzniru, 2022-09-26:
[MH-DS221.json.txt](https://github.com/zwave-js/node-zwave-js/files/9644310/MH-DS221.json.txt)

This is what i added in custom config,  maybe its something useful in there :)

AlCalzone, 2022-09-26:
Can you provide the file as a PR please? Makes it much easier to work with, since checks are automatically done.

itzniru, 2022-09-26:
Yes, Ill do it when I get back home. I didnt see the mail until now. :)

itzniru, 2022-10-01:
When you asked to a PR I had to google it lol but now the config passed the tests it seems:
https://github.com/zwave-js/node-zwave-js/pull/5138#issue-1393577900

apella12, 2023-09-29:
This #5113 should be closed, as noted #5138 merged this device.

closed 2023-10-10 by itzniru