zwave-js/zwave-js #5114: Missing device configuration: Yale YRL256 (using AYR-MOD-ZW3, 700 series module)

vectorcatch, 2022-09-26:
### Checklist

- [X] It is not in the [configuration DB](https://devices.zwave-js.io/?search=3)

- [X] It was not [merged recently](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Apr+is%3Aclosed) or has a [pending PR](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Aopen+is%3Apr)

### Which device is missing?

Yale YRL256 (using AYR-MOD-ZW3, 700 series module)

### Manufacturer ID

0x129

### Product Type

0x800c

### Product ID

0x4f00

### Firmware Version

2.17.0

### Do you have a manual?

It is the same at the YRD256-ZW3 for the most part. They utilize the same z-wave module underneath.

AlCalzone, 2022-09-26:
> for the most part

and where are the differences? :)

thelordzero, 2022-09-26:
I have this issue as well. The new ZW3 modules supporting S2 seem to work well other than not showing the Mfg or Model in the interior locks. For deadbolt locks, the data is provided.

asucrews, 2022-10-01:
@thelordzero and @vectorcatch where did you guys get these modules? I been waiting and looking for them sinces I saw them on zwave alliance site.

vectorcatch, 2022-10-01:
@asucrews They've started showing up on eBay here and there: https://www.ebay.com/itm/403874857232

thelordzero, 2022-10-01:
Yep. Replaced all of my old modules with the new ones. At $100 a pop wasn't fun... That said they are a ton more stable than the old generations.

blhoward2, 2022-10-09:
I'm going to merge in a device file based on the YRD256, but I cannot find a manual for your device with the zwave parameters anywhere. Nor is it certified with the zwave alliance. Someone with the lock will need to test the file once it is merged.

PR #5154 (2022-11-02): feat(config): add all zwa approved yale locks
This PR now merges in all Yale locks with an entry in the ZWA database.

Closes #5144
Closes #5114
closed 2022-11-02 by robertsLando