zwave-js/zwave-js #5118: [Bug]: Cannot write to Everspring AC301 after upgrade

tenfire, 2022-09-26:
### Checklist

- [X]  I am not using Home Assistant. Or: a developer has told me to come here.
- [X] I have checked the troubleshooting section and my problem is not described there.
- [X] I have read the changelog and my problem is not mentioned there.

### Deploy method

Docker

### Z-Wave JS UI version

8.0.0

### ZwaveJS version

10.2.0

### Describe the bug

After upgrading to version 8.0.0 I found that I could no longer control my Everspring AC301. Values are still being reported in the GUI when they update. For example, the mode and temperature are updated when I change it manually on the device, but I cannot send any instructions to it from zwavejs2mqtt. I had not updated for a long time, and I don't know which version I ran before, but I tried to roll back to version 6.0.0 and then it started working again, but then after going back to 8.0.0 it stopped working.

Could it be related to this?: https://forum.z-wave.me/viewtopic.php?t=35497

### To Reproduce

Inlcude Everspring AC301 in zwavejs2mqtt version 8.0.0 and try to change any value from the GUI.

### Expected behavior

Changes in the GUI should update the device

### Additional context

Debug log when trying to change mode from "Heat" to "Off"
```
2022-09-26 12:11:05.097 INFO Z-WAVE: Calling api writeValue with args: [
{ nodeId: 4, commandClass: 64, endpoint: 0, property: 'mode' },
0,
{},
[length]: 3
]
2022-09-26 12:11:05.101 INFO Z-WAVE: Writing 0 to 4-64-0-mode
2022-09-26T10:11:05.128Z SERIAL » 0x010a001304039f01fc2509af (12 bytes)
2022-09-26T10:11:05.129Z DRIVER » [Node 004] [REQ] [SendData]
│ transmit options: 0x25
│ callback id: 9
└─[Security2CCNonceGet]
sequence number: 252
2022-09-26T10:11:05.145Z SERIAL « [ACK] (0x06)
2022-09-26T10:11:05.151Z SERIAL « 0x0104011301e8 (6 bytes)
2022-09-26T10:11:05.151Z SERIAL » [ACK] (0x06)
2022-09-26T10:11:05.152Z DRIVER « [RES] [SendData]
was sent: true
2022-09-26T10:11:06.432Z SERIAL « 0x010700130900008163 (9 bytes)
2022-09-26T10:11:06.432Z SERIAL » [ACK] (0x06)
2022-09-26T10:11:06.435Z DRIVER « [REQ] [SendData]
callback id: 9
transmit status: OK
2022-09-26T10:11:06.458Z SERIAL « 0x011a00040004149f0270016cda089fc6a69d3ff06067155252b45ef6 (28 bytes)
2022-09-26T10:11:06.459Z SERIAL » [ACK] (0x06)
2022-09-26T10:11:06.463Z DRIVER « [Node 004] [REQ] [ApplicationCommand]
└─[Security2CCNonceReport]
sequence number: 112
SOS: true
MOS: false
receiver entropy: 0x6cda089fc6a69d3ff06067155252b45e
2022-09-26T10:11:06.478Z SERIAL » 0x012c001304259f03fd011241f9593680acb31ca39ac28902e47ecdc1756421540 (46 bytes)
ebc4c74eb601b00e9920d250a06
2022-09-26T10:11:06.482Z DRIVER » [Node 004] [REQ] [SendData]
│ transmit options: 0x25
│ callback id: 10
└─[Security2CCMessageEncapsulation]
│ sequence number: 253
│ extensions:
│ · type: SPAN
│ sender EI: 0xf9593680acb31ca39ac28902e47ecdc1
└─[SupervisionCCGet]
│ session id: 49
│ request updates: true
└─[ThermostatModeCCSet]
mode: Off
2022-09-26T10:11:06.497Z SERIAL « [ACK] (0x06)
2022-09-26T10:11:06.498Z SERIAL « 0x0104011301e8 (6 bytes)
2022-09-26T10:11:06.499Z SERIAL » [ACK] (0x06)
2022-09-26T10:11:06.500Z DRIVER « [RES] [SendData]
was sent: true
2022-09-26T10:11:06.517Z SERIAL « 0x010700130a000003e2 (9 bytes)
2022-09-26T10:11:06.517Z SERIAL » [ACK] (0x06)
2022-09-26T10:11:06.518Z DRIVER « [REQ] [SendData]
callback id: 10
transmit status: OK
2022-09-26T10:11:06.571Z SERIAL « 0x011700040004119f0371003669e94c7af4aaba47c545559173 (25 bytes)
2022-09-26T10:11:06.572Z SERIAL » [ACK] (0x06)
2022-09-26T10:11:06.573Z DRIVER « [Node 004] [REQ] [ApplicationCommand]
└─[Security2CCMessageEncapsulation]
│ sequence number: 113
└─[SupervisionCCReport]
session id: 49
more updates follow: false
status: Fail
duration: 0s
2022-09-26 12:11:06.578 INFO Z-WAVE: Success zwave api call writeValue true
2022-09-26T10:11:11.588Z SERIAL » 0x01150013040e9f03fe00e54c02f25dc0680ded2c250bdf (23 bytes)
2022-09-26T10:11:11.594Z DRIVER » [Node 004] [REQ] [SendData]
│ transmit options: 0x25
│ callback id: 11
└─[Security2CCMessageEncapsulation]
│ sequence number: 254
└─[ThermostatModeCCGet]
2022-09-26T10:11:11.597Z SERIAL « [ACK] (0x06)
2022-09-26T10:11:11.602Z SERIAL « 0x0104011301e8 (6 bytes)
2022-09-26T10:11:11.607Z SERIAL » [ACK] (0x06)
2022-09-26T10:11:11.609Z DRIVER « [RES] [SendData]
was sent: true
2022-09-26T10:11:12.913Z SERIAL « 0x010700130b00008464 (9 bytes)
2022-09-26T10:11:12.915Z SERIAL » [ACK] (0x06)
2022-09-26T10:11:12.918Z DRIVER « [REQ] [SendData]
callback id: 11
transmit status: OK
2022-09-26T10:11:12.941Z SERIAL « 0x0115000400040f9f0372008686c4e45d8df5c19b749aba (23 bytes)
2022-09-26T10:11:12.943Z SERIAL » [ACK] (0x06)
2022-09-26T10:11:12.944Z CNTRLR [Node 004] [~] [Thermostat Mode] mode: 1 => 1 [Endpoint 0]
2022-09-26 12:11:12.945 INFO Z-WAVE: Node 4: value updated: 64-0-mode 1 => 1
2022-09-26T10:11:12.946Z DRIVER « [Node 004] [REQ] [ApplicationCommand]
└─[Security2CCMessageEncapsulation]
│ sequence number: 114
└─[ThermostatModeCCReport]
mode: Heat
```

robertsLando, 2022-09-26:
At first sight seems no issues on my side @AlCalzone

AlCalzone, 2022-09-26:
If you haven't done this after upgrading yet, please re-interview the device.

tenfire, 2022-09-26:
Did not help unfortunately, same goes for excluding and re-including the device. I also tried to reboot the device by removing it's batteries.

AlCalzone, 2022-09-26:
Ok, looks like we need to disable Supervision for that one, like in the issue you linked.

AlCalzone, 2022-09-26:
will be fixed in the next release

closed 2022-09-26 by AlCalzone