zwave-js/zwave-js #5120: S2: checkLifelineHealth -> ZWaveError: The node failed to decode the message. (ZW1405)

marcus-j-davies, 2022-09-26:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within Z-Wave JS UI (formerly ZwaveJS2MQTT)?

NO, my problem is NOT within Z-Wave JS UI

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

Im not entirely sure where this lies:
 - is the device simply not supporting a thing needed for lifeline checks (and maybe a more user freindly approach to be used)
 - or is it an overlook driver side 😅 
 
 Back story:
 @fabnavigator alerted me to an error that he is suddenly seeing in NR -> `checkLifelineHealth`.
 specifically this error happens quite consistently with only devices enrolled in S2.
 
 I decide to try this my self, and I received the same error.
 Some factors that may be useful.
  - I don't get the error on non secured devices
  - The S2 device I did try is a FLiRS device
  
  @fabnavigator, it will help us a lot if you can also provide a log of the error happening

### Device information

My Test device: Ring Keypad Gen 1 (Node 32)
https://devices.zwave-js.io/?jumpTo=0x0346:0x0101:0x0201:0.0

### How are you using `node-zwave-js`?

- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (latest)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (dev)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [ ] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [X] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

[image]

### Did you change anything?

no

### Did this work before?

Yes (please describe)

### If yes, where did it work?

Previously I did get results - they were shockingly bad! given its a FLiRS device 😅  but I did at least get results instead of an exception.

### Attach Driver Logfile

Checklist Begins here (1 round)
`2022-09-26T15:51:50.326Z NDERED « [API: DriverAPI] [Method: checkLifelineHealth]`

Fails here
`2022-09-26T15:51:53.410Z NDERED   [ERROR] [INPUT] The node failed to decode the message. (ZW1405)`

[zwave.log](https://github.com/zwave-js/node-zwave-js/files/9647876/zwave.log)

marcus-j-davies, 2022-09-26:
Cool.

I Wasn't sure if my report was a device specific problem, if @fabnavigator  doesn't deliver any logs - I see no reason to keep this open, if my issue is isolated to the device 😅

fabnavigator, 2022-09-26:
Here is the log for the failure I'm seeing.

[zwave-js.log](https://github.com/zwave-js/node-zwave-js/files/9650181/zwave-js.log)

AlCalzone, 2022-09-27:
> DRIVER   version 10.0.0-beta.4

@fabnavigator you should to update to latest before doing any further tests involving this

fabnavigator, 2022-09-27:
@marcus-j-davies Shouldn't I have the latest if I'm running your latest debug version?

marcus-j-davies, 2022-09-27:
@fabnavigator 
Then you Haven't updated to the release version 😅 

v8 was released the other day (with Driver version 10.2.0)
https://github.com/zwave-js/node-red-contrib-zwave-js/releases/tag/v8.0.0

fabnavigator, 2022-09-27:
@marcus-j-davies Got it. I'm at v8 now. I kicked off my normal health check. If you don't hear back from me, assume all is well.

fabnavigator, 2022-09-27:
Same problem as before. Log file attached.
[zwave-js.log](https://github.com/zwave-js/node-zwave-js/files/9659996/zwave-js.log)

AlCalzone, 2022-09-28:
Ok this one is a little bit different. 1 second after starting the healthcheck, the driver queries the node to update the status. At the same time the health check status is received, causing a collision, which causes the query to be re-transmitted, but now with the wrong/outdated SPAN. I've raised https://github.com/zwave-js/node-zwave-js/issues/5126 to track this.
We can actually go a step further in this case, take a shortcut, and not re-transmit the query at all, since the received report is the expected response to it (accidentally): https://github.com/zwave-js/node-zwave-js/issues/5127