zwave-js/zwave-js #5122: Investigate xstate memory leak

AlCalzone, 2022-09-26:


zener82, 2022-09-28:
Here is my log contribution if needed during this night until today 8AM. 

[debugzwavejs.log.txt](https://github.com/zwave-js/node-zwave-js/files/9661890/debugzwavejs.log.txt)

I badly configured to info level instead of debug.
So I've changed it to debug and restart the add-on.

spraguej248, 2022-09-28:
Log taken after fixes applied.  Thank you!

[zwave_js (1).log](https://github.com/zwave-js/node-zwave-js/files/9667686/zwave_js.1.log)

AlCalzone, 2022-09-30:
Thanks for those logs. I probably won't be able to review them in the near future, but good to have them anyways.

AlCalzone, 2023-01-30:
Possibly fixed, need to update and test: https://github.com/statelyai/xstate/issues/3616

AlCalzone, 2023-08-11:
Confirmed fixed

closed 2023-08-11 by AlCalzone