zwave-js/zwave-js #5124: After successful supervised MultilevelSwitchCCSet (255) query the actual value quicker

AlCalzone, 2022-09-27:
```
2022-09-26T14:38:44.203Z DRIVER « [Node 004] [REQ] [ApplicationCommand]
└─[MultiChannelCCCommandEncapsulation]
│ source: 1
│ destination: 0
└─[SupervisionCCReport]
session id: 19
more updates follow: false
status: Success
duration: 0s
```

We don't need to wait 5 seconds after such a report.

PhilipTrauner, 2025-07-04:
at the risk of having overlooked something, wouldn't calling `CCAPI.pollValue` instead of `CCAPI.schedulePoll` in [`verifyChanges`](https://github.com/zwave-js/zwave-js/blob/570fa667e41853fc594a7ded53972ae354649b99/packages/cc/src/cc/MultilevelSwitchCC.ts#L471) be sufficient then?

[`schedulePoll`](https://github.com/zwave-js/zwave-js/blob/570fa667e41853fc594a7ded53972ae354649b99/packages/zwave-js/src/lib/node/mixins/60_ScheduledPoll.ts#L113) [already calls `pollValue` internally](https://github.com/zwave-js/zwave-js/blob/570fa667e41853fc594a7ded53972ae354649b99/packages/zwave-js/src/lib/node/mixins/60_ScheduledPoll.ts#L149), therefor i assume there would be no difference from the perspective of the request handling machinery (`Driver.handleRequest` → `Node.handleCommand` → `CCHandlers.handleMultilevelSwitchCommand` → `ValueDB.setValue`), right?

AlCalzone, 2025-07-04:
I think that would be okay - for the specific case of successful supervised commands.

PR #7963 (2025-07-11): fix: query actual value immediately after successful supervised `Multilevel Switch Set` with value 255
fixes #5124

previously a poll was scheduled after the `transitionDuration`.
scheduling a poll for a later  is not neccessary, because `verifyChanges` is only run *after* the value has already changed.