zwave-js/zwave-js #5126: S2: Increment SPAN before re-transmitting a command after a CAN

AlCalzone, 2022-09-28:
```
2022-09-27T22:15:16.635Z SERIAL » 0x011a00a9012a0e9f03b100828c48961b18ac2a835c25000000009873          (28 bytes)
2022-09-27T22:15:16.635Z DRIVER » [Node 042] [REQ] [SendDataBridge]
                                  │ source node id:   1
                                  │ transmit options: 0x25
                                  │ callback id:      152
                                  └─[Security2CCMessageEncapsulation]
                                    │ sequence number: 177
                                    └─[PowerlevelCCTestNodeGet]
2022-09-27T22:15:16.636Z SERIAL « 0x011b00a800012a129f03d100828fdaeb2f30ad78382a8115da5a00c60e        (29 bytes)
2022-09-27T22:15:16.638Z SERIAL » [ACK]                                                                   (0x06)
2022-09-27T22:15:16.639Z DRIVER « [Node 042] [REQ] [BridgeApplicationCommand]
                                  │ RSSI: -58 dBm
                                  └─[Security2CCMessageEncapsulation]
                                    │ sequence number: 209
                                    └─[PowerlevelCCTestNodeReport]
                                        test node id:        1
                                        status:              Success
                                        acknowledged frames: 10
2022-09-27T22:15:16.640Z CNTRLR « [Node 042] received expected response prematurely, remembering it...
2022-09-27T22:15:16.641Z SERIAL « [CAN]                                                                   (0x18)
2022-09-27T22:15:16.643Z CNTRLR   Failed to execute controller command after 1/3 attempts. Scheduling next try i
                                  n 100 ms.
2022-09-27T22:15:16.743Z SERIAL » 0x011a00a9012a0e9f03b100828c48961b18ac2a835c25000000009873          (28 bytes)
2022-09-27T22:15:16.744Z DRIVER » [Node 042] [REQ] [SendDataBridge]
                                  │ source node id:   1
                                  │ transmit options: 0x25
                                  │ callback id:      152
                                  └─[Security2CCMessageEncapsulation]
                                    │ sequence number: 177
                                    └─[PowerlevelCCTestNodeGet]
```
The second `PowerlevelCCTestNodeGet` is identical to the first one, which guarantees that the device won't understand it.

fabnavigator, 2022-11-11:
@AlCalzone Any thoughts about this issue and 5127?

AlCalzone, 2022-11-11:
See https://github.com/zwave-js/node-zwave-js/issues/5129 😇

AlCalzone, 2023-03-08:
Closing this for now. I rewrote the nonce handling in #5475 to be more deterministic. The change should also be able to handle this. If not, I'll reopen.

closed 2023-03-08 by AlCalzone