zwave-js/zwave-js #5134: Older Qubino Flush 1 Relay shows up in HA's 'energy' view, new one doesn't

tanuva, 2022-09-30:
G'day,

I have one older and one new Qubino Flush 1 Relay (ZMNHAD). The new one is going to be connected to a small photovoltaics setup, thus I tried to configure it as 'energy input' in Home Assistant's new-ish energy tab. However, its energy entity doesn't show up as selectable there. Interestingly, the old ZMNHAD is indeed listed there.

According to zwavejs2mqtt, the old one runs firmware 6.10, the new one runs 71.0. (Both didn't want to connect securely.)
What could I investigate to make the new one show up in HA's energy panel as well?

AlCalzone, 2022-09-30:
You can start by filling out the issue template (bug report) and providing the necessary information.
The most important part is a driver log on loglevel "debug" which includes a re-interview of the problematic device.

tanuva, 2022-09-30:
I'm sorry I didn't catch the bug report form. Apparently Github didn't show the issue type chooser to me because I needed to log in first. Strange. Given the differentiation between zwave-js and the zwave-js integration in HA I'll turn to HA/core first. If they redirect me here again I'll open a new issue through the bug report form.

By the way, congratulations to your future `fork()`, have a great time!

closed 2022-09-30 by tanuva