zwave-js/zwave-js #5135: NAS-PD07Z incorrect info

shadowroamer, 2022-09-30:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within Z-Wave JS UI (formerly ZwaveJS2MQTT)?

NO, my problem is NOT within Z-Wave JS UI

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

It appears the configuration parameters that were automatically retrieved by Z-Wave JS HA integration (shown on https://devices.zwave-js.io/?jumpTo=0x0258:0x0010:0x0720:0.0#) do not match the configuration parameters shown in the manual (found on https://manuals.plus/m/ee5fc2ba34679979a2ed9b8feeaa0d302f0a91252eb7c0836bc7f9183fcc9a74_optim.pdf)

For example, the Config DB only goes up to 15, while the manual goes up to 18 parameters (excluding 99 on both counts).

### Device information

Manufacturer: Shenzhen Neo Electronics Co., Ltd. 
Model name:  NAS-PD07Z

### How are you using `node-zwave-js`?

- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (latest)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (dev)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [X] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

Driver Version:
10.2.1-0-pr-5108-7ec35dd
Server Version:
1.23.1

### Did you change anything?

no

### Did this work before?

No, it never worked anywhere

### Attach Driver Logfile

No log available, this is a config mismatch

AlCalzone, 2022-09-30:
Please verify that this is actually a mistake in our files. We've had this complaint at least twice now and each time the device was not what it was supposed to be.

See https://github.com/zwave-js/node-zwave-js/issues/4249 and other closed issues when searching for PD07Z.

shadowroamer, 2022-09-30:
Ok, so assuming the device isn't what it's supposed to be, how do I correct it?

AlCalzone, 2022-10-01:
The last times either the manual packed with the device was wrong, and the device was either a different one sold as the 07Z.
In the linked issue you'll find some pointers at testing the configuration parameters to see if they actually match the manual, or if the ones we have are still correct.

If you find that the manual is indeed correct, we need to figure out which firmware version changed the config parameters, so we can [change them depending on that version](https://zwave-js.github.io/node-zwave-js/#/config-files/conditional-settings).

closed 2023-01-13 by zwave-js-assistant[bot]