zwave-js/zwave-js #5139: Device does not change parameters/changes leads to same value

TutenStain, 2022-10-03:


TutenStain, 2022-10-03:
Attaching logs. Should be the last thing in the files as I enabled logging, attempted to change value and then downloaded the logs.

[zwave-js-ui-store.zip](https://github.com/zwave-js/node-zwave-js/files/9697604/zwave-js-ui-store.zip)

robertsLando, 2022-10-03:
```
2022-10-03T12:45:21.387Z DRIVER « [Node 084] [REQ] [ApplicationCommand]
                                  └─[SupervisionCCReport]
                                      session id:          42
                                      more updates follow: false
                                      status:              Fail
                                      duration:            0s
2022-10-03T12:45:21.392Z CNTRLR   [Node 084] [setValue] result of SET_VALUE API call for ThermostatModeCCAPI: (S
                                  upervisionResult)
                                    status:   Fail
2022-10-03T12:45:21.393Z CNTRLR   [Node 084] [setValue] not updating value
```

@AlCalzone I think the problem could be because the node is reporting to support SpervisionCC but it actually doesn't support it, may need a compat flag too work?

robertsLando, 2022-10-03:
Ok so just wait for https://github.com/zwave-js/zwave-js-ui/pull/2702 to be merged

TutenStain, 2022-10-03:
Awesome, thanks!

robertsLando, 2022-10-05:
Merged and released (8.1.0)

closed 2022-10-05 by robertsLando