zwave-js/zwave-js #5141: Missing device configuration: Nexia  ZSENS930A

seajack0, 2022-10-05:
### Checklist

- [X] It is not in the [configuration DB](https://devices.zwave-js.io/?search=3)

- [X] It was not [merged recently](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Apr+is%3Aclosed) or has a [pending PR](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Aopen+is%3Apr)

### Which device is missing?

Nexia 

### Manufacturer ID

0x0178:0x5253:0x5433:1.63

### Product Type

0x5253

### Product ID

0x5433

### Firmware Version

1.63

### Is the device listed on the Z-Wave Alliance website?

https://products.z-wavealliance.org/products/1939?selectedFrequencyId=2

### Do you have a manual?

Link to manual [https://wdi.rfwel.com/cdn/techdocs/ZSENS930AW00MAnual.pdf](url)

This device appears to be identical to [TH100NX](https://devices.zwave-js.io/?jumpTo=0x0178:0x5448:0x3130:0.0)

Thanks!

seajack0, 2022-10-06:
Added configuration and submitted pull request.

AlCalzone, 2022-10-07:
Let's keep this open until the PR is reviewed and merged

PR #5142 (2022-10-08): feat(config): add support for Nexia ZSENS930

Added config file for Nexia ZSENS930

fixes: #5141
closed 2022-10-08 by robertsLando