zwave-js/zwave-js #5144: Missing device configuration: <enter manufacturer and device name>

tamorgen, 2022-10-06:
### Checklist

- [X] It is not in the [configuration DB](https://devices.zwave-js.io/?search=3)

- [X] It was not [merged recently](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Apr+is%3Aclosed) or has a [pending PR](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Aopen+is%3Apr)

### Which device is missing?

Yale Assure Lock 2

### Manufacturer ID

297

### Product Type

33028

### Product ID

0x8104

### Firmware Version

1.78

### Is the device listed on the Z-Wave Alliance website?

YES

### Do you have a manual?

[400 series Keyed.pdf](https://github.com/zwave-js/node-zwave-js/files/9724376/400.series.Keyed.pdf)

AlCalzone, 2022-10-06:
Can you give all the ids in their 0x... form and share the link to the entry on the Alliance website?

tamorgen, 2022-10-06:
@AlCalzone , where can I find the 0x codes in zwave js ui?   All I was able to find was what I provided.

Here is a link to the product.  Note, there are several variations (Keyed, not keyed, Touchpad, button)
https://products.z-wavealliance.org/products/4651?selectedFrequencyId=2

Variations:
https://products.z-wavealliance.org/Search/Index?regionId=2&searchText=assure+2

AlCalzone, 2022-10-06:
> where can I find the 0x codes in zwave js ui?

When you open the drop down panel for the node, it should be right at the start, sth. like 0x1234-0x2345-0xabcd

tamorgen, 2022-10-06:
297-1490-33028 (0x0129-0x8104-0x05d2

PR #5154 (2022-11-02): feat(config): add all zwa approved yale locks
This PR now merges in all Yale locks with an entry in the ZWA database.

Closes #5144
Closes #5114
closed 2022-11-02 by robertsLando