zwave-js/zwave-js #5145: Missing device configuration: <Ring Glass Break Sensor >

Cloakedman, 2022-10-07:
### Checklist

- [X] It is not in the [configuration DB](https://devices.zwave-js.io/?search=3)

- [X] It was not [merged recently](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Apr+is%3Aclosed) or has a [pending PR](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Aopen+is%3Apr)

### Which device is missing?

Ring Glass Break Sensor

### Manufacturer ID

0x0346

### Product Type

0x0301

### Product ID

0x0a01

### Firmware Version

1.11.0

### Is the device listed on the Z-Wave Alliance website?

https://products.z-wavealliance.org/products/4554/configs

AlCalzone, 2022-10-07:
@zwave-js-bot import config 4554 from zwa

Cloakedman, 2022-10-07:
> @zwave-js-bot import config 4554 from zwa

Is that for me or the bot? I'm hoping this will be an easy add... I don't know how to import it myself.

Cloakedman, 2022-10-07:
> Sorry Cloakedman, you're not authorized to do that 🙁!

@zwave-js-bot ok cool... Can you do it? :) look forward to being able to get this sensor to work hopefully.

blhoward2, 2022-10-08:
@Cloakedman that's the bot. Go to Pull Requests. Then you can PR your changes to that branch and we can review and merge them in.

blhoward2, 2022-10-08:
Actually, I don't see a PR so it looks like the import failed with the bot. You may have to make it by hand. See the docs for a style guide and instructions.

Cloakedman, 2022-10-08:
> Actually, I don't see a PR so it looks like the import failed with the bot. You may have to make it by hand. See the docs for a style guide and instructions.

This stuff is way over my head. I'm just trying to get this ring glass break sensor to work. I thought if the parameters were known on the zwave alliance page that I posted, someone can add that to the db. I have no clue how to do that.

blhoward2, 2022-10-08:
We can, eventually. It'll take time until we get it it. In the meantime you can set any parameter in zwave-js-ui under advanced under the config parameters. 

The file really isn't needed for it to work, though, unless you need to set a parameter and can't use zwave-js-ui.

AlCalzone, 2022-10-08:
@blhoward2 Can you  access the actions logs? Then you can trigger botty and check the corresponding "react to comments" action for errors.

blhoward2, 2022-10-08:
Yes. It's a malformed database entry on the alliance's side. It's missing a device fingerprint so the formatting for the type/id is failing. I've been meaning to make it more resilient to this so it imports with a placeholder.

blhoward2, 2022-10-08:
@AlCalzone #5148 fixes the importConfig routine, if you ever have a chance to look at it. If not, it can wait until you're back.

blhoward2, 2022-10-08:
@Cloakedman #5149 is the draft device file for this device. It needs to be cleaned up in accordance with our style guide (https://zwave-js.github.io/node-zwave-js/#/config-files/style-guide). You can either try to do so and merge in your changes, or it'll have to wait until I have more time to get to it.

Doing it yourself involves forking the node-zwave-js repo, going to your fork, switching to the branch that matches my pull request (ring_glass_break_sensor), entering VS Code (hit the . on your keyboard), making the changes, committing the changes within VS Code, then creating a PR asking to merge in those changes.

Or, as an in-between solution, I suppose you could download the json, correct it all, then send it to me and I'll upload it when I have time.

Cloakedman, 2022-10-09:
@blhoward2 thanks so much to you all for your hard work. I was able to change parameter 11 to a value of 1 using freshcoasts suggestion which will get my running for now. This config is Def important though since this sensor is useless for HA without that parameter being set to 1. Maybe that can be the default. Thanks again

blhoward2, 2022-10-09:
Glad it's working. We can't and don't set any parameters on our own. The default in our files is just for the UI to indicate which setting is the default. Otherwise, we'd clobber people's settings.

AlCalzone, 2023-03-06:
fixed in #5149

closed 2023-03-06 by AlCalzone