zwave-js/zwave-js #5146: JASCO Plug-in 1-Outlet Switch, 500S (28169 / ZW4103) - Parameter 3

bearhntr, 2022-10-08:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

YES, BUT a Home Assistant developer has told me to come here

### Is your problem within Z-Wave JS UI (formerly ZwaveJS2MQTT)?

NO, my problem is NOT within Z-Wave JS UI

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?  I have 6 of these in my home and all of them are now running Firmware 5.25 (thank you HA team for updates).

What do you observe?  I have discovered something (to me) frustrating. This parameter is wrong: [9-112-0-3] LED Indicator

According to everything I can find and from the ZWave JS site - if this is set to “0” - the light is ON when the Device is OFF (which is default and what I want)…so that you can find as needed in the dark.

What did you expect to happen?   Everyone of mine, when I set them to “0” - the light will be OFF when the device is OFF (which would indicate parameter 3 is set to a ‘1’. 

Steps to reproduce the behavior:
In order to get the LED to be ON when the switch is OFF - I have to set the parameter to “1” - not “0” as the site would indicate.
I have also tried the (press 10 times feature) and it does change the LED - but when you interview the node - Parameter 3 = “1” not “0”.

[image]

### Device information

Manufacturer:   JASCO/GE
Model name:  28169 / ZW4103
Node ID in your network: <5, 6, 7, 8, 9, 10>

### How are you using `node-zwave-js`?

- [X] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (latest)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker image (dev)
- [ ] `zwave-js-ui` (formerly `zwavejs2mqtt`) Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [ ] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version:
`node-zwave-js` branch:  Z-Wave JS  Current version: 0.1.74
`zwave-js-ui` branch:  Z-Wave JS UI  Current version: 1.1.0 

### Did you change anything?

no

### Did this work before?

Yes (please describe)

### If yes, where did it work?

Before these devices were updated - I did not know if PARAMETER 3 was set to "0" or "1" - if they were not working I would perform the 10-taps method to change the LED.

### Attach Driver Logfile

I am not able to find this file

blhoward2, 2022-10-09:
Did this same behavior happen before the firmware upgrade? Or is this new behavior after the firmware upgrade?

bearhntr, 2022-10-10:
@blhoward2 

That I cannot really answer.  I was using the Z-Wave JS option and not the Z-Wave UI that I am now.  I was not really able to see the parameters which were in use and their values.  In the past, when I would add something and if the LED was not what I wanted (LED ON when the device was OFF) - I would just press the button 10x and the light would change.  
  
  I like having the LED illuminated when the device is OFF (specifically for the lamps) as in the dark you can find the button.  :-)

bearhntr, 2022-10-11:
I just tested a FACTORY RESET of one of these plugs with the issue --- it does the same thing.  Parameter 3 set to "0" and the LED comes on when the SWITCH = ON.  If I press the button 10x times.  It fixes the LED and parameter 3 is now showing "1"

blhoward2, 2022-10-11:
That doesn’t really help. I believe that your plugs do this. What I’m trying to figure out is if that behavior changes at a certain firmware version so that we can properly condition the change so both work correctly.

bearhntr, 2022-10-12:
@blhoward2,

  My switches were at different FW levels before the HA push of the FW on them now.  As I stated - I cannot say that this changed when FW was updated.  How I fixed in the past was the 10x press on the button (as I had read that on a forum long ago).  It was not until I moved to HA and realized that I could "see" the parameter values - that I learned of them.

  It appears that I can downgrade the FW on them...and I am willing to test this if you wish to see where it might have changed.  

  As I also have a few other plugs which are doing the same thing...while others are not.   Should I make a spreadsheet for you with this info?  Anything to help.

blhoward2, 2022-10-12:
Open issues for anything that needs to be changed, one issue per device, please. So we can keep track. What's important is at what fw these happened if we can determine that. If not it's not a major issue so I guess we'll just change it and see if anyone complains.

kpine, 2022-10-12:
Can Jasco help you [confirm](https://github.com/zwave-js/node-zwave-js/pull/5038#issuecomment-1243839055)?

closed 2023-03-19 by zwave-js-assistant[bot]